betaflight/betaflight #9351: Can't upgarde firmaware chip image is larger

Pos3idonas, 2020-01-11:
**Describe the bug**
When im trying to flash new firmware appear the message.
" Error: Supplied image is larger then flash available on the chip! Image: 448.0 KiB, limit = -2354048.0 KiB"

**Setup / Versions**
SP RACING H7 EXTREME

[HERE IS AN IMAG](<blockquote class="imgur-embed-pub" lang="en" data-id="a/jKpoykB"><a href="//imgur.com/a/jKpoykB">SP RACING EXTREME FIRMWARE FLASH ISSUE</a></blockquote><script async src="//s.imgur.com/min/embed.js" charset="utf-8"></script>)E

etracer65, 2020-01-11:
I just tested with that flight controller and had no problems flashing the 4.1.2 firmware.

Make sure you have the official Configurator release. The version you have doesn't appear to be the correct build.

https://github.com/betaflight/betaflight-configurator/releases/latest

jflyper, 2020-01-11:
@Pos3idonas 
The error you've encountered happens when you try to flash a local image while board and version are specified by the dropdown. Reselect those two dropdown lists to unselected and try again.

Pos3idonas, 2020-01-11:
> 
> 
> @Pos3idonas
> The error you've encountered may happen when you try to flash a local image while board and version are specified by the dropdown. Reselect those two dropdown lists to unselected and try again.

No, all the procedure done with the online choice.I was load the firmware from online and i was try to flash. 
I am one step before to try to flash by sd card.

Pos3idonas, 2020-01-11:
> 
> 
> I just tested with that flight controller and had no problems flashing the 4.1.2 firmware.
> 
> Make sure you have the official Configurator release. The version you have doesn't appear to be the correct build.
> 
> https://github.com/betaflight/betaflight-configurator/releases/latest

No was the bf 10.6.0, but i uninstall it and re install it again, with no luck.
Same problem!!.

Pos3idonas, 2020-01-11:
The FC is full wired setup up to the quad!. You think may be this a issue?.
Before i complete the setup i was flash it without problem but now something is going wrong!.
Also i cant login to the configurator for use the functions of the FC.

McGiverGim, 2020-01-12:
The size of the limit that appears in your message, negative, let me think on an overflow. Is the H7 firmware installed in an SDCard? If true, which size?

Pos3idonas, 2020-01-13:
The FC have 2 buttons one "boot" and one "bind", I was make mistake and i was try to make it work with the "boot" but want with the "bind" for flashing firmware, so easy solution, front of my face. Sorry guys!.

closed 2020-01-13 by Pos3idonas