betaflight/betaflight #9353: User variable for version control

TwoToneEddy, 2020-01-11:
Hi,
This is an odd request but people may find it useful. I'd like there to be a user variable with no function of int type. The reason for this is so it would be possible use this variable as a configuration number. The reason I would find this useful is because I have 3 quads that are identical, if I want to change something in the configuration I do the change on 1 quad, test it, then apply it to the others. It would be good if I could increment a variable once i've changed a configuration so it is obvious which configuration is on which quad. Does a user variable already exist? Does this sound like a reasonable request or am I anal?

etracer65, 2020-01-11:
Use the “craft name” field (on the Configuration) tab.

TwoToneEddy, 2020-01-11:
Cool, I didn't think about that. Thanks!

closed 2020-01-11 by TwoToneEddy