betaflight/betaflight #9355: autonomous drive ,,, dronekit

anwar1awad, 2020-01-12:
hi
am kinda new to drones science , am just trying to make an autonomous drone , 
but i cant get the FC (matek f405 std) to work with dronekit , i couldnt find anything to state if betaflight is supported by dronekit and mavlink , 
so my question is , does my FC (betaflight) work with dronekit ?
or if not , is there any way to accomplish that ? is there any other protocol or special wiring i need to do so i can get my onboard computer to drive the drone ?

 thanks for all for your efforts

mikeller, 2020-01-12:
@anwar1awad: For what I can tell DroneKit only supports Mavlink as the control protocol for the controlled craft - this protocol isn't supported in Betaflight for RC control.
You might want to look into using ardupilot instead - they support the Matek F405 as well: https://ardupilot.org/copter/docs/common-matekf405.html

anwar1awad, 2020-01-12:
thanks @mikeller 
what is the protocol betaflight uses ?? is there anyway to achieve the same autonomous drive functionality ? i mean control over telemetry or even usb

mikeller, 2020-01-12:
@anwar1awad: Have a look into https://github.com/betaflight/betaflight/tree/master/src/main/rx - these are the RC protocols supported by Betaflight. Feel free to open a pull request to add support for any of these to DroneKit.

closed 2020-02-18 by stale[bot]

iottrends, 2021-08-30:
I am also exploring Beta-flight... wanted to check does Beta-flight exposes any API's which can be used for autonomous flying.
with my research online adrupilot and pix4 both supports dronekit api's. Any document which points, if there is work in Progress, or Beta flight with stay away with autonomous flying!