betaflight/betaflight #9364: RANGEFINDER - Working or not?

kc10kevin, 2020-01-14:
Searched a couple of other issues concerning the, but not clear.  Also noticed that the unified target STM32xxx target.mk does not include Rangefinder drivers.  Manufacturer I work for asking LIDAR would still be supported or not?

Thanks in advance.

etracer65, 2020-01-14:
Without researching (and I can't now), I'd guess that rangefinder support hasn't been implemented for unified targets. Even so it doesn't mean much. Betaflight wan't do anything with the rangefinder data other than possibly reporting altitude. There is no altitude hold support.

mikeller, 2020-01-15:
@kc10kevin: Pretty much what @etracer65 said. Rangefinder support was added to Betaflight at some point, but a short time later altitude hold support was dropped as it was unmaintained and had not been working for a long time. Now we have rangefinder drivers, but nothing that makes use of their output.
As a consequence, adding dynamic configuration support for the rangefinder drivers never was a priority, and has not yet been done. This is usually not much work, and you could do it yourself if you wanted, happy to help with advice.

kc10kevin, 2020-01-15:
@mikeller, @etracer65:  Thanks for the info!   Something I may look at if I can find the time.  Much appreciated.

closed 2020-01-15 by kc10kevin