betaflight/betaflight #9366: pinio active for a second on powerup

robertlacroix, 2020-01-15:
Betaflight 4.1.2 (and possibly earlier versions) briefly pull pinio high on powerup and cause my Unify Pro32 Nano to briefly transmit video although it shouldn't.

This is somewhat related to #7478, which concluded that the initialization glitch on powerup is short enough that it doesn't cause an issue, but with newer VTX it actually does become an issue because they apparently transmit earlier.

Is there anything that can be improved on this front @jflyper? Is about a second expected?

```
diff

# version
# Betaflight / STM32F405 (S405) 4.1.2 Dec 22 2019 / 02:19:35 (89051e256) MSP API: 1.42
# manufacturer_id: AIRB   board_name: OMNIBUSF4SD   custom defaults: YES

# start the command batch
batch start

board_name OMNIBUSF4SD
manufacturer_id AIRB

# name: JFK T2

# resources
resource SERIAL_RX 1 NONE
resource PINIO 1 A10

# feature
feature TELEMETRY

# beacon
beacon RX_SET

# map
map TAER1234

# serial
serial 2 64 115200 57600 0 115200
serial 5 2048 115200 57600 0 115200

# aux
aux 0 0 3 1900 2100 0 0
aux 1 27 0 900 1050 0 0
aux 2 27 0 1100 2100 0 0
aux 3 13 1 1900 2100 0 0
aux 4 26 1 1400 1600 0 0
aux 5 35 1 1900 2100 0 0
aux 6 40 0 1050 1100 0 0
aux 7 49 2 1900 2100 0 0

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 LOWBAND  L CUSTOM  5362 5399 5436 5473 5510 5547 5584 5621
vtxtable powerlevels 3
vtxtable powervalues 14 20 26
vtxtable powerlabels 25 100 400

# master
set gyro_sync_denom = 2
set gyro_lowpass_hz = 0
set gyro_lowpass2_hz = 0
set yaw_spin_threshold = 1000
set dyn_notch_width_percent = 0
set dyn_notch_q = 200
set dyn_notch_min_hz = 200
set dyn_lpf_gyro_min_hz = 0
set dyn_lpf_gyro_max_hz = 0
set acc_calibration = 37,0,-26
set min_check = 1024
set max_check = 2000
set max_aux_channels = 4
set serialrx_provider = FPORT
set serialrx_halfduplex = ON
set blackbox_p_ratio = 64
set dshot_idle_value = 500
set dshot_burst = OFF
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set vbat_max_cell_voltage = 440
set vbat_warning_cell_voltage = 300
set vbat_scale = 111
set ibata_scale = 153
set beeper_dshot_beacon_tone = 3
set yaw_motors_reversed = ON
set small_angle = 180
set pid_process_denom = 1
set transient_throttle_limit = 10
set frsky_vfas_precision = 1
set osd_vbat_pos = 353
set osd_rssi_pos = 40
set osd_tim_1_pos = 54
set osd_tim_2_pos = 33
set osd_flymode_pos = 333
set osd_throttle_pos = 2424
set osd_vtx_channel_pos = 377
set osd_ah_pos = 200
set osd_current_pos = 2454
set osd_mah_drawn_pos = 2434
set osd_craft_name_pos = 2049
set osd_gps_speed_pos = 218
set osd_gps_lon_pos = 82
set osd_gps_lat_pos = 65
set osd_gps_sats_pos = 51
set osd_home_dir_pos = 302
set osd_home_dist_pos = 303
set osd_compass_bar_pos = 266
set osd_altitude_pos = 247
set osd_pid_roll_pos = 423
set osd_pid_pitch_pos = 455
set osd_pid_yaw_pos = 487
set osd_debug_pos = 1
set osd_power_pos = 321
set osd_pidrate_profile_pos = 345
set osd_warnings_pos = 2441
set osd_avg_cell_voltage_pos = 2401
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_battery_usage_pos = 392
set osd_disarmed_pos = 138
set osd_nheading_pos = 311
set osd_nvario_pos = 279
set osd_esc_tmp_pos = 82
set osd_esc_rpm_pos = 83
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set osd_stat_bbox = OFF
set debug_mode = GYRO_SCALED
set scheduler_optimize_rate = ON
set vtx_band = 5
set vtx_channel = 8
set vtx_power = 1
set vtx_freq = 5917
set pinio_box = 40,0,0,0
set gyro_1_align_yaw = 2700
set name = JFK T2

profile 0

# profile 0
set dyn_lpf_dterm_min_hz = 60
set dyn_lpf_dterm_max_hz = 225
set dterm_lowpass2_type = BIQUAD
set dterm_lowpass2_hz = 200
set iterm_relax = RPY
set throttle_boost = 0
set p_pitch = 55
set i_pitch = 87
set d_pitch = 30
set f_pitch = 105
set p_roll = 55
set i_roll = 83
set d_roll = 30
set f_roll = 105
set p_yaw = 55
set i_yaw = 0
set d_yaw = 45
set abs_control_gain = 10
set use_integrated_yaw = ON
set d_min_roll = 23
set d_min_pitch = 24
set d_min_boost_gain = 35
set d_min_advance = 0
set launch_control_mode = PITCHONLY
set launch_angle_limit = 45

rateprofile 0

# rateprofile 0
set thr_mid = 30
set thr_expo = 45
set roll_rc_rate = 175
set pitch_rc_rate = 175
set yaw_rc_rate = 175
set roll_expo = 47
set pitch_expo = 47
set yaw_expo = 47
set roll_srate = 38
set pitch_srate = 38
set yaw_srate = 38
set tpa_rate = 80
set tpa_breakpoint = 1500

# end the command batch
batch end
```

etracer65, 2020-01-15:
Pins on the MCU have different default states on power up. This is before Betaflight starts to run and outside our control. You might try a different pin or possibly invert your power control logic.

mikeller, 2020-01-15:
@robertlacroix: If the 'glitch' is a second in length then this is probably the glitch caused by the 'default' value for the RC channel in question (usually 1500) resulting in the pin being set to 'VTX active' - check the respective RC range to make sure that the default value results in 'VTX disabled'.
The initial delay between power up and RC channel initialisation is only 50 milliseconds or so, which isn't enough for the VTX to actually start emitting.

robertlacroix, 2020-01-15:
@mikeller I thought so too initially, but that didn’t turn out to be it (see diff). It also happens without an RX connected fwiw.

```
aux 6 40 0 1050 1100 0 0
```

jflyper, 2020-01-15:
@robertlacroix What is PA10 on board?

etracer65, 2020-01-15:
From the STM32F405 data sheet it looks like PA10's initial state is floating. You could try to add a week pull-down resistor (like 10K) to hold it low during initialization.

robertlacroix, 2020-01-15:
> @robertlacroix What is PA10 on board? And, what level do you expect on that pin with the given config diff?

@jflyper RX1. And it should be low

jflyper, 2020-01-15:
All GPIO pins are initialized to input pull-up at initialization by `enableGPIOPowerUsageAndNoiseReductions` (or similar).
So, PA10 would change from floating (at reset) to pull-up (at system init) and then to some device specific state.

In addition to this, I observe several ups and downs on pinio pins, which I can't explain why at the moment.

@robertlacroix 
> @jflyper RX1. And it should be low

The line stays high with your config...

robertlacroix, 2020-01-15:
@jflyper 
> The line stays high with your config...

Hmm! Why is that? Pinio box 40 (user1) is configured from 1050 to 1100, so it should stay low, shouldn’t it (unless aux1 is in that range - which it isn’t)?

jflyper, 2020-01-15:
I don't know, what is the exact board you have? Genuine Omnibus F4 V2~V5?

robertlacroix, 2020-01-15:
> I don't know, what is the exact board you have? Genuine Omnibus F4 V2~V5?

The diff is from a Hobbywing FC G2, but it happens on G3, and SynergyF4 as well

jflyper, 2020-01-15:
Hmmm... My Omnibus F4 V3 may have a pull-up or something on RX1.

Can you temporary switch to PA1 (or PA8) so that we are looking at the same problem? I see 0.5sec high period during initialization on pinio pins configured on these pins.

robertlacroix, 2020-01-16:
@jflyper had a busy day at work today, I’ll check it out tomorrow!

jflyper, 2020-01-16:
@robertlacroix Can you test the STM32F405 unified target firmware in #9376 ?

robertlacroix, 2020-01-16:
Sweet, thanks! Will test first thing tonight

robertlacroix, 2020-01-17:
@jflyper awesome, tested with your change and it's a lot better 🎉! Is there anything else we can do? There's still a – now much smaller – blip: https://imgur.com/sbKuG6E.

The reason why it's important that powering up doesn't cause any interference is team racing, where a team mate is in the air flying while you get ready to take off as soon as his battery is done.

mikeller, 2020-01-17:
Looks like this could have been triggered by the recent changes in initialisation order. We are playing tag there with what is getting done when, with everything contesting for the pole position.

@jflyper: Could we extend the existing pull up / pull down facility to be applied to any pin _before_ it is configured for its original purpose? I think that adding this as a parameter for pinio will solve it for some applications, but still leave other applications broken.

etracer65, 2020-01-17:
@robertlacroix Have you tried a pull-down resistor like I suggested above? As I stated, the power on state for that pin is floating so that could cause your vtx to be powered on before Betaflight can initialize the pin.

robertlacroix, 2020-01-17:
@etracer65 I ordered some when you suggested it, but they haven’t gotten here yet. Where do you see which pins are default floating? Are there some that are default low that I could use instead?

etracer65, 2020-01-17:
I don't know the state of the other pins, but in the data sheet PA10 and PB12 are noted to be excepted from pull-up/pull-down support. So any other pin might behave better.

[image]

robertlacroix, 2020-01-17:
@etracer65 thanks! Will try another pin in a bit

robertlacroix, 2020-01-17:
@etracer65 awesome, with @jflyper’s fix and on A09 it works flawlessly on the Hobbywing G2. Will check tonight if the same solution works for the SynergyF4 and HW G3. Thanks for all your guys’ help so far

robertlacroix, 2020-01-18:
HW G3 and synergy good, too!! Thanks again @jflyper @etracer65 @mikeller

mikeller, 2020-01-25:
@robertlacroix: Can we close this then?

robertlacroix, 2020-01-25:
Yes, thanks again everyone

closed 2020-01-25 by robertlacroix