betaflight/betaflight #9374: Altitude limitation

Elemmacil, 2020-01-16:
Helo,

In Europe a new law will ask our quad to limit the maximum altitude. Is it possible to manage this request with Betaflight (for example limit throttle) ?

etracer65, 2020-01-16:
Betaflight has no altitude control or limiting. You can add the altitude to your OSD as well as set an altitude alarm. Once you reach the limit the altitude element will begin blinking to indicate the warning condition.

Juzzle1, 2020-02-15:
You can do this in a rudimentary way in OpenTX if you have a flight controller with a barometer. 

Create a new logical switch that turns on when ALT > 400ft. 
Then create a special function using the logical switch as a trigger and set it to override the throttle channel to a low value.

I'm not sure if this would work with the GALT (GPS) telemetry sensor as I think it always reads altitude above sea level.

mikeller, 2020-02-15:
@Juzzle1:

> Create a new logical switch that turns on when ALT > 400ft.
Then create a special function using the logical switch as a trigger and set it to override the throttle channel to a low value.

Wow, **DO NOT DO THIS!**
I suspect the resulting oscillation between 'normal' and low throttle will result in loss of control and a crash into whatever is below the craft in seconds.

Juzzle1, 2020-02-15:
@mikeller I have done this and with air mode enabled the quad was controllable until it had gone back below the set altitude. 

I completely understand this might not be an idea that the Betaflight would want to promote so I stand down and have deleted my comment.

rFlightFPV, 2020-02-16:
If I am not mistaken the discussed altitude restriction is based on the aircrafts's AGL, so limiting altitude vs. homepoint would be pointless.

closed 2020-03-25 by stale[bot]

VHF-ACADDICT, 2022-02-24:
alti lim should be enabled by default, as we don't have the same freedom everywhere in the world, unfortunately for some of us only...