betaflight/betaflight #9377: Betaflight JoyStick HID not working on Linux

fr4nk3n5731n, 2020-01-16:
**Describe the bug**
No games (neither native nor proton) detect any inputs from the betaflight HID

**To Reproduce**
enable HID on the flight controller, connect transmitter, launch some simulator and watch nothing happen with the input channels in the game.

**Expected behavior**
input from the transmitter is picked up by the game.

**Flight controller configuration**
```
# 

# diff

# version
# Betaflight / STM32F7X2 (S7X2) 4.1.2 Dec 22 2019 / 02:21:16 (89051e256) MSP API: 1.42
# manufacturer_id: MTKS   board_name: MATEKF722SE   custom defaults: YES

# start the command batch
batch start

board_name MATEKF722SE
manufacturer_id MTKS

# name: Lemmy

# feature
feature -AIRMODE
feature MOTOR_STOP
feature ESC_SENSOR

# beacon
beacon RX_LOST

# map
map TAER1234

# serial
serial 2 2048 115200 57600 0 115200
serial 5 1024 115200 57600 0 115200

# aux
aux 0 0 0 1825 2100 0 0
aux 1 1 1 900 1100 0 0
aux 2 13 2 1300 1700 0 0
aux 3 28 1 1350 1600 0 0
aux 4 35 2 1925 2100 0 0

# master
set gyro_calib_duration = 0
###ERROR: CORRUPTED CONFIG: gyro_calib_duration = 0###
set gyro_calib_noise_limit = 0
set gyro_overflow_detect = OFF
set acc_trim_pitch = -4
set acc_calibration = 161,43,215
set mag_hardware = AUTO
set rssi_channel = 16
set rc_smoothing_auto_smoothness = 0
set fpv_mix_degrees = 30
set motor_poles = 0
###ERROR: CORRUPTED CONFIG: motor_poles = 0###
set vbat_warning_cell_voltage = 360
set small_angle = 180
set osd_vbat_pos = 71
set osd_rssi_pos = 2419
set osd_link_quality_pos = 373
set osd_flymode_pos = 359
set osd_vtx_channel_pos = 2093
set osd_ah_sbar_pos = 2254
set osd_ah_pos = 2126
set osd_current_pos = 2375
set osd_mah_drawn_pos = 2385
set osd_craft_name_pos = 2129
set osd_compass_bar_pos = 330
set osd_altitude_pos = 364
set osd_warnings_pos = 14569
set osd_avg_cell_voltage_pos = 2119
set osd_disarmed_pos = 2315
set osd_esc_tmp_pos = 2407
set osd_esc_rpm_pos = 311
set osd_esc_rpm_freq_pos = 194
set usb_hid_cdc = ON
set gyro_1_align_pitch = 1800
set gyro_1_align_yaw = 1800
set gyro_2_align_yaw = 900
set name = Lemmy

profile 2

# profile 2
set i_pitch = 70
set i_roll = 60
set p_yaw = 35
set i_yaw = 100

rateprofile 0

# rateprofile 0
set roll_rc_rate = 200
set pitch_rc_rate = 200
set yaw_rc_rate = 200
set tpa_rate = 50
set tpa_breakpoint = 1500

# end the command batch
batch end

# 
```
resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: FREE
A03: SERIAL_RX 2
A04: FREE
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: LED 2
A14: LED 1
A15: FREE
B00: MOTOR 3
B01: MOTOR 4
B02: GYRO_CS 1
B03: FREE
B04: MOTOR 1
B05: MOTOR 2
B06: FREE
B07: FREE
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: SERIAL_TX 3
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_CURR
C02: ADC_BATT
C03: FREE
C04: GYRO_EXTI
C05: FREE
C06: FREE
C07: SERIAL_RX 6
C08: PINIO 1
C09: PINIO 2
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: BEEPER
C14: FREE
C15: GYRO_CS 2
D00: FREE
D01: FREE
D02: SDCARD_CS
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3:
    CH1: MOTOR 1
    CH2: MOTOR 2
    CH3: MOTOR 3
    CH4: MOTOR 4
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: TIMUP 3
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

# 
```

Use `resource show all` to create a resource allocation list and post it here in a code block. Put ``` (three backticks) at the start and end of the output block.

**Setup / Versions**
 - MATEKF722SE (Betaflight 4.1.2)
 - RX: FrSky XM+ (LBT Firmware)
 - TX: FrSky QX7 (LBT Firmware)
 - Linux Distro: Arch Linux (5.3.13-arch1-1-vfio)

**Additional context**
Receiver input is being picked up in the receiver tab in betaflight.
The HID gets detected in Drone Race League Simulator in this specific  case but no inputs are being detected (it's the same for the native version of FPV Freerider).
The user used to play games is in the input group.
jstest-gtk detects inputs from the HID

etracer65, 2020-01-16:
It looks like your configurations corrupted.

```
# master
set gyro_calib_duration = 0
###ERROR: CORRUPTED CONFIG: gyro_calib_duration = 0###
set gyro_calib_noise_limit = 0
set gyro_overflow_detect = OFF
set acc_trim_pitch = -4
set acc_calibration = 161,43,215
set mag_hardware = AUTO
set rssi_channel = 16
set rc_smoothing_auto_smoothness = 0
set fpv_mix_degrees = 30
set motor_poles = 0
###ERROR: CORRUPTED CONFIG: motor_poles = 0###
```

I would recommend setting back to `defaults` and reconfigure manually (don't paste a `diff` or `dump`).

fr4nk3n5731n, 2020-01-16:
oh yeah, thanks for pointing that out, I didn't see that. The config is fixed now. But it looks like the corrupted config isn't the cause of my issue.

mikeller, 2020-01-17:
@fr4nk3n5731n:

> jstest-gtk detects inputs from the HID

Not sure what the problem here is, but pretty sure it is not Betaflight. If `jstest-gtk` detects the input it means it makes its way into linux, and something isn't picking it up in the apps - I personally only ever tried FPV FreeRider in linux, but whenever `jstest-gtk` worked FPV FreeRider worked as well.

fr4nk3n5731n, 2020-01-18:
I'm very sorry for wasting everyones time here. I went ahead and bought yet another simulator (Liftoff) and it picked up everything just fine.

closed 2020-01-18 by fr4nk3n5731n

jrial, 2020-01-19:
@fr4nk3n5731n I assume all these tests are with Steam on Linux? Because I have a similar issue where jstest and jstest-gtk detect the HID device just fine (albeit with only 8 out of 10 axes), but neither Steam nor Liftoff detect it at all.

fr4nk3n5731n, 2020-01-19:
@jrial yes with steam on linux, but like I said in my previous message Liftoff detects everything just fine. So I'm guessing FPV Freerider used some ancient library andDRL relies on Windows libraries that I lack with proton. Make sure your user is in the `input` group

jrial, 2020-01-20:
@fr4nk3n5731n Like I said, it gets detected just fine under jscal and jscal-gtk, so it's not a permission issue. I just want to rule out the Steam/Linux combo as the possible culprit. If you got it working with Steam on Linux, it means my problem lies elsewhere.

ps: do you see the controller only in Liftoff, or does it also show up in Steam?

fr4nk3n5731n, 2020-01-20:
the controller only shows up in Liftoff, steam can't see it.

jrial, 2020-01-20:
Finally got it working. Thanks a bundle for the help, @fr4nk3n5731n!

mkroehnert, 2020-06-23:
Hi all, this issue seems a bit old already, but I thought it might fit here as well.
Please let me know, if I should open a new issue or if this is out of scope here.

I am facing the issue, that a `/dev/input/eventX` gets created when I connect my BetaFlight controller as USB HID device, but `sdl2-jstest` does not list the device as gamepad/joystick.
However, `/dev/input/js0` gets created as well and `sdl1-jstest` picks it up (but from reading a lot, the SDL1 method and the jsX device do not seem to be used by Steam anymore).

dmesg prints the following output
```
[ 1274.379528] usb 1-4: New USB device found, idVendor=0483, idProduct=3256, bcdDevice= 2.00
[ 1274.379530] usb 1-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1274.379531] usb 1-4: Product: Betaflight STM32F405
[ 1274.379532] usb 1-4: Manufacturer: Betaflight
[ 1274.379533] usb 1-4: SerialNumber: 0x8000000
[ 1274.382074] input: Betaflight Betaflight STM32F405 as /devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.0/0003:0483:3256.0007/input/input33
[ 1274.382182] hid-generic 0003:0483:3256.0007: input,hidraw1: USB HID v1.11 Gamepad [Betaflight Betaflight STM32F405] on usb-0000:00:14.0-4/input0
[ 1274.382260] cdc_acm 1-4:1.1: ttyACM0: USB ACM device
```
Also running `evtest` on the `input0` or `input33` prints events to the console.

Running Steam on the console prints the following output, but the controller is not detected by Steam or by Liftoff.
```
Local Device Found
  type: 0483 3256
  path: sdl://0
  serial_number:  - 0
  Manufacturer:
  Product:      Betaflight STM32F405
  Release:      111
  Interface:    -1
```

To me, this looks like maybe a driver is not loaded for the device.
@jrial @fr4nk3n5731n do you maybe know which driver is associated on your machine that I could try loading?
Thanks in advance.

fr4nk3n5731n, 2020-06-24:
@mkroehnert try starting Liftoff outside of Steam instead, that "fixed" it for me. So I'm suspecting Steam or rather Steams input detection/mapping to be the culprit.

jrial, 2020-06-24:
@mkroehnert For me the issue was that I assumed the controller to show up in Steam, with the other controllers I have connected. It doesn't. It also doesn't work to navigate the menus in Liftoff, only for actual flying, IIRC. Can't really test it now as I'm at work, but I gathered some info via remote shell.

My dmesg looks similar to yours:
```
[3338484.799728] usb 3-7: new full-speed USB device number 8 using xhci_hcd
[3338484.948993] usb 3-7: New USB device found, idVendor=0483, idProduct=3256, bcdDevice= 2.00
[3338484.948998] usb 3-7: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[3338484.949002] usb 3-7: Product: Betaflight STM32F405
[3338484.949005] usb 3-7: Manufacturer: Betaflight
[3338484.949008] usb 3-7: SerialNumber: 0x8000000
[3338484.951889] input: Betaflight Betaflight STM32F405 as /devices/pci0000:00/0000:00:14.0/usb3/3-7/3-7:1.0/0003:0483:3256.000F/input/input38
[3338484.952330] hid-generic 0003:0483:3256.000F: input,hidraw8: USB HID v1.11 Gamepad [Betaflight Betaflight STM32F405] on usb-0000:00:14.0-7/input0
[3338485.738982] cdc_acm 3-7:1.1: ttyACM0: USB ACM device
[3338485.739142] usbcore: registered new interface driver cdc_acm
[3338485.739142] cdc_acm: USB Abstract Control Model driver for USB modems and ISDN adapters
```

My list of loaded usb/hid/joystick related kernel modules:
```
nighty@arcanum ~ $ lsmod | grep -i 'hid\|usb\|joy' | sort
hid                   126976  9 hid_steam,usbhid,hid_generic,hid_logitech_dj,hid_logitech_hidpp
hid_generic            16384  0
hid_logitech_dj        24576  0
hid_logitech_hidpp     40960  0
hid_steam              20480  0
joydev                 28672  0
mac_hid                16384  0
usbhid                 53248  1 hid_logitech_dj
```

mkroehnert, 2020-06-24:
@JohnieBraaf @fr4nk3n5731n thanks for the quick response and the pointers. I'll see if they help me getting this to work.

mkroehnert, 2020-08-01:
@JohnieBraaf @fr4nk3n5731n wanted to let you know, that I got it to work eventually.
Even though I had tried it before asking here, setting 'Steam Input Per-Game' to 'Forced Off' in the Steam controller options did the trick in the end.

AlienTux, 2021-07-22:
I know this is an old thread, but just wanted to say thank you: running LiftOff outside of steam worked immediately with my BetaFPV Literadio SE 2.

The option 'Steam Input-Per-Game' is no longer available, tho there is a Controller tab and forcing everything off also didn't work for me.

zigmar, 2021-08-14:
Just in case it might be useful to someone, as it took me quite a long time to figure out: after adding a user to 'input' group a restart/relogin is required (a groups are applied on login). Without it jstest-gtk detected the betaflight HID but the game didn't. I got clued in by having "evtest /dev/input/event0" not working due to permission which was solved by simple restart.

makerio90, 2022-01-14:
having this same issue with a orangerx usb rx.

i saw somewhere that if you put ```usbhid.quirks=0451:16a5:0x040``` in your /boot/cmdline.txt file, it forces hid into treating it as a game pad device.

i can see it show up in jstest-gtk and the reported  channels react how i think they should. however nothing on steam, and an online game pad tester wont work either. any help would be appreciated.