betaflight/betaflight #9379: Please add a driver for the Memsic MMC5883A compass

twbrkfd1, 2020-01-17:
The Memsic MMC5883A is an advanced version of the ubiquitous HMC5883. It is a newer state-of-the art component. It has much higher data rate capability. However, it's register set is a bit different than the HMC part; similar just uses different locations.

jflyper, 2020-01-18:
Thank you for the head ups on newer and better sensor devices. They really do seem interesting.
However, there are couples of reasons devs are not very positive on adding drivers for these devices, as they do not contribute to Betaflight's focus. The way the current unified target is built also poses some limitations on what devices we can support.

twbrkfd1, 2020-01-18:
1.admin issue: I submitted this as a feature request, but in Github it comes up as just an "issue" with no "feature request" tab...?!

jflyper, 2020-01-18:
It's okay. The nature of the ticket is clear; a feature request.

tavdog, 2022-06-04:
Stale issue¿

closed 2022-06-04 by haslinghuis