betaflight/betaflight #9380: Please add driver for the Bosch BMM150.

twbrkfd1, 2020-01-17:
The BMM150 compass is a solid component, and it has been around for some time yet it is not supported. Would be nice to have.

etracer65, 2020-01-18:
We generally don't speculatively add drivers for sensors. If we did that we would have dozens of unmaintained drivers included bloating the firmware. If you can provide examples where this sensor is used in an existing flight controller or is popular through some easy to use add-on board then we might consider it.

haslinghuis, 2024-05-12:
Outdated.

closed 2024-05-12 by haslinghuis

iottrends, 2025-09-06:
Parameter	BMM150	  QMC5883L	HMC5883L
Resolution	 0.3 µT	   2 µT	         5 µT
Range	        ±1300 µT  ±800 µT	 ±800 µT
Current	        170 µA	  2.5 mA	  0.1 mA
Temp Stability	Excellent	Good	  Fair
Package	     1.56×1.56mm 	3×3mm 	3×3mm 

BMM150	  is supported in Ardupilot, costs 0.5USD in LCSC also HMC5883L is hard to source