betaflight/betaflight #9381: Please add driver for Infineon DPS310 barometer.

twbrkfd1, 2020-01-17:
The DPS310 has been out for a couple of years and is an advanced barometris pressure sensor. It has a capacitive sensor rather than a resistive sensor like most ubiquitous barometric sensors (MS5611, BMP280, etc). The capacitive sensor is immune to light sensitivity unlike the resistive sensors. In addition, they have a higher resolution and accuracy.

MicroMontis, 2020-11-16:
Matek H743-WING Flight Controller has STM32H743VI and DPS310. Also iFlight Beast H743 FC has one. As I have read in the #10195, the DPS310 support will be from BF 4.3.

goodnerdboy, 2020-12-07:
It seems the barometer of the F722-SE board is now the DPS310 too which won’t  be recognized by betaflight.

Bugz000, 2020-12-15:
matek F405-SE uses DPS310, flashed to most recent 4.3 unstable to no avail, despite the fact it's a selectable option in the console, guess it's still not implemented fully :( eagerly awaiting this, i bought the board specifically for the far superior barometer

mikeller, 2020-12-26:
@Bugz000: DPS310 should work on the 4.3 development builds - if not then this is probably something to be raised with @MATEKSYS.

MATEKSYS, 2020-12-26:
@mikeller  @Bugz000  https://github.com/betaflight/betaflight/issues/10041

Bugz000, 2020-12-30:
> @mikeller @Bugz000 #10041

now i'm excited! thanks for tagging this!

ht099, 2021-05-19:
What other flight controllers except IFlight Beast H7 and Matek have this barometer ?

haslinghuis, 2023-12-16:
Fixed in #9957

closed 2023-12-16 by haslinghuis