betaflight/betaflight #9387: Defaults for Success-D F7 gyros is wrong

ghden, 2020-01-18:
I upgraded a quad from FC SucceX F7 V1 to the SucceX-D V2, nothing else changed and arming it was doing roll of death, after checking everything possible, I found the Gyros alignment incorrect, I verified comparing to the F7 V1 wich their defaults I tested good in 4.1.1 and 4.1.3.

After I fixed manually the alignment of both gyros it worked well.

Here are the diff for the gyros 1 and 2

gyro_1_sensor_align = CW0
Allowed values: DEFAULT, CW0, CW90, CW180, CW270, CW0FLIP, CW90FLIP, CW180FLIP, CW270FLIP, CUSTOM
Default value: CW270

I have to set it to CW0 manually, you can see the default is 270

gyro_2_sensor_align = CW90
Allowed values: DEFAULT, CW0, CW90, CW180, CW270, CW0FLIP, CW90FLIP, CW180FLIP, CW270FLIP, CUSTOM
Default value: DEFAULT

The default value DEFAULT not sure what it means but apparently for the other SucceX F7 V1 is CW90 which is correct.. 

Thanks!
Denis

RipperDrone, 2020-01-18:
confirmed, same thing here, I had to manually check each gyro's orientation in single gyro mode and found out the same. Did the correction manually. Board maintainer should have a look and change the Unfied Target defaults accordingly.

etracer65, 2020-01-18:
@Linjieqiang

etracer65, 2020-01-21:
see https://github.com/betaflight/unified-targets/pull/165

RipperDrone, 2020-01-25:
@Linjieqiang pls re-instate modified Unified Target file, otherwise your board is now deleted / unusable in terms of Unified Targets

mikeller, 2020-01-25:
@RipperDrone:

[image]

RipperDrone, 2020-01-25:
@mikeller Thank you - so board maintainer has updated meanwhile? (checking if file for TWIN_G_D is available as well...)

mikeller, 2020-01-25:
@RipperDrone: Not sure what you mean by 'updated' - this target was always working for non-defective boards.

RipperDrone, 2020-01-25:
@mikeller I realized that gyro orientations from @Linjieqiang 's UT file did NOT match physical gyro orientation, it had to be manually re-configured. Thus, I never felt it was about defective boards, just  a mismatch btw hardware and UT defatuls. I understood that @Linjieqiang as board maintainer was asked to correct this in UT defaults, but due to no reaction from his side in this thread the UT file was deleted by you. Maybe I got it wrong...

TWIN_G_D firmware is not listed in UT flash files yet, as far as I remember it was a different one than TWIN_G
[image]

mikeller, 2020-01-25:
@RipperDrone: From betaflight/unified-targets#165:

> According to iFlight, the need for a different gyro orientation on some IFF7_TWIN_G boards is caused by a hardware issue with a certain batch of boards - all correctly working IFF7_TWIN_G require the same gyro orientation.

RipperDrone, 2020-01-25:
@mikeller - Ok. So both the _G and _G_D versions go by the same firmware file? Since one has BF OSD (analog video), the other one supports DJI (no BF OSD), I felt it was sensible to have 2 different UT firmware files _G and _G_D?

mikeller, 2020-01-25:
@RipperDrone:

> Ok. So both the SucceX and SucceX_D FC versions go by the same firmware file now?

Yes.

> I felt it was sensible to have 2 different UT firmware files _G (analog) and _G_D (digital), at least reflecting OSD differences.

Most users will find it confusing to have to choose between different firmware versions - having only one removes this confusion.

RipperDrone, 2020-01-25:
Got it, thanks - and found some side-by-side reference (rev change obviously had to do w/ yaw twitching issues, though gyro re-orientation remains shady...): 
[image]

ghden, 2020-01-25:
Well, the problem I had was with the IFF7_TWIN_G_D 30x30, the error was as
you mentioned, physically the Gyros are correctly installed but the config
default wasn't, I have the original V1 (non DJI)  that is almost identical
to this DJI version only the new conector for the DJI Air unit and better
regulators and filters, the one I have is not the V2.1 out there, I bought
it as the 2.1 but actually received something older than 2.1 maybe is the
2.0....  And the hardware have some issues using a USB_C to USB_C cable, it
doesn't power the FC and even if powered with the Lipo it won't connect to
BF, I must use a USB3 to USB-C in order to have communication with the PC..

Thanks guys for your help!
Denis

closed 2020-01-25 by mikeller