betaflight/betaflight #9391: BetaFPV A01 VTX SA broken in 4.x.x

SteveCEvans, 2020-01-19:
The A01 VTX fitted to the BetaFPV Beta75X works with the default installed BF 3.5.7, but when 4.x.x versions are installed the ability to change channel is lost. Power selection of 25/200mW still works however, so the SA is configured correctly on the right port etc.

Config for 4.1.3 below.

Is this a known issue?

```
# 

# diff all

# version
# Betaflight / STM32F411 (S411) 4.1.3 Dec 26 2019 / 00:13:12 (6b13cfb9d) MSP API: 1.42
# manufacturer_id: MTKS   board_name: MATEKF411   custom defaults: YES

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name MATEKF411
manufacturer_id MTKS
mcu_id 006a00473437510c35313736
signature 

# name: X series

# feature
feature -AIRMODE
feature LED_STRIP

# beacon
beacon RX_LOST
beacon RX_SET

# serial
serial 0 2048 115200 57600 0 115200
serial 1 64 115200 57600 0 115200

# led
led 0 0,0::A:0
led 1 1,0::A:0
led 2 2,0::L:0
led 3 3,0::L:0

# mode_color
mode_color 6 0 2

# aux
aux 0 0 0 950 1050 0 0
aux 1 1 3 1450 1550 0 0
aux 2 2 3 1950 2050 0 0
aux 3 13 2 1950 2050 0 0
aux 4 28 1 1950 2050 0 0
aux 5 35 2 1450 1550 0 0
aux 6 36 1 1450 2050 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E CUSTOM  5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 2
vtxtable powervalues 0 1
vtxtable powerlabels 25 200

# master
set gyro_sync_denom = 2
set gyro_lowpass_hz = 0
set gyro_lowpass2_hz = 0
set dyn_notch_range = LOW
set dyn_notch_width_percent = 0
set dyn_lpf_gyro_min_hz = 0
set dyn_lpf_gyro_max_hz = 700
set acc_trim_pitch = 8
set acc_trim_roll = -4
set acc_calibration = 12,-31,-87
set rssi_channel = 12
set rc_smoothing_derivative_hz = 20
set rc_smoothing_derivative_type = PT1
set serialrx_provider = CRSF
set dshot_idle_value = 500
set dshot_bidir = ON
set motor_poles = 12
set vbat_warning_cell_voltage = 360
set ibat_lpf_period = 100
set ibata_scale = 533
set ibata_offset = 200
set yaw_motors_reversed = ON
set small_angle = 180
set pid_process_denom = 1
set osd_units = IMPERIAL
set osd_warn_batt_warning = OFF
set osd_warn_visual_beeper = OFF
set osd_warn_core_temp = OFF
set osd_rssi_alarm = 70
set osd_cap_alarm = 900
set osd_alt_alarm = 400
set osd_vbat_pos = 2456
set osd_rssi_pos = 2433
set osd_tim_1_pos = 97
set osd_tim_2_pos = 55
set osd_flymode_pos = 2081
set osd_throttle_pos = 280
set osd_vtx_channel_pos = 2101
set osd_current_pos = 438
set osd_mah_drawn_pos = 2424
set osd_craft_name_pos = 54
set osd_gps_speed_pos = 257
set osd_home_dir_pos = 43
set osd_home_dist_pos = 48
set osd_altitude_pos = 278
set osd_power_pos = 0
set osd_pidrate_profile_pos = 409
set osd_avg_cell_voltage_pos = 385
set osd_battery_usage_pos = 8
set osd_stat_tim_1 = ON
set osd_stat_tim_2 = OFF
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set vtx_band = 4
set vtx_channel = 2
set vtx_power = 2
set vtx_freq = 5760
set gyro_1_align_yaw = 1800
set name = X series

profile 0

# profile 0
set dyn_lpf_dterm_min_hz = 150
set dyn_lpf_dterm_max_hz = 250
set dterm_lowpass_type = BIQUAD
set vbat_pid_gain = ON
set iterm_relax_type = GYRO
set iterm_relax_cutoff = 10
set pidsum_limit = 1000
set p_pitch = 55
set d_pitch = 46
set p_roll = 50
set d_roll = 42
set p_yaw = 36
set d_min_roll = 24
set d_min_pitch = 26
set d_min_boost_gain = 30
set d_min_advance = 0

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set thr_mid = 20
set thr_expo = 60
set tpa_rate = 80
set tpa_breakpoint = 1800

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
# 
```

etracer65, 2020-01-19:
Use a proper SA 2.0 vtxtable.

https://github.com/betaflight/betaflight/wiki/VTX-tables

etracer65, 2020-01-19:
Your primary problem is that you have the bands set to "CUSTOM" and they should be "FACTORY".

When set to "CUSTOM" the control is attempted by the custom frequency. Not every VTX (particularly cheap clones) will support custom frequencies and will refuse to change channels. When the band is set to "FACTORY" then the channel is changed by sending the band number and channel number. The VTX has an internal table to convert that into a frequency.

rob-herb, 2020-01-29:
I have this exact VTX  (on a BetaFPV Pro 2, running 4.1.0) and can confirm that the VTX table config needs to be set to "FACTORY" for all bands, as etracer65 also stated.

SteveCEvans, 2020-01-29:
@rob-herb @etracer65 That fixed it. Thank you.

closed 2020-01-29 by etracer65