betaflight/betaflight #9403: Automatic ARM without flipping the switch

febeks, 2020-01-21:
I have assembled my drone and everything seemed to work fine without the propellers but later when I put propellers on and the drone was ready to fly I put it on the floor while my crossfire and taranis remote control was connected to the receiver but but everything was off meaning not armed and the moment when I put the drone on the floor the props start to spin and my finger is cut into pieces really bad. I'm just wondering if anyone had ever such a problem because this is very scary and I cannot trust it anymore. I'm sure  I had set up everything correctly (it's my fourth quad) but I cannot understand why did this happen. 

Thanks for any help.

bizmar, 2020-01-23:
Hi, I'm sorry that happened to you. Let's try to find out what went wrong. From the info you provided it's impossible to tell what went wrong. Betaflight has a multitude of safety features designed to specifically prevent this. They do work, unless specifically disabled. There's a very small possibility of you finding a bug, I would say zero, but let's be diligent and figure it out. 

Did all the motors spin or just one? (one motor indicates an ESC problem) Let's continue on the presumption that all the motors spun up because the quad actually armed. If you had your DVR running, please post that as well.

Here's what happens in most situations of accidental arming. The person has their radio on their neck strap, and as they bend down, the arm switch hits their body. If they don't have a pre-arm configured that requires two actions to arm. The quad arms.

If you don't think that was the case, and would like us to investigate. Please don't alter any settings on the radio or on the quad. Treat it as a crime scene, you want to preserve and document. 

First remove the props.
Connect it to a computer and go to the bottom CLI tab and type in `diff all` and press `ENTER`
Paste the result in a comment here, and to prevent Github as treating it as styling, please type or copy paste ``` one line above it. 

Then try to recreate what happened. If at all possible please record it with your phone. If you can successfully recreate it, please do the following and record it.

With props off and your quad connected to your computer.
Go to the motors tab and press the "Enable Motors" switch on the lower right.
Switch to the Modes tab, and record the position of the Arm mode
(recreate what happened when it armed on you but leave the quad connected to your computer)
Show the Modes and the Receiver tab.

febeks, 2020-01-23:
Thank you for reply.

I will connect the quad very soon and post the output here but I can assure you, the arm switch was off because I did not hold the switch in my hands, I had it on the floor, turned on with crossfire lite, all switches were in default position.

That's why it doesn't make a sense but I have to see what are the settings in betaflight to make sure I am not missing something.

febeks, 2020-01-23:
This is the outcome of diff all:
```
Entering CLI Mode, type 'exit' to return, or 'help'

# 
# Building AutoComplete Cache ... Done!
# 
# diff all

# version
# Betaflight / LUXF4OSD (LUX4) 3.5.4 Dec 17 2018 / 12:04:47 (66b8ea143) MSP API: 1.40

board_name LUXF4OSD
manufacturer_id 
mcu_id 001d002c3548501520313753
signature 

# reset configuration to default settings
defaults nosave

# name

# resources

# mixer

# servo

# servo mix

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature TELEMETRY

# beeper

# beacon

# map
map TAER1234

# serial
serial 5 64 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1875 2100 0 0
aux 1 1 1 1300 1700 0 0
aux 2 27 2 1300 1700 0 0
aux 3 13 3 1300 1700 0 0
aux 4 28 1 1925 2100 0 0

# adjrange

# rxrange

# vtx

# rxfail

# master
set acc_calibration = 12,-41,-19
set serialrx_provider = CRSF
set osd_vbat_pos = 2092
set osd_tim_1_pos = 2072
set osd_flymode_pos = 2049
set osd_vtx_channel_pos = 2548
set osd_mah_drawn_pos = 2085
set osd_avg_cell_voltage_pos = 2118

# profile
profile 0

# profile
profile 1

# profile
profile 2

# restore original profile selection
profile 0

# rateprofile
rateprofile 0

# rateprofile
rateprofile 1

# rateprofile
rateprofile 2

# rateprofile
rateprofile 3

# rateprofile
rateprofile 4

# rateprofile
rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save

bizmar, 2020-01-23:
That looks ok. You are on a version of Betaflight that is over a year old, there have been thousands of changes since then, so maybe there is a bug somewhere, it would be very hard to figure out. If you are on 3.5 because you like it, go to 3.5.7 which is the last version, with the most issues addressed. 

In any case, I want to figure it out. The best path forward is to replicate the issue. I know too well how it sucks when I quad does something like this, one of mine used to reset a ESC about once a month, and it would just fall out of the sky. But only about once a month, I never felt comfortable flying over trees or water.

etracer65, 2020-01-23:
If the motors spun by themselves, then how did you disarm it and get the motors to stop spinning?

febeks, 2020-01-23:
The quad flipped over and that's why it stopped. I truly don't remember how
long did it take or any other details, because I was in a shock from the
wound. But I will investigate when I have more time next week, because I
want to know what happened, as some of you too.

closed 2020-02-29 by stale[bot]