betaflight/betaflight #9404: Slow / stuttering OSD Element refresh rate Betaflieght 4.1.3

MacMen01, 2020-01-22:
**Describe the bug**
The refresh rate of the OSD Elements slow down and stuttering a lot.
The first half of a Lipo pack round every second the OSD Values will refresh. Second half of the pack the "Battery current draw" will slow down and "Battery current mAh draw will make huge steps from 550mAh to 1030mAh in one step. Current and used mAh are precisely.

**To Reproduce**
With every pack i got the same result. I flew 4 pack around me near by ang got the same Issue.

**Expected behavior**
I expect the same fine steps / refresh rate like in the first half of the pack.

**Flight controller configuration**
```
# 

# diff all

# version
# Betaflight / STM32F7X2 (S7X2) 4.1.3 Jan 16 2020 / 11:35:53 (543a5e1ba) MSP API: 1.42
# manufacturer_id: TMTR   board_name: TMOTORF7   custom defaults: YES

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name TMOTORF7
manufacturer_id TMTR
mcu_id 0045003a3238510f36353235
signature 

# name: 7 Source One CF

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature GPS
feature TELEMETRY

# map
map TAER1234

# serial
serial 1 1 115200 57600 0 115200
serial 3 2 115200 57600 0 115200
serial 4 64 115200 57600 0 115200

# aux
aux 0 0 0 1800 2100 0 0
aux 1 1 2 1350 1650 0 0
aux 2 46 2 1800 2100 1 0
aux 3 46 0 1800 2100 1 0
aux 4 13 1 1800 2100 0 0
aux 5 26 0 1800 2100 1 0
aux 6 26 1 1800 2100 1 0
aux 7 35 1 1400 1600 0 0

# rxfail
rxfail 3 h
rxfail 6 s 1500
rxfail 7 s 1000

# master
set gyro_sync_denom = 2
set gyro_lowpass2_hz = 350
set dyn_notch_range = LOW
set dyn_notch_width_percent = 0
set dyn_notch_q = 200
set dyn_notch_min_hz = 90
set dyn_lpf_gyro_min_hz = 280
set dyn_lpf_gyro_max_hz = 700
set acc_calibration = 3,10,291
set baro_hardware = NONE
set min_check = 1020
set max_check = 1980
set rssi_channel = 8
set serialrx_provider = CRSF
set min_throttle = 1020
set max_throttle = 1980
set dshot_idle_value = 300
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set failsafe_procedure = GPS-RESCUE
set bat_capacity = 1200
set vbat_scale = 111
set ibata_scale = 146
set ibata_offset = -5100
set small_angle = 180
set gps_provider = UBLOX
set gps_sbas_mode = EGNOS
set gps_auto_baud = ON
set gps_rescue_angle = 35
set gps_rescue_initial_alt = 70
set gps_rescue_descent_dist = 30
set gps_rescue_ground_speed = 1000
set gps_rescue_min_sats = 5
set pid_process_denom = 1
set osd_warn_rssi = ON
set osd_cap_alarm = 1200
set osd_vbat_pos = 2539
set osd_rssi_pos = 2537
set osd_link_quality_pos = 2535
set osd_current_pos = 2541
set osd_mah_drawn_pos = 2545
set osd_gps_lon_pos = 2112
set osd_gps_lat_pos = 2080
set osd_gps_sats_pos = 2048
set osd_home_dir_pos = 2510
set osd_home_dist_pos = 2144
set osd_battery_usage_pos = 2057
set osd_disarmed_pos = 2220
set osd_stat_max_dist = ON
set osd_stat_endbatt = ON
set osd_stat_max_alt = ON
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set osd_stat_total_time = ON
set osd_stat_total_dist = ON
set name = 7 Source One CF

profile 0

# profile 0
set dyn_lpf_dterm_min_hz = 98
set dyn_lpf_dterm_max_hz = 238
set dterm_lowpass2_hz = 210
set iterm_relax = RPY
set iterm_relax_type = GYRO
set p_pitch = 76
set i_pitch = 99
set d_pitch = 54
set f_pitch = 105
set p_roll = 73
set i_roll = 94
set d_roll = 51
set f_roll = 99
set p_yaw = 45
set i_yaw = 99
set f_yaw = 99
set d_min_roll = 29
set d_min_pitch = 32

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
# 
```
Use `resource show all` to create a resource allocation list and post it here in a code block. Put ``` (three backticks) at the start and end of the output block.
```
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: SERIAL_TX 4
A01: SERIAL_RX 4
A02: SERIAL_TX 2
A03: SERIAL_RX 2
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: FREE
A14: FREE
A15: FREE
B00: MOTOR 3
B01: MOTOR 4
B02: LED 1
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: FREE
B07: FREE
B08: FREE
B09: CAMERA_CONTROL
B10: I2C_SCL 2
B11: I2C_SDA 2
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_BATT
C02: ADC_CURR
C03: FREE
C04: GYRO_EXTI
C05: FREE
C06: MOTOR 1
C07: MOTOR 2
C08: FLASH_CS
C09: FREE
C10: FREE
C11: FREE
C12: SERIAL_TX 5
C13: BEEPER
C14: PINIO 1
C15: PREINIT 2
D00: FREE
D01: FREE
D02: SERIAL_RX 5
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1: DSHOT_BITBANG 3
    CH2: DSHOT_BITBANG 2
TIM9: FREE
TIM10: FREE
TIM11:
    CH1: CAMERA_CONTROL
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 3
DMA2 Stream 3: DSHOT_BITBANG 2
DMA2 Stream 4: ADC
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

**Setup / Versions**
 - Flight controller [what type is it, where was it bought from];
FC = T-Motof F7 HD bought from German Dealer www.N-Factory.de

 - Other components [RX, VTX, brand / model for all of them, firmware version where applicable];
 - how are the different components wired up.
ESC = T-Motor F55A PRO II with BLHeli 32.7 (Current Read yout ofer analog pin to FC)
VTX = DJI Air Unit V1.0.4.0 UART 2 Configuration / MSP enabled
Rx = TBS Crossfire Diversity V3.23 UART 5 seriel Reseiver enabled
GPS = Matek M8Q-5883 UART 4 Sensor Input set to GPS

**Additional context**
When i only fly the first half of the pack the numbers for used Current mAh are equal to the recarged numbers for the pack.

etracer65, 2020-01-22:
This looks very questionable:
```
set ibata_offset = -5100
```
Why do you have such a large negative offset?

MacMen01, 2020-01-22:
I have measure voltage and Current externaly with my good and trusted meters. I have set the Voltage scale to 111 so that the Voltage matches the external measurment.

For the Current the same. I have spin the Motors and measure external Current for 1A / 3A / 5A / 8A. Than i have compared the Betaflight Values and calculated that Value. Betaflight is showing me around 5,1A to much compared to the external measurment. With this offset off -5100 the Values will macht not perfectly but close enough.

etracer65, 2020-01-22:
> Betaflight is showing me around 5,1A to much compared to the external measurment. With this offset off -5100 the Values will macht not perfectly but close enough.

Well that's a problem. You shouldn't need a large (or any) offset. The manufacturer specifies a current scale of 152. Normal analog current sense resistor setups like this don't require an offset.

You likely have electrical noise getting into the current sense wire. Try adding a 0.1uF capacitor from the current sense wire to ground.

MacMen01, 2020-01-22:
Ok i will try it with tha capacitor. I have the same issue with a diffrent Quad also with a T-Motor F4 HD FC + F55A PRO II 4in1 ESC. Here i have offset of -7400.

downset, 2020-01-30:
I have the same Problem 
after some time some osd information are getting stuck
today for example the used mAh jumped from 411mAh to 1220mAh also actual current was stuck while gps cords and Voltage updates came normally

do you also use the DJI HD System? If so, maybe its a bug in the custom osd there - that was my first thought

MacMen01, 2020-02-01:
Yes i also use the DJI HD System (Air Unit + Goggles). And yes i can confirm that not all values will stuck. Voltage, GPS coordinates and RSSI will refresh normaly. Only my used mAh will work fine at the beginning and than it stuck and make a huge jump like yours.

Now i have soldered a capacitor between Current sens and GND. But the weather doesn´t allow a test.

downset, 2020-02-01:
Do you also have disabled the max7456 to get constant pid loops as i suggest [here](https://forum.dji.com/forum.php?mod=redirect&goto=findpost&ptid=203712&pid=2030924) ?
maybe this could be a problem.
But I think we have to address this problem to DJI
I opened a new thread [there](https://forum.dji.com/forum.php?mod=viewthread&tid=208229)

MacMen01, 2020-02-01:
No i have not disabled it. My PID loops are consisent 8k. I think the problem from the Tread you linked is that the Betaflight Model for specific Flight Controllers are wrong. Normaly that must be disabled by default in the *.hex files for DJI HD Flight Controllers.

Anyway my looptimes are fine and consisent.

downset, 2020-02-01:
Okay than that could not be the Point
anyway I can`t believe that the Capacitor can fix it.
The measurements are correct the OSD just get no update.
After a "jump" the showen flight time and Voltage makes sense to the just updated used mAh, but it takes minutes to the next OSD update

etracer65, 2020-02-01:
Nowhere in the original problem report was it mentioned that this was only affecting the DJI OSD. The capacitor can help cases where the current readings are inaccurate and jump around because of electrical noise. If it's only affecting DJI OSD then it's probably best to check with DJI support as they poll the data from the flight controller over MSP messaging.

MacMen01, 2020-02-01:
The capacitor helps me to reduce the Current - Offset. I think it have nothing to do with the OSD stuttering.

So i think the T-Motor F7 HD Flight Controller have the same Problem. In the CLI i entered the command.

"get max7456_spi_bus" ->
max7456_spi_bus = 2

Now i have set the default 2 to 0. I will try this and will keep the result updated here.

mikeller, 2020-02-01:
@MacMen01:

> Now i have set the default 2 to 0. I will try this and will keep the result updated here.

I am surprised that this is still needed - #9360 in 4.1.3 should have fixed this. Can you try again to see if you get constant loop times even without these changes?

downset, 2020-02-02:
> I am surprised that this is still needed - #9360 in 4.1.3 should have fixed this. Can you try again to see if you get constant loop times even without these changes?

you are right, I resettet the max7456 to it`s default spi bus and still constant PID Loops.
Thanks for this clarification.

But I think it has nothing to do with the actual Problem. It was just my first thought cause it was a change i did from defaults.

But as macMen hasn't done this change and i did, and we both have the same Problem, now i think it has nothing to do with the Stucking Elements.

mikeller, 2020-02-02:
@MacMen01: Ok. I just wanted to verify that the fix is indeed working as it should. As @etracer65 already stated, it sounds very much like the OSD problems are on the DJI side. You should raise an issue with them.

MacMen01, 2020-02-02:
I have setup my Radio Telemetry Sensors to my Display. So at the next test I can compare the OSD information when it starts to stutter and the original Telemetry Data on the Radio. Than we have more Information.
What have i changed at the moment from start of this issue to now.
1) Add a capacicator 100nF between Current sens and GND to the FC. And i have set the Current Offset in Betaflight from -5100 back to 0.

2) max7456_spi_bus = 2 (is at default)

3) I have changed the Gyro update frequency and PID loop frequency to 8k/8k

closed 2020-03-11 by stale[bot]

btemtd, 2020-03-24:
@MacMen01 hey macmen you have same setup as me and same problem can you tell me if anyting fixed the problem?

MacMen01, 2020-03-25:
Yes no recording on the Air Unit SD (only rec. on Goggles) and the OSD is working fine. That is no fix but a workaround until a fix is there

btemtd, 2020-03-25:
hmm what firmware are you using Im pretty sure i tried this but mine still happened? WHat else have you done since the issue raised

MacMen01, 2020-03-25:
FC = Betaflight V4.1.3
DJI Air Unit = v01.00.04.00

btemtd, 2020-03-25:
I just went into Goggles and recording and set recording to Goggles only.
I reflash 4.1.3 

EVerything else stock... Same problem

When armed (motors spinning)

Current Drawn = Frozen
Mah Used = Frozen 

Anything else? Like what exactly have you got different  - ANy other settings you change , calibration, capacitor

MacMen01, 2020-03-25:
You can check / compare my Settings above with your own. Nothing changed from Stock Settings. Maybe your FC is broken. My first T-Motor F7 HD have a broblem on the Ait Unit Connector. The pin from the Current Sens Wire was not soldered correctly.

btemtd, 2020-03-25:
This is unlikely I have 2 Identical boards which do the same thing.

Can you when you have time go into the betaflight and select backup and then send me this betaflight backup file TO my email ah94gtr@hotmail.com

btemtd, 2020-03-25:
you added a capacitor from Current sensor and GND to the FC?? WHat is current sensor, which pad is this?
 the only thing i did was add a capacitor to the battery terminals

btemtd, 2020-03-25:
All SOLVED.. In ports under telemetry Remove default Smartport change to disabled all good now. EVerything works perfectly I can record , i can Not record  etc everything works no need for workaround
--

MacMen01, 2020-03-25:
No this capacitor has nothing to do with the stuttering issue. At this moment i have removed this capacitor. The Battery Capacitor is only requiered for 6S Setups for the F55A PRO II.

btemtd, 2020-03-25:
Well I have a 6s and no capacitor, everything is fine. I may install it later just for backup though. btw I can record on both unit and goggles no issue

MacMen01, 2020-03-25:
> you added a capacitor from Current sensor and GND to the FC?? WHat is current sensor, which pad is this?
> the only thing i did was add a capacitor to the battery terminals

And now you write you have no capacitor an the Battery Pins......strange. Anyway the stuttering error is not on every flight when you record on the Air Unit and the Goggles. BUT the issue is there and can be workaround with no Air Unit recording at this moment.

btemtd, 2020-03-25:
sorry I meant I tried to do the capacitor and i removed it this was days ago it made no difference at tha point

mikeller, 2020-03-25:
@MacMen01: Can you check if your problem isn't the same as the one that @btemtd presented the solution to in https://github.com/betaflight/betaflight/issues/9404#issuecomment-603789598: In configurator on the Ports tab, check that for the port connected to the air unit's UART TX / RX wires, only MSP is enabled and nothing else (like telemetry for instance)?

MacMen01, 2021-09-22:
> @MacMen01: Can you check if your problem isn't the same as the one that @btemtd presented the solution to in [#9404 (comment)](https://github.com/betaflight/betaflight/issues/9404#issuecomment-603789598): In configurator on the Ports tab, check that for the port connected to the air unit's UART TX / RX wires, only MSP is enabled and nothing else (like telemetry for instance)?

No I can not confirm that! I have the right Settings like in the other Topic but still have the same Problem.

Also testet with new DJI Firmware V1.00.0606 and Betaflight 4.2.9

Still the Same Problem! Only working solution is to not have a SD Card in the Air Unit! So i think the Caddx Vista is here the way to go for new Builds...