betaflight/betaflight #9405: throttle act like yaw and speed up uncontroble

eurodriguez89, 2020-01-22:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:

**Expected behavior**
A clear and concise description of what you expected to happen.

**Flight controller configuration**
Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/)

Use `resource show all` to create a resource allocation list and post it here in a code block. Put ``` (three backticks) at the start and end of the output block.

**Setup / Versions**
 - Flight controller [what type is it, where was it bought from];
 - Other components [RX, VTX, brand / model for all of them, firmware version where applicable];
 - how are the different components wired up.

**Additional context**
Add any other context about the problem here.

eurodriguez89, 2020-01-22:
succex-e f4
throttle isnt working properly. i need help

IvoFPV, 2020-01-22:
@eurodriguez89 Without props on? If so, that is normal, see this video...
https://www.youtube.com/watch?v=xBaefeQtA1c

closed 2020-01-23 by no-response[bot]