betaflight/betaflight #9409: Dshot 600 motors keep beeping when unarmed ?

ochbob, 2020-01-23:
Diatone Taycan
BF: 4.1.x
FC: MAMBA F405MINI MK3 (FURYF4OSD)
PDB/ESC: MAMBA F35 /35A 6S ESC Dshot 600 (Seems to be BLHeli_S ESCs )

Hello everyone,

I know this is not really the right place, but I'm a rookie in the hobby, and I can't figure out what's my issue... and it drives me crazy.

I received recently my Diatone Taycan, and after power on the drone with Lipo (and setup basic BF mods, etc...):
https://youtu.be/oKsIQ-BCAbI

My motors keep beeping... 
When I armed the drone, beeps stops.

After googling all night, I can't find anyone else had the same issue...

Maybe it's  normal, if yes, how can I disable this beeping when unarmed ?
If not, what's the beeping meaning ?

Otherwise, I know i have an bad contact too with the wire between the FC and the PDB...
I need to checkout the wires connector and FC connector...

Not a good experience to begin :(

Anyway, thank you very much for any helps about the beeping.

etracer65, 2020-01-23:
Issues are for reporting problems with the firmware, bug reports, and feature requests.

For configuration help try other resources like:

Facebook group: https://www.facebook.com/groups/betaflightgroup/

Betaflight Slack: https://slack.betaflight.com/

closed 2020-01-23 by etracer65

StanMan11, 2020-04-08:
Hi both , i've the same problem on the BNF Diatone Taycan , it keep beeping until unarmed !! may i know what is the solution ? Thanks in advance