betaflight/betaflight #9412: VTX device not ready BF 4.1.2 / IRC Tramp issue ?

ochbob, 2020-01-24:
Hello,

I can't make Mamba TX400 (VTX comes with the Taycan PNP) work with my BF,
 The factory settings is UART3 for IRC Tramp (TX3 on the FC)
The Diatone wiring is good, I check multiple times.

But when I go to the VTX tab in BF, my device is not ready:

[image]

I can change VTX channel / band through VTX buttons, but I can't have any signal with the auto search from my video receivers...

It seems something is wrong with my BF config or with the VTX...

But LED on the VTX seems OK:

[image]

Config dump:

`# 

# dump

# version
# Betaflight / MAMBAF405US (MBF4) 4.1.2 Dec 28 2019 / 15:49:00 (89051e256) MSP API: 1.42

# start the command batch
batch start

board_name MAMBAF405US
manufacturer_id 

# name: TAYCAN

# resources
resource BEEPER 1 C13
resource MOTOR 1 A09
resource MOTOR 2 A08
resource MOTOR 3 C09
resource MOTOR 4 C08
resource MOTOR 5 NONE
resource MOTOR 6 NONE
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 B09
resource PWM 1 NONE
resource PWM 2 NONE
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource LED_STRIP 1 B03
resource SERIAL_TX 1 B06
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 A00
resource SERIAL_TX 5 C12
resource SERIAL_TX 6 C06
resource SERIAL_TX 7 NONE
[... 942 lines cut ...]
set rates_type = BETAFLIGHT
set roll_rc_rate = 70
set pitch_rc_rate = 70
set yaw_rc_rate = 75
set roll_expo = 0
set pitch_expo = 0
set yaw_expo = 0
set roll_srate = 60
set pitch_srate = 55
set yaw_srate = 60
set tpa_rate = 65
set tpa_breakpoint = 1250
set tpa_mode = D
set throttle_limit_type = OFF
set throttle_limit_percent = 100
set roll_rate_limit = 1998
set pitch_rate_limit = 1998
set yaw_rate_limit = 1998

# end the command batch
batch end

# `

Thank you

McGiverGim, 2020-01-24:
Please paste a diff and not a dump. Is far easier for the people to review.

ochbob, 2020-01-24:
diff: https://pastebin.com/2Z8rF99C

Thank you

etracer65, 2020-01-24:
The device being reported as "not ready" means that Betaflight is unable to communicate with the VTX. So one or more of these are the problem:

- VTX doesn't support Tramp
- VTX doesn't support Tramp properly (broken implementation)
- Not connected properly (wiring problem)
- Correct UART not selected and set to Tramp
- VTX is "locked" (see the manual)

ochbob, 2020-01-24:
But I should be able to see video signal if I put my video receiver on the signal which is indicate by the VTX (for example R4, like my photo), right ?

So it could be a VTX issue too ?

Anyway:

- VTX support tramp (watch lot of video about the Taycan PNP)
https://youtu.be/QU8yuBDySTY?t=170 / https://youtu.be/QU8yuBDySTY?t=343
- Maybe, but works for the others ?
- I double checked, seems OK to me and I have the power on the VTX (so 5v/ground are well wired) Maybe the video and audio are not, but visually it's ok, liem5v and ground.
- UART is the number 3, and the audio wire is on the TX3, so I should be good
-VTX is unlocked, I have all the frequency / channel when I switch from the VTX buttons (conversely when it is locked, I can't go to the E frequency)

So my guess is that the VTX is broken, if my BF settings seems ok to you...

etracer65, 2020-01-24:
Tramp control does not use the "audio" wire. Tramp compatible VTX's will have a dedicated telemetry connection. Usually labeled "tramp", "telem", "tlm" or maybe "rx". This is what you'd connect to TX3 on the flight controller.

ochbob, 2020-01-24:
The audio wire from VTX goes to TX3 on the FC.

So you sugest to use smart audio ?

Diagram:

https://cdn.shopify.com/s/files/1/0027/2708/4144/products/diatone-mxc-taycan-duct-3-inch-cinewhoop-preorder-designer-series-new-official_919_1024x1024@2x.jpg?v=1577536094

mikeller, 2020-01-24:
@ochbob: Looking at https://cdn.shopify.com/s/files/1/0027/2708/4144/products/diatone-mxc-taycan-duct-3-inch-cinewhoop-preorder-designer-series-new-official_543_1024x1024@2x.jpg?v=1577536094, it does not state anywhere that this VTX supports any form of remote control. The instructions make it look like the only way to change frequencies / power levels are the buttons on the device.

ochbob, 2020-01-24:
Hum, manual is specific about frequency /band power control: « support BF OSD remonte control »

mikeller, 2020-01-24:
@ochbob: All that this states is that the Betaflight OSD CMS can be used with this VTX (which is stating the obvious), and giving instructions on how to enter the CMS. It does not actually state anything about the VTX being controllable through Betaflight.

etracer65, 2020-01-24:
Also nowhere in the documentation link for the VTX does is state that it supports Tramp vtx control. And like I said before, the "audio" wire would never connect to TX3. The audio connection on the VTX is used to connect a microphone.

ochbob, 2020-01-25:
So, the VTX is broken, because even without VTX remote control, I should see video signal if I put my video receiver on the same channel that the VTX is configure, right ?

mikeller, 2020-01-25:
@ochbob: Yes - contact the manufacturer.

closed 2020-01-25 by mikeller

ash2482, 2021-01-19:
I kinda have the same issue but I couldn't even change the channels in my vtx . It is locked in some kind of way and I cant unlock it . I tried to unlock it using the manufacturers instructions but it doesnt seem to work. I can see video thought but it works sometime and doesn't on other times. I am total beginner to fpv  so sommeone please help me
[image]
[image]
[image]