betaflight/betaflight #9413: KAKUTEF7 V2 dshot bidir makes gps unusable

TregliaRaphael, 2020-01-24:
**Describe the bug**
Here is a 3 minute video: [issue video](https://www.youtube.com/watch?v=u5IrQ9Xicgo&t=45s)
I can't use betaflight firmware on configurator. It's stuck on "waiting for data"
[image]

**To Reproduce**
Just activate dshot bidir with gps, lets check the video

**Expected behavior**
Gps should works

**Flight controller configuration**
It happen with default kakutef7v2 config and also with my custom cli

**Setup / Versions**
 - Flight controller : KAKUTEF7 V2
 - R9mm, tbs unify pro32, bn 880q gps, esc flycolor 60A
 - R9MM fport one wire on T6, tbs unify smart audio on UART1 and bn 880q on UART4

**Additional context**
video explain it very well and she is very short

TregliaRaphael, 2020-01-25:
After trying some other more basics test, i realized that it was a bit different, i'll explain:
The probleme is not between dshot and gps only finally.
To trigger the bug, with default kakuteF7V2 firmware, i juste need to activate GPS on any serial port.
Then on configuration, activate gps with default values, any dshot bidir.
Until there it works.
After, just add telemetry + FPort protocol frsky and here it trigger the bug.

mikeller, 2020-01-25:
@TregliaRaphael:

> `set use_unsynced_pwm = OFF`

This should not do anything, but just to exclude it can you test again with this set to `on`?

> feature TELEMETRY

Since you have not configured any protocol that can actually receive telemetry, I would switch this off, as it does not do anything.

TregliaRaphael, 2020-01-25:
The CLI that i give is just a cli to reproduce the bug as fast as possible.
I'm using telemetry on my other cli and i tested gps alone on all other uart and it is not working but compass work, i'm using bn 880q

mikeller, 2020-01-25:
@TregliaRaphael: Sorry, no interest in wasting time trying to help you if you do not want to provide the information that you are asked to provide.

closed 2020-01-25 by mikeller

TregliaRaphael, 2020-01-25:
it was just to declare a bug easy to reproduce and not to declare my own. i can deal without gps but it was just to try to resolv it.
here is my CLI used, sry for th emiss communication, it's mine.
```
# 

# diff all

# version
# Betaflight / KAKUTEF7V2 (KT76) 4.1.3 Jan 16 2020 / 11:23:12 (543a5e1ba) MSP API: 1.42

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name KAKUTEF7V2
mcu_id 003900433236510d31393330
signature 

# name: Raph

# resources
resource MOTOR 1 B01
resource MOTOR 2 E11
resource MOTOR 3 B00
resource MOTOR 4 E09

# feature
feature TELEMETRY

# beacon
beacon RX_LOST

# map
map RETA1234

# serial
serial 0 2048 115200 57600 0 115200

# aux
aux 0 0 1 1300 2100 1 0
aux 1 2 1 1800 2100 0 0
aux 2 27 3 1800 2100 0 0
aux 3 35 2 1300 1700 0 0
aux 4 36 0 1800 2100 0 0
aux 5 49 2 1800 2100 0 0

# rxrange
rxrange 0 988 2011
rxrange 1 987 2011
rxrange 2 987 2011
rxrange 3 988 2008

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745    0
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 UNKNOWN  U FACTORY    0    0    0    0    0    0    0    0
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860    0
vtxtable band 5 RACEBAND R FACTORY    0    0 5732 5769 5806 5843    0    0
vtxtable band 6 IMD6     I CUSTOM  5732 5765 5828 5840 5866 5740    0    0
vtxtable powerlevels 4
vtxtable powervalues 14 20 26 30
vtxtable powerlabels 25 100 400 1W

# master
set gyro_lowpass2_hz = 350
set dyn_notch_range = LOW
set dyn_notch_width_percent = 0
set dyn_notch_q = 200
set dyn_notch_min_hz = 90
set dyn_lpf_gyro_min_hz = 280
set dyn_lpf_gyro_max_hz = 700
set acc_calibration = 87,115,-179
set align_mag = CW180FLIP
set mag_align_pitch = 1800
set mag_align_yaw = 1800
set mag_hardware = NONE
set min_check = 1005
set max_check = 2000
set rssi_scale = 200
set rc_smoothing_input_hz = 40
set rc_smoothing_derivative_hz = 20
set rc_smoothing_input_type = PT1
set rc_smoothing_derivative_type = PT1
set fpv_mix_degrees = 35
set serialrx_provider = FPORT
set serialrx_inverted = ON
set serialrx_halfduplex = ON
set blackbox_device = SERIAL
set dshot_bidir = ON
set use_unsynced_pwm = OFF
set motor_pwm_protocol = DSHOT600
set failsafe_off_delay = 30
set failsafe_throttle = 1100
set failsafe_switch_mode = STAGE2
set failsafe_procedure = AUTO-LAND
set bat_capacity = 1550
set ibata_scale = 135
set pid_process_denom = 1
set tlm_inverted = ON
set tlm_halfduplex = OFF
set osd_warn_rc_smoothing = OFF
set osd_warn_no_gps_rescue = OFF
set osd_warn_gps_rescue_disabled = OFF
set osd_warn_rssi = ON
set osd_cap_alarm = 0
set osd_alt_alarm = 150
set osd_tim1 = 2561
set osd_vbat_pos = 2103
set osd_rssi_pos = 2281
set osd_link_quality_pos = 258
set osd_rssi_dbm_pos = 192
set osd_tim_1_pos = 2465
set osd_flymode_pos = 2510
set osd_g_force_pos = 2113
set osd_throttle_pos = 2145
set osd_vtx_channel_pos = 2497
set osd_crosshairs_pos = 2318
set osd_ah_pos = 142
set osd_current_pos = 119
set osd_mah_drawn_pos = 2136
set osd_gps_speed_pos = 298
set osd_home_dist_pos = 89
set osd_flight_dist_pos = 88
set osd_compass_bar_pos = 2091
set osd_altitude_pos = 2081
set osd_power_pos = 121
set osd_warnings_pos = 2410
set osd_avg_cell_voltage_pos = 87
set osd_battery_usage_pos = 393
set osd_disarmed_pos = 2187
set osd_esc_tmp_pos = 471
set osd_flip_arrow_pos = 224
set osd_core_temp_pos = 2519
set osd_stat_tim_1 = ON
set osd_stat_tim_2 = OFF
set osd_stat_max_spd = OFF
set osd_stat_min_batt = OFF
set osd_stat_endbatt = ON
set osd_stat_max_alt = ON
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set osd_stat_max_g_force = ON
set vtx_band = 1
set vtx_channel = 1
set vtx_power = 1
set vtx_freq = 5865
set vcd_video_system = PAL
set name = Raph

profile 0

# profile 0
set dyn_lpf_dterm_min_hz = 98
set dyn_lpf_dterm_max_hz = 238
set dterm_lowpass2_hz = 210
set anti_gravity_gain = 7500
set acro_trainer_angle_limit = 25

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set roll_rc_rate = 129
set pitch_rc_rate = 126
set yaw_rc_rate = 108
set roll_expo = 21
set pitch_expo = 20
set yaw_expo = 7
set roll_srate = 75
set pitch_srate = 74
set yaw_srate = 67

rateprofile 1

# rateprofile 1
set roll_rc_rate = 175
set pitch_rc_rate = 175
set yaw_rc_rate = 150
set roll_expo = 22
set pitch_expo = 22
set yaw_expo = 10
set yaw_srate = 67

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
# 
```