betaflight/betaflight #9415: Dyn_notch not following settings from configurator

HrrrPippi, 2020-01-25:
Hi
I have setup the dyn notch through the configurator with dyn_notch_min_hz = 90 and dyn_notch_range = low

I used debug_mode = fft_freq to track the dyn notch, and the center goes below 90Hz, and the header file says dyn_notch_range is set to high. I dont know if this is  an logging error, or dyn notch issue

I tried setting the min_hz = 120 and have seen it track to 105Hz, again lower than my preset.

set dyn_notch_range = LOW
set dyn_notch_width_percent = 0
set dyn_notch_q = 240
set dyn_notch_min_hz = 90
set debug_mode = fft_freq

I'm using mamba f7 with blheli_32 esc with dshot telemertry and rpm notch
configurator 10.6.0
```
board_name MAMBAF722

# name: Skinny-Pip

# feature
feature -SOFTSERIAL
feature -TELEMETRY

# beacon
beacon RX_LOST
beacon RX_SET

# serial
serial 0 0 115200 57600 0 115200
serial 2 8192 115200 57600 0 115200
serial 3 64 19200 57600 0 115200

# aux
aux 0 0 0 1800 1875 0 0
aux 1 1 1 1800 1900 0 0
aux 2 13 2 1425 1525 0 0
aux 3 35 1 1450 1550 0 0

# rxrange
rxrange 0 1156 1843
rxrange 1 1156 1843
rxrange 2 1156 1843
rxrange 3 1156 1843

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E CUSTOM  5705 5685 5665    0 5885 5905    0    0
vtxtable band 4 FATSHARK F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 IMD6     I CUSTOM  5732 5765 5828 5840 5866 5740    0    0
vtxtable powerlevels 5
vtxtable powervalues 25 100 200 400 600
vtxtable powerlabels 25 100 100 200 400

# master
set gyro_sync_denom = 2
set gyro_lowpass2_hz = 500
set dyn_notch_range = LOW
set dyn_notch_width_percent = 0
set dyn_notch_q = 240
set dyn_notch_min_hz = 110
set dyn_lpf_gyro_min_hz = 400
set dyn_lpf_gyro_max_hz = 1000
set acc_calibration = -82,79,-381
set min_check = 1005
set max_check = 1995
set serialrx_provider = FPORT
set blackbox_p_ratio = 128
set dshot_idle_value = 200
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set yaw_motors_reversed = ON
set small_angle = 180
set pid_process_denom = 1
set osd_warn_batt_not_full = OFF
set osd_rssi_alarm = 50
set osd_cap_alarm = 10000
set osd_tim1 = 0
set osd_tim2 = 17
set osd_vbat_pos = 8609
set osd_rssi_pos = 10697
set osd_tim_2_pos = 406
set osd_g_force_pos = 88
set osd_throttle_pos = 8225
set osd_current_pos = 8656
set osd_mah_drawn_pos = 14807
set osd_motor_diag_pos = 8248
set osd_craft_name_pos = 10282
set osd_altitude_pos = 366
set osd_debug_pos = 8545
set osd_warnings_pos = 14666
set osd_avg_cell_voltage_pos = 14785
set osd_core_temp_pos = 34
set osd_stick_overlay_left_pos = 4353
set osd_stick_overlay_right_pos = 4374
set osd_stat_tim_2 = OFF
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set osd_stat_battery = ON
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set osd_stat_max_fft = ON
set osd_stat_total_time = ON
set debug_mode = FFT_FREQ
set vtx_band = 5
set vtx_channel = 2
set vtx_power = 1
set vtx_low_power_disarm = ON
set vtx_freq = 5695
set name = Skinny-Pip

profile 0

# profile 0
set dyn_lpf_dterm_min_hz = 140
set dyn_lpf_dterm_max_hz = 340
set dterm_lowpass2_hz = 300
set pid_at_min_throttle = OFF
set anti_gravity_gain = 2000
set feedforward_transition = 100
set iterm_relax_type = GYRO
set iterm_relax_cutoff = 11
set pidsum_limit = 1000
set pidsum_limit_yaw = 1000
set p_pitch = 60
set d_pitch = 40
set f_pitch = 0
set p_roll = 60
set d_roll = 40
set f_roll = 0
set p_yaw = 55
set d_min_roll = 0
set d_min_pitch = 0
set idle_min_rpm = 18

rateprofile 1

# rateprofile 1
set roll_rc_rate = 103
set pitch_rc_rate = 103
set yaw_rc_rate = 103
set roll_srate = 66
set pitch_srate = 66
set yaw_srate = 66
```

ctzsnooze, 2020-01-26:
There is a blackbox explorer issue where `HIGH` is reported as `LOW` and vice versa.  That has been fixed in nightly builds.  The CLI value is the correct one.
There is a check in the code to not allow the FFT frequency to go below the minimum configured.
However the FFT frequency itself is smoothed with a biquad filter which can 'ring' low, briefly.
If you could share the log I could see perhaps if this is what's happening.

HrrrPippi, 2020-01-28:
Sorry for the late reply.. Life happened.
It sounds very probable that this is a matter of the smoothing making the dyn_notch go lower.

I will attach a log, with dyn_notch_min_hz = 120
At 2,611 sek there is an example of the dyn fft_freq showing the dyn_notch to go below the presset for a brief second.

[btfl_002.zip](https://github.com/betaflight/betaflight/files/4124365/btfl_002.zip)

closed 2020-01-31 by HrrrPippi