betaflight/betaflight #9419: Battery profile future request

JuStAdsor, 2020-01-27:
Hi!

I am recently experimenting with liion on a 6'' quad, but I have the following issue:
I have to manually adjust the warning voltage eachi time I am switching in between so the liion will not continuously beep on lipo settings.

Looking around I have seen that inav has a great future that lets you switch by stick, cli, or lua up the 3 battery profiles.

The profiles look like the one bellow:
battery_profile 1
set vbat_scale = 1100
set vbat_max_cell_voltage = 430
set vbat_warning_cell_voltage = 340
set vbat_min_cell_voltage = 330

Also the 3 profiles can be automatically rotated depending on input voltage for instance.

Would it be possible to add something like this to betaflight also.

Thank you!

Juzzle1, 2020-02-03:
My case is different, however would also make use of battery profiles:

I am currently using profiles with auto_profile_cell_count and motor_output_limit as I have a huge collection of both 4s and 6s packs that I use interchangeably on my quads. 
The 4s packs are 1300mah, and the 6s packs are 1000mah, so I can't make use of OSD capacity warnings or remaming time estimate depending on which battery I'm using.

Being able to set battery profiles by aux channel or auto_profile_cell_count would be a great addition.

Dropsleutel, 2020-04-06:
Maybe we could enter an IR value ? Or even Betaflight may be even possible to roughly compute it and evaluate battery quality?  Comparing Current with Volt fluctuations. 

IF IR is known, it would be possible to calculate the open pack voltage at land, which is different than the pack under load.

Juzzle1, 2020-04-07:
> Maybe we could enter an IR value ? Or even Betaflight may be even possible to roughly compute it and evaluate battery quality? Comparing Current with Volt fluctuations.
> 
> IF IR is known, it would be possible to calculate the open pack voltage at land, which is different than the pack under load.

@Dropsleutel I added a feature request for this a while back (https://github.com/betaflight/betaflight/issues/9298)

I had a look at the code to see if it would be possible to port over from iNav, but in all honesty it's _way_ beyond my ability.

peterfoleyusa, 2020-10-22:
> Hi!
>
> I am recently experimenting with liion on a 6'' quad, but I have the following issue:
> I have to manually adjust the warning voltage eachi time I am switching in between so the liion will not [...]

I've recently started flying my 6 inch with both Lipo and LiIon batteries as well.  The ability to be able to switch battery profiles with a switch would be super helpful.

rs3hc, 2021-01-07:
Yes bring it.
Same here. 
On one 6" quad I use 4s and 5s LiPo and Li-ion 🤟🤟🤟🤟

wbertrand, 2021-01-20:
Ho Yes, it would be great to be able to switch from one battery profile to another with a switch ;)

ClickNciak, 2021-01-22:
I think will Be a good improvement, you use different battery and sometimes the Volt/measurements in osd is different battery(brands) than other "brands", 100% and you have a battery arm and after landing, you can see that you still had voltage available to continue the flight...(damn) 
so as for pid / rate first and Vtx power now, this would be an added value for everyone, right?
Thanks for your great work🙌🏽
Still Amazing 🤩

haslinghuis, 2021-01-23:
is there a reason to not measure voltage at startup and apply warning and min dynamic and if so what are the pitfalls to address to going forward?

haslinghuis, 2021-01-23:
Would it  be possible to calculate instead of requiring user intervention? Would only work with fully loaded batteries though.

wbertrand, 2021-01-26:
operator intervention is necessary to choose the type of battery used: Lipo, Liion ... because the low battery alarms are different

cptX, 2023-08-01:
I'm also interested in this. Recently I built a battery with li-ion cells. They have the needed capacity but because they have higher internal resistance the produce a higher voltage drop compared to my other lipo batteries, so this triggers the warnings too early. I would like to switch the battery profile with a switch...

Martivip, 2023-09-17:
I need this feature too, INAV aproach looks good

gleavek, 2023-10-02:
I often fly my Flywoo Explorer LR4 with a mixture of packs. I'll be using a 750/900mah LIPO where I need my warnings for landing at 3.5A for example. However also fly with LION too where I can obviously fly down to 3A. As when out int he field I'll take both sets of batteries and use them both in the same session, it's frustrating I can't alter the voltage warnings.

In the POWER AND BATTERY tabs you would just need an option to set 2 lots of values. At the top have the option for LIPO and then below LION. So you can set your defaults for both. Then just have a switch to change to LIPO or LION on your controller and no need to mess with anything else then.

Martivip, 2023-11-09:
@gleavek that is a brilliant idea ! battery setting profiles selectable by switch !

Dear Devs there any chances to implement that ?

Martivip, 2024-01-18:
Dears Devs what do you think about that functionality?

pegurov, 2024-01-18:
I actually figured out a way to change warning and min voltage in the field.
You can:
1. Use Betaflight OSD menu to change these settings.
2. Install Betaflight LUA script on your radio and change it from there.

Martivip, 2024-02-13:
> I actually figured out a way to change warning and min voltage in the field. You can:
> 
> 1. Use Betaflight OSD menu to change these settings.
> 2. Install Betaflight LUA script on your radio and change it from there.

 via OSD menu I can change only the battery alarm not the battery capacity that is presented under battery section and via BF Lua scrip there is no both possibilities neither.
So the battery profiles selectable by switches under the battery section is the way to go...

Martivip, 2024-06-22:
Dear Devs, it will be really handy for users with Li-Ion and Lipo batteries

Mick51, 2024-07-31:
Why is this feature ignored by Devs when it makes perfect sensé ?

Martivip, 2024-07-31:
> Why is this feature ignored by Devs when it makes perfect sensé ?

I think maybe we need to promote this PR on main Betaflight Facebook group. Many of us don't have any idea that PR exist 😎

Mick51, 2024-07-31:
> > Why is this feature ignored by Devs when it makes perfect sensé ?
> 
> I think maybe we need to promote this PR on main Betaflight Facebook group. Many of us don't have any idea that PR exist 😎

I think it's a good idea many now use lipo and also li-ion. So it interests a lot of pilots I think....

mfiferna, 2024-07-31:
> Why is this feature ignored by Devs when it makes perfect sensé ?

Because work takes time and noone has so far decided to donate their time to implement this? Anyone can implement it and submit a PR you know...

berturion, 2024-09-07:
This feature would be absolutely awesome.

insane78, 2024-09-14:
4 years and nothing, please consider to implement this feature. Thanks

Andreas-Menzel, 2024-11-06:
This would be a great feature! I have a 10" that I use for freestyle with a LiPo and for Long Range with a LiIon.

Martivip, 2024-12-02:
Dear Devs, There are any chances to implement tis feature ? if donation needed we community will donate it

ramkrishna757575, 2025-01-11:
This is a very important feature for a lot of pilots. It would be awesome if this could be made available in betaflight

Martivip, 2025-02-03:
Dear Devs, There are any chances to implement this feature ?

KondratokGK, 2025-03-09:
We need it!

anti-vaxxer, 2025-03-12:
> any chances to implement this feature ?

I think it hasn't been implemented in the present because it's a _**future**_ request.