betaflight/betaflight #9424: SD card no recognized by beta Flight 10.6.0 using Omnibus F7 V2

iskymaster, 2020-01-29:
I am using the FC with 8 GB Sandisk SD card.

SD card not recognized by Beta Flight 10.6.0 using Omnibus F7 V2

BetaFlight version 10.6.0

No matter what method I try, the BetaFlight doesn't see the SD card. It says "No card inserted". I have changed the config. settings to SD Card, 1Khz, NONE.

The SD Card was formatted using, windows file format. if I reboot the FC using, "mass storage device (MSC) mode" I can access the card.

Any ideas what I am doing wrong here?

etracer65, 2020-01-29:
Reformat the card, try a different card. Also provide the information that was requested when you opened the issue. 10.6 is not a Betaflight version.

iskymaster, 2020-01-30:
What information do I need to provide?

mikeller, 2020-01-30:
@iskymaster: The information you were asked to provide when you opened this request:

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:

**Expected behavior**
A clear and concise description of what you expected to happen.

**Flight controller configuration**
Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/)

Use `resource show all` to create a resource allocation list and post it here in a code block. Put ``` (three backticks) at the start and end of the output block.

**Setup / Versions**
 - Flight controller [what type is it, where was it bought from];
 - Other components [RX, VTX, brand / model for all of them, firmware version where applicable];
 - how are the different components wired up.

**Additional context**
Add any other context about the problem here.

closed 2020-01-31 by no-response[bot]