betaflight/betaflight #9426: Servo Tilt not workong on 4.x.x with rpm filters

HABTomas, 2020-01-29:
I have tried to activate servo tilt on the 4.1.3 software with RPM filters with no luck.
I have made the servo out on 3.6 and it workt on same quad and hardware .
I use DJI FPV controler with DALR/DALRCF405(STM32F405)
I activate servo tilt , go to servo tab , activate the A3 on servo 0
go to CLI and change the LED strip to servo 1 on B06.

this worked on 3.6 but not on 4.X.X
any ideas how to solve that ??
this is how it looks after assigning resource B06 to Servo instead of LED.

# Building AutoComplete Cache ... Done!
#
# resource
resource BEEPER 1 C13
resource MOTOR 1 B00
resource MOTOR 2 C06
resource MOTOR 3 A10
resource MOTOR 4 A08
resource MOTOR 5 C08
resource MOTOR 6 B01
resource MOTOR 7 C07
resource MOTOR 8 C09
resource SERVO 1 B06
resource PPM 1 B15
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 A00
resource SERIAL_TX 5 C12
resource SERIAL_RX 1 B07
resource SERIAL_RX 2 A03
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 A01
resource SERIAL_RX 5 D02
resource I2C_SCL 1 B08
resource I2C_SDA 1 B09
resource LED 1 C14
resource SPI_SCK 1 B03
resource SPI_SCK 2 B13
resource SPI_SCK 3 C10
resource SPI_MISO 1 A06
resource SPI_MISO 2 B14
resource SPI_MISO 3 C11
resource SPI_MOSI 1 A07
resource SPI_MOSI 2 C03
resource SPI_MOSI 3 B05
resource ESCSERIAL 1 A03
resource CAMERA_CONTROL 1 A05
resource ADC_BATT 1 C02
resource ADC_RSSI 1 A00
resource ADC_CURR 1 C01
resource FLASH_CS 1 B12
resource OSD_CS 1 A15
resource GYRO_EXTI 1 C04
resource GYRO_CS 1 A04

#
# Building AutoComplete Cache ... Done!
#
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: FREE
A03: FREE
A04: GYRO_CS 1
A05: CAMERA_CONTROL
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: MOTOR 4
A09: FREE
A10: MOTOR 3
A11: USB
A12: USB
A13: FREE
A14: FREE
A15: OSD_CS
B00: MOTOR 1
B01: FREE
B02: FREE
B03: SPI_SCK 1
B04: FREE
B05: SPI_MOSI 3
B06: SERVO 1
B07: SERIAL_RX 1
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: FREE
B11: FREE
B12: FLASH_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: FREE
C00: FREE
C01: ADC_CURR
C02: ADC_BATT
C03: SPI_MOSI 2
C04: GYRO_EXTI
C05: FREE
C06: MOTOR 2
C07: FREE
C08: FREE
C09: FREE
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SERIAL_TX 5
C13: BEEPER
C14: LED 1
C15: FREE
D00: FREE
D01: FREE
D02: SERIAL_RX 5
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2:
CH1: CAMERA_CONTROL
TIM3: FREE
TIM4:
CH1: SERVO 1
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
CH1: DSHOT_BITBANG 2
CH2: DSHOT_BITBANG 3
CH3: DSHOT_BITBANG 1
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC 1
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 2
DMA2 Stream 3: DSHOT_BITBANG 3
DMA2 Stream 4: DSHOT_BITBANG 1
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

best regards
Tomas

**Flight controller configuration**
Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/)

Use `resource show all` to create a resource allocation list and post it here in a code block. Put ``` (three backticks) at the start and end of the output block.

**Setup / Versions**
 - Flight controller DALR/DALRCF405(STM32F405) from Himodel
 - Other components DJI FPV system
 -standard servo to LED strip out.

**Additional context**
this works on 3.6 but not on 4

etracer65, 2020-01-29:
Provide the settings `diff` as requested when you opened the issue. Also there is no Betaflight 3.6 version.

HABTomas, 2020-01-30:
hello 
it was the previous betaflight version before 4.x 
# Building AutoComplete Cache ... Done!
# 
# diff

# version
# Betaflight / STM32F405 (S405) 4.1.3 Jan 16 2020 / 11:34:00 (543a5e1ba) MSP API: 1.42
# manufacturer_id: DALR   board_name: DALRCF405   custom defaults: YES

# start the command batch
batch start

board_name DALRCF405
manufacturer_id DALR

# resources
resource SERVO 1 B06
resource LED_STRIP 1 NONE

# servo
servo 0 1000 2000 1500 100 6

# feature
feature SERVO_TILT

# serial
serial 0 64 115200 57600 0 115200
serial 4 1 115200 57600 0 115200

# aux
aux 0 0 0 900 1100 0 0
aux 1 1 1 1300 1700 0 0
aux 2 2 1 1750 2100 0 0
aux 3 8 2 1300 1700 0 0
aux 4 13 3 900 1175 0 0

# master
set gyro_sync_denom = 2
set gyro_lowpass2_hz = 325
set dyn_notch_range = LOW
set dyn_notch_width_percent = 0
set dyn_notch_q = 90
set dyn_notch_min_hz = 200
set dyn_lpf_gyro_min_hz = 260
set dyn_lpf_gyro_max_hz = 650
set mag_hardware = NONE
set baro_hardware = NONE
set fpv_mix_degrees = 15
set min_throttle = 1070
set dshot_bidir = ON
set use_unsynced_pwm = OFF
set motor_pwm_protocol = DSHOT300
set motor_pwm_rate = 480
set bat_capacity = 1700
set ibata_scale = 170
set yaw_motors_reversed = ON
set pid_process_denom = 1
set osd_vbat_pos = 2503
set osd_current_pos = 2515
set osd_mah_drawn_pos = 2509
set osd_power_pos = 2176
set osd_warnings_pos = 14664
set osd_rol_ang_pos = 32
set osd_battery_usage_pos = 2537

profile 0

# profile 0
set dyn_lpf_dterm_min_hz = 91
set dyn_lpf_dterm_max_hz = 221
set dterm_lowpass2_hz = 195
set p_pitch = 58
set i_pitch = 81
set d_pitch = 48
set f_pitch = 86
set p_roll = 53
set i_roll = 77
set d_roll = 44
set f_roll = 81
set p_yaw = 38
set i_yaw = 81
set f_yaw = 81
set d_min_roll = 25
set d_min_pitch = 28

rateprofile 0

# rateprofile 0
set roll_srate = 60
set pitch_srate = 60
set yaw_srate = 60

# end the command batch
batch end

jflyper, 2020-02-04:
@HABTomas Do you still have this issue?

HABTomas, 2020-02-04:
switched off the RPM filter 
here is like it looks like, Sorry I´m not that good at this.
# version
# Betaflight / STM32F405 (S405) 4.1.3 Jan 16 2020 / 11:34:00 (543a5e1ba) MSP API: 1.42
# manufacturer_id: DALR   board_name: DALRCF405   custom defaults: YES

# start the command batch
batch start

board_name DALRCF405
manufacturer_id DALR

# resources
resource SERVO 1 B06
resource LED_STRIP 1 NONE

# servo
servo 0 1000 2000 1500 100 6

# feature
feature SERVO_TILT

# serial
serial 0 64 115200 57600 0 115200
serial 4 1 115200 57600 0 115200

# aux
aux 0 0 0 900 1100 0 0
aux 1 1 1 1300 1700 0 0
aux 2 2 1 1750 2100 0 0
aux 3 8 2 1300 1700 0 0
aux 4 13 3 900 1175 0 0

# master
set gyro_sync_denom = 2
set gyro_lowpass2_hz = 325
set dyn_notch_range = LOW
set dyn_notch_width_percent = 0
set dyn_notch_q = 90
set dyn_notch_min_hz = 200
set dyn_lpf_gyro_min_hz = 260
set dyn_lpf_gyro_max_hz = 650
set mag_hardware = NONE
set baro_hardware = NONE
set fpv_mix_degrees = 15
set min_throttle = 1070
set use_unsynced_pwm = OFF
set motor_pwm_protocol = DSHOT300
set motor_pwm_rate = 480
set bat_capacity = 1700
set ibata_scale = 170
set yaw_motors_reversed = ON
set pid_process_denom = 1
set osd_vbat_pos = 2503
set osd_current_pos = 2515
set osd_mah_drawn_pos = 2509
set osd_power_pos = 2176
set osd_warnings_pos = 14664
set osd_rol_ang_pos = 32
set osd_battery_usage_pos = 2537

profile 0

# profile 0
set dyn_lpf_dterm_min_hz = 91
set dyn_lpf_dterm_max_hz = 221
set dterm_lowpass2_hz = 195
set p_pitch = 58
set i_pitch = 81
set d_pitch = 48
set f_pitch = 86
set p_roll = 53
set i_roll = 77
set d_roll = 44
set f_roll = 81
set p_yaw = 38
set i_yaw = 81
set f_yaw = 81
set d_min_roll = 25
set d_min_pitch = 28

rateprofile 0

# rateprofile 0
set roll_srate = 60
set pitch_srate = 60
set yaw_srate = 60

# end the command batch
batch end

# resource
resource BEEPER 1 C13
resource MOTOR 1 B00
resource MOTOR 2 C06
resource MOTOR 3 A10
resource MOTOR 4 A08
resource MOTOR 5 C08
resource MOTOR 6 B01
resource MOTOR 7 C07
resource MOTOR 8 C09
resource SERVO 1 B06
resource PPM 1 B15
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 A00
resource SERIAL_TX 5 C12
resource SERIAL_RX 1 B07
resource SERIAL_RX 2 A03
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 A01
resource SERIAL_RX 5 D02
resource I2C_SCL 1 B08
resource I2C_SDA 1 B09
resource LED 1 C14
resource SPI_SCK 1 B03
resource SPI_SCK 2 B13
resource SPI_SCK 3 C10
resource SPI_MISO 1 A06
resource SPI_MISO 2 B14
resource SPI_MISO 3 C11
resource SPI_MOSI 1 A07
resource SPI_MOSI 2 C03
resource SPI_MOSI 3 B05
resource ESCSERIAL 1 A03
resource CAMERA_CONTROL 1 A05
resource ADC_BATT 1 C02
resource ADC_RSSI 1 A00
resource ADC_CURR 1 C01
resource FLASH_CS 1 B12
resource OSD_CS 1 A15
resource GYRO_EXTI 1 C04
resource GYRO_CS 1 A04

# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: FREE
A03: FREE
A04: GYRO_CS 1
A05: CAMERA_CONTROL
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: MOTOR 4
A09: FREE
A10: MOTOR 3
A11: USB
A12: USB
A13: FREE
A14: FREE
A15: OSD_CS
B00: MOTOR 1
B01: FREE
B02: FREE
B03: SPI_SCK 1
B04: FREE
B05: SPI_MOSI 3
B06: SERVO 1
B07: SERIAL_RX 1
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: FREE
B11: FREE
B12: FLASH_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: FREE
C00: FREE
C01: ADC_CURR
C02: ADC_BATT
C03: SPI_MOSI 2
C04: GYRO_EXTI
C05: FREE
C06: MOTOR 2
C07: FREE
C08: FREE
C09: FREE
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SERIAL_TX 5
C13: BEEPER
C14: LED 1
C15: FREE
D00: FREE
D01: FREE
D02: SERIAL_RX 5
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1:
    CH1: MOTOR 4
    CH3: MOTOR 3
TIM2:
    CH1: CAMERA_CONTROL
TIM3:
    CH3: MOTOR 1
TIM4:
    CH1: SERVO 1
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1: MOTOR 2
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: TIMUP 3
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC 1
DMA2 Stream 1: TIMUP 8
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: TIMUP 1
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

jflyper, 2020-02-04:
Ah, no, I meant if switching off RPM filtering makes any difference in terms of servo functionality.

HABTomas, 2020-02-04:
hi tested again , tested the servo again and found that the servo is burned, I don´t know how that happend , but I measured the signal out and it seem to work now.
sorry for this confusion.
best regards
Tomas
Thank for the great job you are doing !!!!!!!!

closed 2020-02-04 by HABTomas