betaflight/betaflight #9427: BMP280 not visible on TMOTOR F7 HDV

angeldimitrov, 2020-01-29:
**Describe the bug**
The barometer can not be activated on the TMOTOR F7. The Barometer is connected to the SDA and SCL pins. Power is also connected as expected 3.3V and GND (I also tried on 5V) 

**To Reproduce**
After connecting the barometer i activated in from the configuration tab and then executed the following commands:
```
set baro_bustype = I2C
set baro_hardware = BMP280
save
```
**Expected behavior**
The barometer should work. 

**Flight controller configuration**
``` 

# version
# Betaflight / STM32F7X2 (S7X2) 4.1.3 Jan 16 2020 / 11:35:53 (543a5e1ba) MSP API: 1.42
# manufacturer_id: TMTR   board_name: TMOTORF7   custom defaults: YES

# start the command batch
batch start

board_name TMOTORF7
manufacturer_id TMTR

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature GPS
feature TELEMETRY

# map
map TAER1234

# serial
serial 1 1 115200 57600 0 115200
serial 3 2 115200 57600 0 115200
serial 4 64 115200 57600 0 115200

# aux
aux 0 0 0 1300 1700 0 0
aux 1 1 1 900 1300 0 0
aux 2 13 3 1700 2100 0 0

# master
set gyro_lowpass2_hz = 375
set dyn_notch_range = LOW
set dyn_notch_width_percent = 0
set dyn_notch_q = 200
set dyn_notch_min_hz = 90
set dyn_lpf_gyro_min_hz = 300
set dyn_lpf_gyro_max_hz = 750
set acc_calibration = 83,-7,312
set baro_bustype = I2C
set baro_hardware = BMP280
set rssi_channel = 16
set fpv_mix_degrees = 30
set serialrx_provider = FPORT
set serialrx_halfduplex = ON
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set yaw_motors_reversed = ON
set gps_provider = UBLOX
set osd_rssi_pos = 2497
set osd_rssi_dbm_pos = 176
set osd_mah_drawn_pos = 2432
set osd_gps_lon_pos = 2048
set osd_gps_lat_pos = 2080
set osd_gps_sats_pos = 2458
set osd_avg_cell_voltage_pos = 2546
set gyro_rpm_notch_harmonics = 1

profile 0

# profile 0
set dyn_lpf_dterm_min_hz = 105
set dyn_lpf_dterm_max_hz = 255
set dterm_lowpass2_hz = 225

rateprofile 0

# end the command batch
batch end

# 
```
```
resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: SERIAL_TX 4
A01: SERIAL_RX 4
A02: SERIAL_TX 2
A03: SERIAL_RX 2
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: FREE
A14: FREE
A15: FREE
B00: MOTOR 3
B01: MOTOR 4
B02: LED 1
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: FREE
B07: FREE
B08: FREE
B09: CAMERA_CONTROL
B10: I2C_SCL 2
B11: I2C_SDA 2
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_BATT
C02: ADC_CURR
C03: FREE
C04: GYRO_EXTI
C05: FREE
C06: MOTOR 1
C07: MOTOR 2
C08: FLASH_CS
C09: FREE
C10: FREE
C11: FREE
C12: SERIAL_TX 5
C13: BEEPER
C14: PINIO 1
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1: DSHOT_BITBANG 3
    CH2: DSHOT_BITBANG 2
TIM9: FREE
TIM10: FREE
TIM11:
    CH1: CAMERA_CONTROL
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 3
DMA2 Stream 3: DSHOT_BITBANG 2
DMA2 Stream 4: ADC
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

# 
```

**Setup / Versions**
 - TMOTOR F7 HDV Stack
 - LITEBEE Matek GPS & QMC5883L Compass Module M8Q-5883 Ublox SAM-M8Q
version where applicable];
 - SDA and SLC are used for Baro and Compass. Compass works fine.

**Additional context**
I have the same compass and barometer working on MAMBA405 with the latest legacy firmaware for mamba

mikeller, 2020-01-30:
@angeldimitrov: The internal pull ups on STM32 MCUs are on the weak side of things for I2C - this might be a case of the pull ups being enough for the compass, but not for the baro. Try adding external pull ups.

angeldimitrov, 2020-01-30:
@mikeller i am not sure i am getting the idea. I also tried the baro without the compass - in case there is not enough energy to supply them both. Is this what you are suggesting? 

Thank you!

mikeller, 2020-01-30:
@angeldimitrov: No, not quite. Have a read of https://learn.sparkfun.com/tutorials/i2c/i2c-at-the-hardware-level. I think the 4k7 value that they recommend should work for your case too.

angeldimitrov, 2020-02-03:
@mikeller I connected 2 resistors 4k7 - one between SDA and 5V and one between SCL and 5V. I also tried using 3.3V as well. The barometer is however this not visible in betaflight. Another suggestions are always welcome. Thank you!

jflyper, 2020-02-04:
@angeldimitrov A board pic from the URL you've referenced does not show a BMP280 chip onboard. T-Motor product page mentions the baro chip is optional upon user's request. Make sure your board has one onboard.

[image]

angeldimitrov, 2020-02-04:
@jflyper there is no build-in barometer. I connected one using the SDA / SCL pins. I added the exact same way a barometer to my MAMBA405.

jflyper, 2020-02-04:
Oh, in that case, you are on your own.

I would just scope the SCL/SDA lines with a logic/protocol analyzer to see what's going on. If you don't have one yet, you should get one if you are planning to play around with random peripheral modules.

closed 2020-03-12 by stale[bot]

comhic, 2023-01-05:
I asked T-Motor and they said if I could solder a BMP280 to the pad on the FC (in the red cercle on the picture) it should work.