betaflight/betaflight #9434: "gps_rescue_alt_mode = FIXED_ALT" not working properly

DamianFPV, 2020-02-01:
When I set "gps_rescue_alt_mode = FIXED_ALT" then 
my qwad is still climbing to the max alt during flight. I don't see any difference between MAX_ALT and FIXED_ALT

[BTFL_cli_KWAD1_20200201_135043.txt](https://github.com/betaflight/betaflight/files/4142813/BTFL_cli_KWAD1_20200201_135043.txt)

[BTFL_BLACKBOX_LOG_KWAD1_20200201_135117.zip](https://github.com/betaflight/betaflight/files/4142814/BTFL_BLACKBOX_LOG_KWAD1_20200201_135117.zip)

etracer65, 2020-02-01:
I think I found the problem. Please try this firmware (unzip to use).

[betaflight_4.1.4_STM32F405.hex.zip](https://github.com/betaflight/betaflight/files/4142859/betaflight_4.1.4_STM32F405.hex.zip)

To flash, select your board in the drop-down at the top "KAKUTEF4V2 (HBRO)" then click the "Load Firmware [local]" button in the bottom-right. Select the unzipped .hex file and then flash.

DamianFPV, 2020-02-02:
I did just a few tests, but it looks like it works as it should. Thanks

etracer65, 2020-02-02:
@DamianFPV Thanks for the feedback.

PR #9433 (2020-02-03): Fix GPS Rescue altitude mode behavior
Fixes #9434 

Fixed unassigned variable controlling the altitude mode. Unassigned value was defaulting to 0 forcing `MAX_ALT` mode at all times instead of the user setting.
closed 2020-02-03 by mikeller