betaflight/betaflight #9440: millivoltosd no telemetry and voltage info

trojanGoat, 2020-02-02:
Hi,
I'm using the revoltosd firmware on my millivoltosd FC.  I'm also using a spark32 4in1 with blheli32 and rxsr fport with access firmware.  

It seems like telemetry is not working at all and the fc itself is not even getting any voltage information from the esc.   Is there a different firmware I should be using?

It seems like smart audio and all other rx related control is working.

Here is my diff:

`# Betaflight / REVOLTOSD (ROSD) 4.1.3 Jan 16 2020 / 11:29:12 (543a5e1ba) MSP API: 1.42

# start the command batch
batch start

board_name REVOLTOSD

# name: trojanGoat

# resources
resource MOTOR 1 A02
resource MOTOR 2 A03
resource MOTOR 3 B01
resource MOTOR 4 B00

# feature
feature TELEMETRY
feature OSD
feature ESC_SENSOR

# beacon
beacon RX_SET

# serial
serial 2 2048 115200 57600 0 115200
serial 5 1024 115200 57600 0 115200

# aux
aux 0 0 0 1275 2100 0 0
aux 1 27 2 1700 2100 0 0
aux 2 13 1 1700 2100 0 0
aux 3 35 3 1700 2100 0 0
aux 4 36 2 1300 1700 0 0
aux 5 39 3 900 1300 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665    0 5885 5905    0    0
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 3
vtxtable powervalues 14 20 26
vtxtable powerlabels 25 100 400

# master
set acc_hardware = NONE
set serialrx_provider = FPORT
set serialrx_inverted = ON
set blackbox_device = NONE
set min_throttle = 1070
set dshot_idle_value = 350
set dshot_bidir = ON
set use_unsynced_pwm = OFF
set motor_pwm_protocol = DSHOT600
set motor_pwm_rate = 480
set failsafe_delay = 0
set failsafe_switch_mode = KILL
set failsafe_throttle_low_delay = 0
set align_board_roll = 180
set bat_capacity = 1300
set current_meter = ESC
set battery_meter = ESC
set small_angle = 180
set deadband = 2
set pid_process_denom = 2
set tlm_halfduplex = OFF
set osd_warn_core_temp = OFF
set osd_cap_alarm = 1500
set osd_tim2 = 2577
set osd_tim_2_pos = 2067
set osd_mah_drawn_pos = 2416
set osd_craft_name_pos = 2050
set osd_warnings_pos = 2378
set osd_avg_cell_voltage_pos = 2409
set osd_battery_usage_pos = 361
set osd_disarmed_pos = 2347
set osd_esc_tmp_pos = 370
set osd_esc_rpm_pos = 14
set osd_stat_max_spd = OFF
set debug_mode = SMARTAUDIO
set vtx_band = 5
set vtx_channel = 4
set vtx_power = 1
set vtx_freq = 5769
set vcd_video_system = NTSC
set name = trojanGoat`

mikeller, 2020-02-03:
@trojanGoat: The setup looks all good. I am not an expert in FlightOne hardware, so I do not know what target should be used for this board. What is important is to make sure that the wire from the ESCs is connected to the RX pin of the UART (in this case UART6).

trojanGoat, 2020-02-08:
In that case, I will assume that I don't have the correct firmware.  If I flash raceflight, then the current and voltage monitoring works.  Where can I find info on what firmware I should use?  I tried checking BF forums and docs already.

mikeller, 2020-02-08:
I’d ask the manufacturer. We are not provided with engineering samples or information on this board, so all we have to go on are guesses and user feedback.

cudakeer808, 2020-02-23:
battery_meter should at least be ADC. did you jump the current/tx6 pad?

closed 2020-03-31 by stale[bot]