betaflight/betaflight #9442: Just a question not a feature

yo90bosses, 2020-02-03:
I'm building my own flight controller and software for my aerospace degree and wanted to know if the flight controllers like betaflight, KISS etc. hold there angle at a setpoint or if they only try to stop unwanted rotations when in Acro mode.
So if say they have a set orientation that the pilot can rotate and the flight controller tries to hold this set rotation so if it say hits a branch and the quad spins if it will completely return to the original rotation or if the quad only controls rotational speed.

Sorry if I'm posting this in the wrong place. I don't know where else to ask this.

etracer65, 2020-02-03:
In normal acro (sometimes called "rate" mode) the flight controller is only concerned about controlling the commanded rotational rates of each axis. It does not make any effort to control orientation.

The exception would be in self-level modes. Then the flight controller is working to maintain the commanded angle for roll (X) and pitch (Y) based on the gravity reference (down). Yaw (Z) behaves like in acro mode with only an attempt to maintain rotation rate.

yo90bosses, 2020-02-03:
Ok so Betaflight calculates for example the D value in the PID algorithm by calculating the change in the rotation speed?

etracer65, 2020-02-03:
Well if you want to get into the details of the PID controller it's a bigger discussion and probably not appropriate here. Probably should join our slack.

https://slack.betaflight.com

But P is calculated from the error (desired rate - actual rate). D is calculated from the derivative of the gyro rate change.

closed 2020-03-11 by stale[bot]