betaflight/betaflight #9446: CRSF Telemetry not reporting remaining capacity

SteveCEvans, 2020-02-04:
Using an X9D to display telemetry from a crossfire module, the discovered sensors include RxBt, Curr and Capa corresponding to sensor IDs 0008/1, 0008/2 and 0008/3 respectively. 

crsf.c defines the telemetry data sent to be:

```
/*
0x08 Battery sensor
Payload:
uint16_t    Voltage ( mV * 100 )
uint16_t    Current ( mA * 100 )
uint24_t    Fuel ( drawn mAh )
uint8_t     Battery remaining ( percent )
*/
```

I've attempted manually creating a 0008/4 sensor to pick up the ```Battery remaining``` value, but it doesn't receive any data.

I am seeing the same on a number of quads. Is this an issue with the data being transmitted by the ```crsfFrameBatterySensor()``` routine (which looks OK to me!) or the Crossfire LUA script running on the transmitter?

SteveCEvans, 2020-02-07:
Can anybody else confirm this?

klutvott123, 2020-02-11:
Looks like this is an opentx issue that will be fixed for 2.3.6 https://github.com/opentx/opentx/pull/7377

closed 2020-03-19 by stale[bot]