betaflight/betaflight #9447: Is it possible to make an SERIAL_RX to be an SERIAL_TX

exocode, 2020-02-04:
I have a Fortini F4 FC. It says it has 5 UARTS (in there manual they say: "5x Serial UARTs for USB, SBUS, SPORT, TBS Receiver or GPS, Telemetry VTX")
So for what is UART2 and UART3? 

**My idea is, to use the S.Bus (input) as an SERIAL_TX to connect a Blackbox Logger. Is that possible? and if yes/no: WHY/HOW?**

Here is the manual for my exact model: https://www.dronelec.com/Fichiers/produits/document/144804user_manual_fortini_f4_osd.pdf

I know one UART must be left for USB
**THANK YOU IN ADVANCE**
Here is my current usage:
```
1xUART1 --> GPS
1xUART2 --> ???
1xUART3 --> ??? 
1xUART4 --> VTX (TX)
1xUART2 --> USB
1xUART6 --> Crossfire
```
When I write `resource`and `resource show` and align them properly it looks like this:

```
resource PPM 1 B09: FREE

GPS:
resource SERIAL_RX 1 A10: SERIAL_RX 1
resource SERIAL_TX 1 A09

Their manual say: "S.Bus RX3" which is not used because I use Crossfire
resource SERIAL_RX 3 B11: FREE
resource SERIAL_TX 3 B10: FREE

resource SERIAL_RX 5 D02: FREE
resource SERIAL_TX 5 C12: SPI_MOSI 3

VTX: 
resource SERIAL_RX 4 A01: FREE
resource SERIAL_TX 4 A00: SERIAL_TX 4

CROSSFIRE:
resource SERIAL_RX 6 C07: SERIAL_RX 6 
resource SERIAL_TX 6 C06: SERIAL_TX 6
```

If it helps: here is the complete `resource` and `resource show`
```
A00: SERIAL_TX 4
A01: FREE
A02: MOTOR 4
A03: MOTOR 3
A04: FREE
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: GYRO_CS 1
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: FREE
A14: FREE
A15: FREE
B00: MOTOR 1
B01: MOTOR 2
B02: FREE
B03: OSD_CS
B04: BEEPER
B05: LED 1
B06: LED 2
B07: LED_STRIP
B08: FREE
B09: FREE
B10: FREE
B11: FREE
B12: FREE
B13: FREE
B14: FREE
B15: FREE
C00: FREE
C01: ADC_CURR
C02: FREE
C03: FREE
C04: GYRO_EXTI
C05: FREE
C06: SERIAL_TX 6
C07: SERIAL_RX 6
C08: SYSTEM
C09: FREE
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: FREE
C14: SYSTEM
C15: INVERTER 3
D02: FREE

# resource
resource BEEPER 1 B04
resource MOTOR 1 B00
resource MOTOR 2 B01
resource MOTOR 3 A03
resource MOTOR 4 A02
resource PPM 1 B09
resource LED_STRIP 1 B07
resource SERIAL_TX 1 A09
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 A00
resource SERIAL_TX 5 C12
resource SERIAL_TX 6 C06
resource SERIAL_RX 1 A10
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 A01
resource SERIAL_RX 5 D02
resource SERIAL_RX 6 C07
resource INVERTER 3 C15
resource LED 1 B05
resource LED 2 B06
resource SPI_SCK 1 A05
resource SPI_SCK 3 C10
resource SPI_MISO 1 A06
resource SPI_MISO 3 C11
resource SPI_MOSI 1 A07
resource SPI_MOSI 3 C12
resource ESCSERIAL 1 B00
resource ADC_BATT 1 C02
resource ADC_CURR 1 C01
resource FLASH_CS 1 B03
resource OSD_CS 1 B03
resource GYRO_EXTI 1 C04
resource GYRO_CS 1 A08
```

jflyper, 2020-02-04:
@exocode 
> Is it possible to make a SERIAL_RX to be a SERIAL_TX

In general, NO, because an UART's RX pin can not be assigned to a TX function.

Only exception is the new, never been on the FC market yet G4 based boards, on which a certain UART's RX pin can be assigned to TX of another UART.

(Just for a little tip, the other way around is possible; using TX pin as RX by putting the UART into half-duplex mode and forget about TX side.)

> My idea is, to use the S.Bus (input) as an SERIAL_TX to connect a Blackbox Logger. Is that possible? and if yes/no: WHY/HOW?

You can connect your serial logger if you have PB10 exposed directly as SERIAL_TX 3 (or TX3 or T3 or whatever label that suggest UART3's TX pin). It's not converting SERIAL_RX to SERIAL_TX as suggested in the ticket's title.

jflyper, 2020-02-04:
Oh, and
> I know one UART must be left for USB

This is not true.

exocode, 2020-02-04:
ok thank you!

closed 2020-02-04 by exocode