betaflight/betaflight #9451: omnibusf4sd (v2 Pro) unable to map LED strip to PPM

diggit, 2020-02-05:
**Describe the bug**
OmnibusF4 v2 pro is unable to drive LED strip through dedicated LED connector pin (HW design bug). At the same time, my PPM header is unused. I'd like to use this header to drive my LED strip. After unmapping PPM a PWM 1 resource from PPM header (PB8), mapping LED_STRIP to B8, saving and configuring LED strip, there is no output on PPM pin (checked with oscilloscope). When I try to map LED strip pin to any motor output (and remove motor resource from that pin), LED signal is there. I've also verified physical connection from PPM pin to PB8 on STM32F405RG.
[Schematic of newer version here](https://www.dropbox.com/s/b4hq3ikyu77gejm/OmnibusF4-Pro-Sch.pdf).

Is there something hidden preventing LED_STRIP assignment to PPM(unmapped) pin?

**To Reproduce**
 - Try to configure LED strip to PPM header (remap resource)
 - Nothing on PPM pin (even when you try to remap MOTOR 1 there)
 - only constant 0.7V on pin

**Expected behavior**
 - led strip driving signal appears on PPM pin

**Flight controller configuration**
```
# diff

# version
# Betaflight / STM32F405 (S405) 4.1.3 Jan 16 2020 / 11:34:00 (543a5e1ba) MSP API: 1.42
# manufacturer_id: AIRB   board_name: OMNIBUSF4SD   custom defaults: YES

# start the command batch
batch start

board_name OMNIBUSF4SD
manufacturer_id AIRB

# name: ORION

# resources
resource PPM 1 NONE
resource PWM 1 NONE
resource LED_STRIP 1 B08

# feature
feature TELEMETRY
feature LED_STRIP

# beacon
beacon RX_LOST
beacon RX_SET

# serial
serial 2 32 115200 57600 0 115200

# led
led 0 0,0::C:2
led 1 0,1::C:2
led 2 0,2::C:2
led 3 0,3::C:2
led 4 0,4::C:2
led 5 0,5::C:2
led 6 0,6::C:2
led 7 0,7::C:2
led 8 0,8::C:2
led 9 0,9::C:2

# master
set serialrx_provider = SBUS
set motor_pwm_protocol = DSHOT600
set align_board_yaw = 270
set small_angle = 30
set pid_process_denom = 1
set gyro_1_align_yaw = 2700
set name = ORION

profile 0

rateprofile 0

# end the command batch
batch end
```
```
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: MOTOR 4
A03: MOTOR 3
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: FREE
A14: FREE
A15: OSD_CS
B00: MOTOR 1
B01: MOTOR 2
B02: FREE
B03: BARO_CS
B04: BEEPER
B05: LED 1
B06: FREE
B07: SDCARD_DETECT
B08: FREE
B09: FREE
B10: SERIAL_TX 3
B11: FREE
B12: SDCARD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_CURR
C02: ADC_BATT
C03: FREE
C04: GYRO_EXTI
C05: FREE
C06: FREE
C07: SERIAL_RX 6
C08: INVERTER 6
C09: INVERTER 3
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: FREE
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2:
    CH3: MOTOR 4
    CH4: MOTOR 3
TIM3:
    CH3: MOTOR 1
    CH4: MOTOR 2
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8: FREE
TIM9: FREE
TIM10:
    CH1: LED_STRIP
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: TIMUP 3
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: TIMUP 2
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: ADC 2
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```
It's kinda weird, that B08 shows *FREE* even when LED_STRIP is mapped there. (Yeah, resource config was saved using `save` command and this is after reboot.) See that TIM1CH1 is mapped to LED_STRIP.

**Setup / Versions**
 - Flight controller info, identifier: BTFL, version: 4.1.3
 - Running firmware released on: Jan 16 2020 11:34:00
 - Board: AIRB/OMNIBUSF4SD(STM32F405), version: 0
 - Board is OmnibusF4 v2 Pro (SD card)
 - Flight controller [what type is it, where was it bought from];
 - Other components [RX, VTX, brand / model for all of them, firmware version where applicable];
 - how are the different components wired up.

**Additional context**
Tested on 2 identical boards with same result.

etracer65, 2020-02-05:
OmnibusF4 boards like that have a solder jumper that selects between SBUS and PPM for the shared pad. So if the jumper is set to SBUS then the SBUS/PPM pad connects through an inverter to a UART RX. In that case there is no connection to the pin used for PPM on the MCU. IF you set the solder jumper to PPM then the SBUS/PPM pad connects to the pin resourced to PPM, but you won't be able to use SBUS.

diggit, 2020-02-05:
Thanks for reply. *V2 Pro* has 2 0402 jumper resistors (0R) near PPM header. (matching schematic). Both are assembled. One connects PPM to PB8, the other one to inverter input. I've removed one conecting PPM pin to inverter and voltage on PPM pin increased to 3.3V, but still constant value.

etracer65, 2020-02-05:
You're resource output above shows that PB8 is not used for anything.
```
B08: FREE
```

jflyper, 2020-02-05:
Unified target with OMNIBUSF4SD defaults loaded does not define a timer for PB8.

diggit, 2020-02-05:
Oh, wow, I completely missed config of timers and DMA. Unfortunately, DMA1_Stream7 is already used by motors ans I have to disable them to make LED strip working. Otherwise LED strip blinks in random pattern. (probably mix of Dshot 600 and LED signal?)

diggit, 2020-02-05:
Dma TIMUP 2 could be handled by DMA1_Stream1 to release DMA1_Stream7 for LED_STRIP, but is it even possible?

jflyper, 2020-02-05:
It should be (If I've ever coded the config part correctly lol)

jflyper, 2020-02-05:
How about BIDIR?

diggit, 2020-02-05:
`# dma` in CLI does not mention TIMUP assignments at all

jflyper, 2020-02-05:
Sorry, TIMUP configurability is only available for H7 (and G4) ...

diggit, 2020-02-05:
Oh, so it is limitation of Betaflight? That's the better option.

jflyper, 2020-02-05:
You are not planning to run RPM filtering?

jflyper, 2020-02-05:
You will get
```

diggit, 2020-02-05:
:D nice workaround, but my ESCs are Blheli-S based which AFAIK does not support bidirectional Dshot.

jflyper, 2020-02-05:
BLHeli_S has a RPM folk. Did you check it out?

diggit, 2020-02-05:
I saw it recently mentioned just with BlHeli32. Will check it out.

jflyper, 2020-02-05:
I think there's a licensed version called JESC as well as a free version.

closed 2020-03-13 by stale[bot]