betaflight/betaflight #9462: Asgard32 F7 Target Wrong Gyro Alignment

dukey32123, 2020-02-09:
AG3X/target.h states: 
"GYRO_1_ALIGN    CW0_DEG_FLIP" when it should be CW270_DEG_FLIP

You can see the directon of the Gyro in the attatched files 

[image]
[image]

etracer65, 2020-02-09:
I don’t see any alignment indicators on the board so how can you assume the alignment is wrong?

dukey32123, 2020-02-09:
> I don’t see any alignment indicators on the board so how can you assume the alignment is wrong?

That's why i attatched the 2nd picture, you can see the motor outputs ... for better illustration: 

[image]

mikeller, 2020-02-10:
I think @jflyper should know more about this?

closed 2020-03-18 by stale[bot]