betaflight/betaflight #9467: Frsky_X spi rx jitter MATEKF411RX target

Gozzz, 2020-02-09:
Frsky_X on build build 2071 has a bad gitter not present on build 1947. Almost feels like too high P but dropping it doesn't help, suspecting micro failsafes. Only fix is to bind D8 using Frsky_D and it is back to smooth.

Flash build 1947 and use SPI RX/Frsky_X, no issues with flight, then flash build 2071 with SPI RX/Frsky_X, massive amounts of oscillations.

I'd expect build 2071 to fly the same as 1947 with SPI RX/Frsky_X. Diff below if what I flew build 1947 on and same diff I used on 2071 (minus changing PID loop to 4k)

```
# diff all

# version
# Betaflight / STM32F411 (S411) 4.2.0 Nov  6 2019 / 01:07:01 (4ba9602e6) MSP API: 1.43
# manufacturer_id: MTKS   board_name: MATEKF411RX   custom defaults: YES

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name MATEKF411RX
manufacturer_id MTKS
mcu_id 003300193239511334393435
signature 

# name: MGOZZ M65 2

# feature
feature TELEMETRY

# beeper
beeper -ON_USB

# beacon
beacon RX_LOST

# serial
serial 1 2048 115200 57600 0 115200

# aux
aux 0 0 0 1700 2100 0 0
aux 1 35 1 1700 2100 0 0

# adjrange
adjrange 0 0 2 900 2100 12 2 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 1
vtxtable powerlabels 25

# master
set gyro_sync_denom = 2
set gyro_lowpass2_hz = 300
set dyn_lpf_gyro_min_hz = 240
set dyn_lpf_gyro_max_hz = 600
set acc_calibration = -180,-32,-290,1
set mag_hardware = NONE
set baro_hardware = NONE
set min_check = 1005
set airmode_start_throttle_percent = 55
set blackbox_device = NONE
set dshot_idle_value = 1000
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set motor_poles = 12
set vbat_max_cell_voltage = 440
set vbat_min_cell_voltage = 300
set vbat_warning_cell_voltage = 315
set current_meter = NONE
set small_angle = 180
set pid_process_denom = 1
set runaway_takeoff_prevention = OFF
set osd_warn_visual_beeper = OFF
set osd_warn_launch_control = OFF
set osd_warn_no_gps_rescue = OFF
set osd_warn_gps_rescue_disabled = OFF
set osd_cap_alarm = 200
set osd_tim2 = 513
set osd_vbat_pos = 385
set osd_rssi_pos = 2403
set osd_tim_2_pos = 2455
set osd_flymode_pos = 2371
set osd_throttle_pos = 2423
set osd_vtx_channel_pos = 2389
set osd_mah_drawn_pos = 344
set osd_craft_name_pos = 2409
set osd_avg_cell_voltage_pos = 2433
set osd_disarmed_pos = 2442
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set osd_stat_battery = ON
set osd_stat_max_curr = OFF
set osd_stat_used_mah = OFF
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set osd_stat_max_fft = ON
set cpu_overclock = 108MHZ
set vtx_band = 4
set vtx_channel = 4
set vtx_power = 1
set vtx_freq = 5800
set frsky_spi_tx_id = 35,14
set frsky_spi_offset = -31
set frsky_spi_bind_hop_data = 6,164,89,14,174,99,24,184,109,34,194,119,44,204,129,54,214,139,64,224,149,74,234,159,84,9,169,94,19,179,104,29,189,114,39,199,124,49,209,136,59,219,144,69,229,154,79,0,0,0
set frsky_x_rx_num = 5
set gyro_1_align_yaw = 1800
set name = MGOZZ M65 2

profile 0

# profile 0
set dyn_lpf_dterm_min_hz = 84
set dyn_lpf_dterm_max_hz = 204
set dterm_lowpass2_hz = 180
set crash_recovery = ON
set iterm_relax = RPY
set p_pitch = 78
set i_pitch = 108
set d_pitch = 59
set f_pitch = 149
set p_roll = 73
set i_roll = 102
set d_roll = 55
set f_roll = 140
set p_yaw = 47
set i_yaw = 108
set f_yaw = 119
set angle_level_strength = 100
set level_limit = 70
set d_min_roll = 31
set d_min_pitch = 35

profile 1

# profile 1
set dyn_lpf_dterm_min_hz = 77
set dyn_lpf_dterm_max_hz = 187
set dterm_lowpass2_hz = 165
set vbat_pid_gain = ON
set anti_gravity_gain = 7000
set iterm_relax = RPY
set p_pitch = 70
set i_pitch = 25
set d_pitch = 90
set p_roll = 70
set i_roll = 25
set d_roll = 90
set p_yaw = 90
set d_min_roll = 80
set d_min_pitch = 80
set d_min_boost_gain = 30
set d_min_advance = 0

profile 2

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set thr_mid = 100
set roll_rc_rate = 130
set pitch_rc_rate = 130
set yaw_rc_rate = 120
set roll_expo = 17
set pitch_expo = 17
set yaw_expo = 16
set roll_srate = 27
set pitch_srate = 27
set yaw_srate = 28

rateprofile 1

# rateprofile 1
set thr_mid = 70
set thr_expo = 28
set roll_rc_rate = 135
set pitch_rc_rate = 135
set yaw_rc_rate = 125
set roll_expo = 19
set pitch_expo = 19
set yaw_expo = 19
set roll_srate = 27
set pitch_srate = 27
set yaw_srate = 28

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
# 
```

Setup- BNF BetaFPV Meteor65 (happens on 2 different quads), JESC 2.3 w/rpm filtering enabled.

etracer65, 2020-02-10:
Without logging showing the concern I don’t see much we can do.

Gozzz, 2020-02-10:
@etracer65 Yeah no bb on these. I could post dvr but that would probably not help. Maybe someone can recreate the issue? It's there for sure.

Gozzz, 2020-02-11:
So I even tried using a new model and rebinding, still same behavior on d16. I'm going to try to figure out which build it changes at. Might take me a bit of time to figure out but I assume that would help @etracer65?  Only reason it is a concern is because d16 feel much more locked in vs d8 for me and I'd like to figure it out..

Gozzz, 2020-02-15:
OK so as a test I downloaded the betafpv hex from their support site here and flashed it, then bound w/D16 - https://betafpv.freshdesk.com/helpdesk/attachments/27017701927

No jitter issues what so ever.

This is the diff-
`# diff all

# version
# Betaflight / MATEKF411RX (M41R) 3.5.7 Mar 15 2019 / 22:05:33 (e9130527c) MSP API: 1.40

board_name MATEKF411RX
manufacturer_id 
mcu_id 003300193239511334393435
signature 

# reset configuration to default settings
defaults nosave

# name

# resources

# mixer

# servo

# servo mix

# feature

# beeper

# beacon

# map

# serial
serial 1 2048 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1700 2100 0 0
aux 1 35 1 1700 2100 0 0

# adjrange

# rxrange

# vtx

# rxfail

# master
set min_check = 1005
set rc_smoothing_type = FILTER
set dshot_idle_value = 1000
set motor_pwm_protocol = DSHOT300
set small_angle = 180
set frsky_spi_tx_id = 35,14
set frsky_spi_offset = -36
set frsky_spi_bind_hop_data = 6,164,89,14,174,99,24,184,109,34,194,119,44,204,129,54,214,139,64,224,149,74,234,159,84,9,169,94,19,179,104,29,189,114,39,199,124,49,209,136,59,219,144,69,229,154,79,0,0,0
set frsky_x_rx_num = 5

# profile
profile 0

# profile
profile 1

# profile
profile 2

# restore original profile selection
profile 0

# rateprofile
rateprofile 0

set roll_rc_rate = 130
set pitch_rc_rate = 130
set yaw_rc_rate = 120
set roll_expo = 19
set pitch_expo = 19
set yaw_expo = 18
set roll_srate = 27
set pitch_srate = 27
set yaw_srate = 28

# rateprofile
rateprofile 1

# rateprofile
rateprofile 2

# rateprofile
rateprofile 3

# rateprofile
rateprofile 4

# rateprofile
rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save`

Really not sure what is going on here, as soon as I flash 4.1.4 or higher the jitter is back w/D16. Also not sure why it is only happening on my betafpv f4 1s boards. Any pointers in where to looks next would be great.

mikeller, 2020-02-15:
@Gozzz:

> OK so as a test I downloaded the betafpv 3.5.7 hex from their support site here and flashed it, then bound wD16 - betafpv.freshdesk.com/helpdesk/attachments/27017701927
> 
> No jitter issues whatsoever.

Ok, the fact that they need to distribute custom firmware to make this work suggests that the hardware that you have is an incorrect or faulty clone of the MATEKF411RX target, and that betafpv has got changes in their custom firmware that are required to make this clone work.
Your options are to either open a support case with betafpv and ask them to open a pull request with Betaflight for the changes that are needed to make this hardware work, or stick with the custom firmware that they give you.

Gozzz, 2020-02-15:
Thanks @mikeller I will reach out to them. This wouldn't be something I could compare with a resource dump or the like?

Also I didn't have this issue when I was running an older dev firmware (build 1947), it seems to have started when I updated to the most recent 4.2 dev builds but trying to fix it by flashing back to the "older" dev build doesn't seem to make any difference now, I still have the same behavior bound with D16.

mikeller, 2020-02-15:
@Gozzz: 

> This wouldn't be something I could compare with a resource dump or the like?

No, from the symptom description it sounds like it is custom built firmware, which will not show up in anything that the firmware returns, just in behaviour.

ceewo, 2020-02-16:
I have also experienced this issue with Betaflight 4.1.2, mine was shipped with the RC smoothing turned way up to smooth out the jitter which made for a very sluggish control response.  I eliminated the smoothing and changed to D8 which made the quad fly better but I have increased latency on D8 and would rather use D16

eovnu87435ds, 2020-02-26:
@Gozzz did you ever hear back from BetaFPV on this? Just got a betaFPV Meteor65 that came with 4.1.0 on it, and atrocious rx performance. Doing a dump, the hash matches the 4.1.0 release.

Gozzz, 2020-02-26:
@eovnu87435ds I have only had 2 communications with Betafpv, one saying they would look into it and the other generic support reply asking if I could supply a video of the issue along with answering a few other questions. I have replied to both and not heard back.

gilesvessey, 2020-03-08:
Confirming this is an issue. My meteor65 was flying great on 3.5.7, but upgrading to 4.1.4 presented the jittering, very similar to how you've described. Played around with PIDs, followed the community presets for a brushed whoop (on 4.0.x), played around with the filtering, and still getting a very bumpy ride.

Gozzz, 2020-03-08:
@gilesvessey if you could get dvr of how it flies bound with d16 vs. how smooth it flies with d8 and post it on youtube I can share it with betafpv. I just haven't had the time.

mikeller, 2020-03-09:
@Gozzz: I think I found the bug causing the jitter. It seems to have been caused by a per se harmless bug that had always been in the FrSky X SPI code that then resulted in jittery behaviour when RC smoothing was enabled. I fixed it in #9563. Can you try the following firmware:

[betaflight_4.2.0_STM32F745_e387400970.zip](https://github.com/betaflight/betaflight/files/4306095/betaflight_4.2.0_STM32F745_e387400970.zip)
[betaflight_4.2.0_STM32F7X2_e387400970.zip](https://github.com/betaflight/betaflight/files/4306096/betaflight_4.2.0_STM32F7X2_e387400970.zip)
[betaflight_4.2.0_STM32F411_e387400970.zip](https://github.com/betaflight/betaflight/files/4306097/betaflight_4.2.0_STM32F411_e387400970.zip)
[betaflight_4.2.0_STM32F405_e387400970.zip](https://github.com/betaflight/betaflight/files/4306098/betaflight_4.2.0_STM32F405_e387400970.zip)

closed 2020-03-09 by mikeller

Gozzz, 2020-03-09:
Great news @mikeller !! Thank you. This looks like it was merged in build 2116 on the jenkins site? I will flash the latest dev build this evening and give it a test.

FWIW I tried to change the rcsmoothing options a bunch when testing but never disabled it. Sorry I couldn't have been more help troubleshooting.

PR #9563 (2020-03-09): Fixed frame handling for FrSky SPI RX.
Fixes #9467.

A bug in the FrSky X SPI code caused new RC frames to be recognised / handled multiple times, leading to jittery behaviour when RC smoothing was enabled.
This fixes it:

[image]