betaflight/betaflight #9469: Calibration take too long

fgiudice98, 2020-02-10:
Calibration take at least 30s (or even 1 min) with the latest master build (pr #9460). In previous build (pr #9421) all was ok and the calibration is completed soon after the boot.
This happens in two STM32F411 board, one with MPU6000 (NOX) and one with ICM20602 (MATEKF411)

Here the diff of one of the two (the other is almost the same)

```# version
# Betaflight / MATEKF411 (MK41) 4.2.0 Feb  9 2020 / 10:11:29 (norevision) MSP API: 1.4

# resources
resource SERIAL_TX 2 NONE
resource SERIAL_TX 11 A02

# feature
feature GPS
feature LED_STRIP

# beeper
beeper -ON_USB
beeper -CRASH_FLIP

# serial
serial 0 2 115200 57600 0 115200
serial 1 64 115200 57600 0 115200
serial 30 2048 115200 57600 0 115200

# led
led 0 8,6:S:AOI:2
led 1 7,6:S:AOI:2
led 2 7,5:S:AOW:2
led 3 8,5:S:AOW:2

# mode_color
mode_color 6 0 2
mode_color 6 1 6

# aux
aux 0 0 0 900 1300 0 0
aux 1 1 1 900 1300 0 0
aux 2 13 3 900 1300 0 0
aux 3 15 7 900 1200 0 0
aux 4 15 7 1800 2100 0 0
aux 5 19 5 1800 2100 0 0
aux 6 19 5 900 1200 0 0
aux 7 35 6 900 1400 0 0
aux 8 39 2 900 1300 0 0
aux 9 49 6 1600 2100 0 0

# adjrange
adjrange 0 0 7 900 2100 30 7 0 0
adjrange 1 0 5 900 2100 29 5 0 0

# rxrange
rxrange 0 2000 1000
rxrange 2 2000 1000

# master
set gyro_calib_duration = 1950
set gyro_overflow_detect = OFF
set yaw_spin_recovery = OFF
set dyn_notch_width_percent = 0
set dyn_notch_q = 200
set dyn_notch_min_hz = 100
set dyn_notch_max_hz = 500
set acc_calibration = 12,-31,1,1
set baro_hardware = NONE
set min_check = 1080
set rssi_channel = 13
set rssi_scale = 255
set serialrx_provider = SPEK2048
set blackbox_device = NONE
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set motor_poles = 12
set current_meter = NONE
set 3d_deadband_throttle = 10
set small_angle = 40
set gps_provider = UBLOX
set gps_sbas_mode = EGNOS
set gps_auto_baud = ON
set deadband = 5
set yaw_deadband = 5
set pid_process_denom = 2
set ledstrip_visual_beeper = ON
set ledstrip_visual_beeper_color = RED
set ledstrip_profile = RACE
set ledstrip_race_color = LIGHT_BLUE
set ledstrip_beacon_period_ms = 1000
set ledstrip_beacon_percent = 25
set debug_mode = SMARTAUDIO
set cpu_overclock = OFF
set vcd_video_system = NTSC
set displayport_max7456_wht = 0
set gyro_rpm_notch_harmonics = 2

profile 0

# profile 0
set dyn_lpf_dterm_min_hz = 80
set dyn_lpf_dterm_max_hz = 180
set vbat_pid_gain = ON
set yaw_lowpass_hz = 150
set launch_control_mode = PITCHONLY

rateprofile 0

# rateprofile 0
set thr_mid = 40
set thr_expo = 37
set roll_rc_rate = 200
set pitch_rc_rate = 200
set yaw_rc_rate = 200
set roll_expo = 37
set pitch_expo = 37
set yaw_expo = 37
set roll_srate = 75
set pitch_srate = 75
set yaw_srate = 75```

etracer65, 2020-02-10:
Not sure if this is a joke. You have it set for 19.5 seconds.
```
set gyro_calib_duration = 1950
```

Also this is an extremely bad idea for any flight controllers with ICM gyros. On flight controllers with MPU6000 it's already disabled and this setting is unnecessary.
```
set gyro_overflow_detect = OFF
```

etracer65, 2020-02-10:
My guess would be that you flashed a nightly release and didn't do a full chip erase. As a result you ended up with corrupted settings.

fgiudice98, 2020-02-10:
Not a joke, never changed that variable in my life. I checked the diff i made before updating and gyro_calib_duration is not present. The fact that this changed the same way in two boards seems to me a configuration reading error than an user error.

PS: I'm not here to make fun of anybody, i checked all and made some tests before posting here, I always check the chip erase (and this is the reason why i have all previous diffs saved on my pc)

But thanks and sorry for your wasted time

closed 2020-02-11 by fgiudice98