betaflight/betaflight #9470: Betaflight 4.1.1 Jeti EXBus not working on FURYF4OSD

padel-fr, 2020-02-11:
**Describe the bug**
I just flashed Betaflight 4.1.1 on my FURYF4OSD and EXBus is not working anymore (No channels moving no response). Move the connexion to the UART TX Pin (without any resistor) as recommanded in the new Betaflight documentation  

When I flash 3.5.7 EXBus is working fine, with connection to the UART RX Pin + resistor.

**To Reproduce**
I tried also version 4.1.3 with the same result.

**Flight controller configuration**

Spécifications :
Manufacturer : Diatone
Model : GT-M530 
FC : Mamba F405
MCU : STM32F405
ESC: F40 4IN1 50A 3-6S ESC Dshot600

**Setup / Versions**

# version
# Betaflight / FURYF4OSD (FY4O) 4.1.1 Nov 15 2019 / 12:26:24 (1e5e3d369) MSP API: 1.42

# start the command batch
batch start

board_name FURYF4OSD
manufacturer_id DIAT

# feature
feature -AIRMODE
feature MOTOR_STOP
feature TELEMETRY

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -GPS_STATUS
beeper -RX_SET
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CRASH_FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL

# aux
aux 0 0 0 900 1300 0 0

# master
set serialrx_provider = JETIEXBUS
set motor_pwm_protocol = DSHOT600
set bat_capacity = 2200
set current_meter = ADC

profile 0

rateprofile 0

# end the command batch
batch end
 

**Additional context**
Receiver Jeti REX 6 - Version REX Receivers Update V1.13
Radiocontrol Jeti DS16 - 5.02 realease.

mikeller, 2020-02-13:
@padel-fr: Check this guide: 

[How.to.setup.your.RX.and.FC.for.Jeti.Ex-Bus.pdf](https://github.com/betaflight/betaflight/files/4197693/How.to.setup.your.RX.and.FC.for.Jeti.Ex-Bus.pdf)

 (added in #9478)

padel-fr, 2020-02-13:
Hi Mikeller, thanks for this guide, but the telemetry don't work even in release 4.0.6. I can confirm you all channel control is working fine but telemetry is missing and nothing appears on the jeti screen. 
I tried to downgrade in 3.5.7 release, move solder to RX and add resistor and in this case all was working well. Do you have any idea regarding this issue ?

Thanks in advance for your reply.

mikeller, 2020-02-13:
@padel-fr: No, I am not a JetiEx user, and neither do I have test hardware for it. But maybe @marv-t or @RoadyFPV can help?

RoadyFPV, 2020-02-13:
@padel-fr Hi, please try to use another uart. On my radix fc is a uart too that wont work. I solder to another uart and it work fine.

padel-fr, 2020-02-13:
@RoadyFPV  Hi, i tried to connect to UART 6 but nothing changes...I think to a limitation of the FC board. It doesn't matter. I will downgrade in release 3.5.7 and all will be fine.
thanks for your support.

closed 2020-02-13 by padel-fr