betaflight/betaflight #9476: Matek F722-Mini VBT switch not functioning

billyd60, 2020-02-12:
The matek f722-mini has a video tx power connection onboard that can be switched on and off via mode USER1. I have wired my vtx to VBT and the adjacent ground and enabled mode USER1 on an rc channel. The mode clearly enables when the switch is activated on my transmitter, in the configurator. However VBT still produces normal power and the VTX continues to transmit normally. It is my understanding the the power should be cut at VBT when USER1 is enabled.

I have tried the legacy matek f722se and the matek f722-mini hex with custom defaults applied, but neither works as intended.

I can supply a dump or diff all tonight. But I wanted to get this out there now in case there is an obvious step I missed in CLI or something like that, or it's a known issue. I couldn't find anything.

jflyper, 2020-02-12:
We need a `dump` or `diff`, as your issue description does not give what you have done in CLI at all.

billyd60, 2020-02-12:
Here's a dump:
[dumpQAV.txt](https://github.com/betaflight/betaflight/files/4195412/dumpQAV.txt)

billyd60, 2020-02-14:
My mistake wasn't using the correct mode user 1

closed 2020-02-14 by billyd60