betaflight/betaflight #9477: Kakute F7 Bug Betaflight 4.1.3

MacFerran, 2020-02-13:
If I flash Betaflight 4.1.3 in my
Kakute F7 & / ESC Tekko32 metal, one of the four engines does not turn if I return to 4.0.6 everything goes perfect, does anyone know the solution?

Asizon, 2020-02-13:
Please provide requested `diff`

Asizon, 2020-02-13:
Can you try to copy-paste this in cli and trying again?
`# timer
timer E13 AF1
# pin E13: TIM1 CH3 (AF1)
timer B00 AF2
# pin B00: TIM3 CH3 (AF2)
timer B01 AF2
# pin B01: TIM3 CH4 (AF2)
timer E09 AF1
# pin E09: TIM1 CH1 (AF1)
timer E11 AF1
# pin E11: TIM1 CH2 (AF1)
timer C09 AF3
# pin C09: TIM8 CH4 (AF3)
timer A03 AF2
# pin A03: TIM5 CH4 (AF2)
timer D12 AF2
# pin D12: TIM4 CH1 (AF2)
save`

closed 2020-02-14 by no-response[bot]