betaflight/betaflight #9480: Betaflight Blackbox Explorer can't read logs recorded with OpenLog. OpenLog text file contains unreadable text

KramRul, 2020-02-13:
Now I am trying to write logs on a flash drive using an OpenLog board. However, Betaflight Blackbox Explorer produces this error. 
[image]

Log file: 
[image]
[LOG00011 (2).TXT](https://github.com/betaflight/betaflight/files/4199425/LOG00011.2.TXT)

I use official [OpenLog](https://github.com/sparkfun/OpenLog) firmware
Config file settings:
115200,26,3,0,1,1,0
baud,escape,esc#,mode,verb,echo,ignoreRX 

Flight controller settings:
[image]
[image]
[image]
[image]
[image]

Can somebody solve this problem? Thank you.

etracer65, 2020-02-13:
The OpenLogger is not adequate for logging and data will overflow its buffer leading to corruption.

See here: #9101 

Try a more modern logger like OpenLager.

closed 2020-02-13 by etracer65

KramRul, 2020-02-13:
> The OpenLogger is not adequate for logging and data will overflow its buffer leading to corruption.
> 
> See here: #9101
> 
> Try a more modern logger like OpenLager.

Is this problem inherent in any version of Betaflight or just above 4.0? As you can see in the screenshot of the command status, I am currently using version 3.5.3.

etracer65, 2020-02-13:
The problem is that the logger can't keep up with the data. You can try using a faster baud rate and reducing the logging rate on the Blackbox tab, but OpenLogger is just not viable anymore. Too much data going to the log.

jflyper, 2020-02-13:
@KramRul Your line speed is 115.2K. You can try increasing this to 230.4 and see if your OpenLog stably record the entire duration of your flight. If not, there's no other option but to procure a faster serial recording device. Please note that there are no issues reported with the serial blackbox recording itself software-wise, neither with 3.5 or 4.x.

JoeriCapens, 2020-02-13:
@KramRul I've successfully used my OpenLog to record at 1 kHz, see https://github.com/betaflight/betaflight-configurator/commit/74445a4b4b62e3f86ebb5bc4cc37f903149d22d3#commitcomment-37217880

Set the serial port to 500000 baud using the command line (on my Eachine Trashcan I set "serial 1 128 115200 57600 0 500000") **and** get an SD card which can handle fast small writes. I went with the SanDisk Extreme 32 GB because it gets good random 4K writes in this review: https://www.jeffgeerling.com/blog/2018/raspberry-pi-microsd-card-performance-comparison-2018 . The Samsung Evo+ should be even better but I couldn't find it, the newer Evo Plus is only slightly better than the SanDisk.