betaflight/betaflight #9496: Battery percentage in OSD

kristjanbjarni, 2020-02-19:
I think it would be a good feature to be able to display the battery percentage based on voltage only. Voltage is more actual than using capacity which has to be fine tuned to actually work and can handle multiple battery sizes.

So the implementation that I am thinking about would be to use the current voltage, the value for the minimum cell voltage and the value of the voltage at startup as maximum since it's always somewhat lower than the maximum cell voltage. I also think some kind of averaging over a small timeframe would be needed to handle fluctuation in voltage and have more consistent display.

etracer65, 2020-02-19:
Battery voltage is a poor estimate of remaining capacity.

kristjanbjarni, 2020-02-22:
Then how about having battery percentage based on amp draw?

etracer65, 2020-02-22:
There is already a "Battery usage" element that shows a graphical bar of the remaining battery capacity.

kristjanbjarni, 2020-02-24:
Yes but it doesn't show it in percentage. The graphical bar although it can be useful takes up large part of the screen space and also makes it hard to read out the actual percentage while flying. In my opinion having the option of displaying it in percentage would be a good option to have.

ctzsnooze, 2020-03-18:
The upcoming motor output compensation PR has a debug with a 0-100 value that approximates battery voltage, where 0 means battery is at warning voltage (default 3.5), and 100 means battery at full voltage.  This may achieve what you're after.

alexmeliss, 2020-05-17:
I agree with @kristjanbjarni, would be helpful to have battery % in the osd, especially for beginners. Even if its not that accurate, but to have a rough idea for people when you are getting started with fpv. Advanced pilots wont use it anyways, but its cool to have.

I remember when i started, the TBS oblivion had it on his OSD, and it helped me a lot. The more you fly the more you will understand V and mAh. As a side note when enabling this Feature in the OSD could be also written that people should not 100% rely on it and its only here to give a rough estimate.

automatt1c, 2020-07-06:
This was also very useful when i had my TBS powercube. I'm not sure how they calculated the percentage but it was surprisingly accurate, on a smaller quad at least. It seemed like it used the battery voltage along with the amp draw and some smoothing added as well.

adam-ah, 2023-07-01:
+1 for this feature. Estimated % based on voltage is *way* better than simply looking at the bouncing current voltage and doing that estimation in my own head while flying. 

${batt_{percentage}}=123-\frac{123}{1+(x/3.7)^{80})^{0.165}}$

[source](https://electronics.stackexchange.com/questions/435837/calculate-battery-percentage-on-lipo-battery)