betaflight/betaflight #9498: Possible bug

kenthinson, 2020-02-20:
Hello. I have a possible bug. Not sure if it's in the ESC or Betaflight. I am hoping someone can point me in the right direction to check logs and find out where the fault is.

Configuration: Betaflight 4.1.4
[TRANSTECF411HD](https://www.transtechobby.com/products/transtec-f411-hd-for-dji-fpv-system?pr_prod_strat=copurchase&pr_rec_pid=4416263061590&pr_ref_pid=4325849170006&pr_seq=uniform)
[ESC: J-H-30](https://www.transtechobby.com/products/transtec-25a-4in1-blheli_s-esc?pr_prod_strat=copurchase&pr_rec_pid=4325849170006&pr_ref_pid=4416263061590&pr_seq=uniform)

I have tried loading both [JESC](https://jflight.net) and [JazzMaverick](https://github.com/JazzMaverick/BLHeli) to the ESC. In the attempt to enable Bidirectional DSHOT. The motors work normally when bidirectional is not enabled. However once I enable bidirectional I have a problem. Plugging in the battery I get only the first half of the esc startup sound. If I check the motor tab of Betaflight configurator I see 100% error rate.

So it's either 1. A bug in both JESC and JazzMaverick. Betaflight is ok. Or 2. It's a Betaflight bug. Or 3. I'm missing something. 

Can anyone advise me how to start looking at logs to trace this down? Thanks!

kenthinson, 2020-02-20:
@mikeller I believe I added all the missing info from the template. Sorry.

kenthinson, 2020-02-20:
After more googling I found the solution on the Facebook group.
```
dshot_bitbang = ON
feature -LED_STRIP
```

closed 2020-02-20 by kenthinson

vinz486, 2020-12-04:
Hi, i'm just fixed the same issue, using this board (no rx version): https://betafpv.com/products/f4-1s-aio-brushless-flight-controller?variant=34861563019398

ESC is Jazz, 96khz, with telemetry.

But, I had to set:
dshot_bitbang = OFF

Tried AUTO (default) and ON: not working.

Led strip is unrelated for me.

Thankfully I found your post @kenthinson, I was going crazy and I think many people will do too.

What's this bitbang? Never heard.