betaflight/betaflight #9503: Pite mode automatically activates when I unplug/ Reconnect lipo

Seas-ODz, 2020-02-21:
Hi, 
I am currently on betaflight 4.1.4 on Dys f4 pro v2 and a TBS unify pro V3 5.8ghz
since the update the pit mode activates each time I unplug/reconnect.

more when I want to know the version of my SA I get 220 and not 216 or 200 etc...

I can disable this pit mode by forcing on betaflight but i can’t fly with this
```
# 

# diff

# version
# Betaflight / STM32F405 (S405) 4.1.4 Feb  4 2020 / 17:25:51 (bc9715eec) MSP API: 1.42
# manufacturer_id: DYST   board_name: DYSF4PRO   custom defaults: YES

# start the command batch
batch start

board_name DYSF4PRO
manufacturer_id DYST

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL

# beacon
beacon RX_LOST
beacon RX_SET

# serial
serial 0 64 115200 57600 0 115200
serial 5 2048 115200 57600 0 115200

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665    0 5885 5905    0    0
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 IMD6     I CUSTOM  5732 5765 5828 5840 5866 5740    0    0
vtxtable powerlevels 4
vtxtable powervalues 0 1 2 3
vtxtable powerlabels 25 200 500 800

# master
set gyro_sync_denom = 2
set acc_calibration = -102,11,-252
set mag_hardware = NONE
set baro_hardware = NONE
set serialrx_provider = SBUS
set blackbox_device = NONE
set dshot_idle_value = 250
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set beeper_dshot_beacon_tone = 2
set small_angle = 180
set pid_process_denom = 1
set debug_mode = SMARTAUDIO
set vtx_band = 5
set vtx_channel = 7
set vtx_power = 2
set vtx_freq = 5880
set gyro_1_align_yaw = 1800

profile 0

rateprofile 0

# end the command batch
batch end

#
 ```

Asizon, 2020-02-21:
Yes it is related to RACE mode inside the pitmode, just cjange to FREE in osd, we need to add it to Configurator vtx tab

Asizon, 2020-02-21:
Also this vtx is a first generation SA 1.0.
https://github.com/betaflight/betaflight/blob/master/docs/VTX.md#smartaudio-20-device

Seas-ODz, 2020-02-21:
okay it look to work fine now, 
but I’m not sure this is a first genaration beacause now i get "216" in debug mod

closed 2020-03-29 by stale[bot]

fw42, 2020-10-30:
I'm having a similar (same?) issue with a TBS Unity Pro32 Nano, Betaflight 4.2.3, MTKS/MATEKF411RX(STM32F411). Changing to FREE in the OSD doesn't help. Unchecking "Pit mode" in the VTX tab will just active it again immediately after saving. Same thing happens in the betaflight OSD (so not a configurator bug).

Seas-ODz, 2020-10-30:
> I'm having a similar (same?) issue with a TBS Unity Pro32 Nano, Betaflight 4.2.3, MTKS/MATEKF411RX(STM32F411). Changing to FREE in the OSD doesn't help. Unchecking "Pit mode" in the VTX tab will [...]

Hi, it's weird because for me the problem has been just the RACE mod in the OSD