betaflight/betaflight #9504: Can't enter CLI commands

marcus202, 2020-02-21:
hello!

I can't enter some specific CLI commands like (set motor_throttle_limit) (set auto_profile_cell_count).

I hope somebody can help cause I'm clueless.

# set motor_output_limit = 90
###ERROR### Invalid name

# diff

# version
# Betaflight / FURYF4OSD (FY4O) 3.5.1 Sep  8 2018 / 05:21:25 (d9fb5ca13) MSP API: 1.40

board_name FURYF4OSD
manufacturer_id 

# name

# resources

# mixer

# servo

# servo mix

# feature
feature -RX_SERIAL
feature RX_PPM

# beeper

# beacon

# map

# serial

# led

# color

# mode_color

# aux
aux 0 0 1 1950 2100 0 0
aux 1 2 2 1750 2100 0 0
aux 2 26 3 1700 2100 0 0
aux 3 28 2 1350 1625 0 0
aux 4 35 0 1900 2100 0 0

# adjrange

# rxrange

# vtx

# rxfail

# master
set gyro_lowpass_type = BIQUAD
set gyro_lowpass_hz = 50
set gyro_lowpass2_type = BIQUAD
set acc_calibration = 12,-22,232
set rssi_channel = 12
set rc_smoothing_type = FILTER
set fpv_mix_degrees = 20
set motor_pwm_protocol = DSHOT600
set bat_capacity = 1500
set vbat_max_cell_voltage = 44
set current_meter = ADC
set ibata_scale = 260
set osd_tim1 = 2561
set osd_vbat_pos = 2514
set osd_rssi_pos = 2102
set osd_tim_1_pos = 2497
set osd_remaining_time_estimate_pos = 449
set osd_flymode_pos = 2081
set osd_current_pos = 2091
set osd_mah_drawn_pos = 2156
set osd_home_dir_pos = 340
set osd_home_dist_pos = 397
set osd_pid_yaw_pos = 455
set osd_warnings_pos = 2409
set osd_avg_cell_voltage_pos = 2506
set osd_battery_usage_pos = 14
set osd_stat_tim_1 = ON
set osd_stat_tim_2 = OFF

# profile
profile 1

set dterm_notch_cutoff = 0
set d_pitch = 30
set p_roll = 42
set i_roll = 49
set d_roll = 35
set d_yaw = 20

# rateprofile
rateprofile 0

set thr_mid = 49
set roll_rc_rate = 75
set pitch_rc_rate = 75
set yaw_rc_rate = 60
set roll_srate = 73
set pitch_srate = 73
set yaw_srate = 73
set throttle_limit_type = SCALE
set throttle_limit_percent = 80

Asizon, 2020-02-21:
`motor_output_limit` is added in Betaflight 4.0, update your board

marcus202, 2020-02-21:
> `motor_output_limit` is added in Betaflight 4.0, update your board

thanks, i'll try

marcus202, 2020-02-21:
> `motor_output_limit` is added in Betaflight 4.0, update your board

Thank you very much, it worked.   👍

closed 2020-03-18 by marcus202