betaflight/betaflight #9516: TALONF7V2-MPU6000 Overheating 85c and ABOVE   :O

TheZeroBeast, 2020-02-25:
**Describe the bug**
STM32F722RET6 Core Thermal Design Power (**TDP**) is exceeded on **TALONF7V2-MPU6000** when running BF4.1.4 release resulting in an overheating flight controller when not in flight which is still not acceptable.
As I don't have a schematic of this design I can't troubleshoot further. Can you please assist in identifying a solution.

I am thinking the ARM core has more pins set to sink current and dissipate heat than the package is designed to handle.

At a room temperature of 20c I get to 85c core temp within 5 minutes of USB being powered after loading the firmware...

**To Reproduce**
Steps to reproduce the behavior:

1. Buy a **TALONF7V2-MPU6000** (not connected to any external peripherals)
2. Connect to **USB Port**
3. Measure temperature over the coming minutes.
4. Notice the excessive thermal dissipation

**Expected behavior**
A flight controller, and indeed my other CLRacingF7 flight controller can be configured on the bench over USB without destroying itself and maintaining a reasonable ~40c operating temperature.

**Flight controller configuration**
No diff needed, this happens from base **TALONF7V2** firmware loaded

**Setup / Versions**
 - TALONF7V2-MPU6000 Purchased from PhaserFPV.com.au in Feb 2020

**Additional context**
Attached are FLIR videos of temp resting at a comfortable level when flash erased and still powered.
[image]

I also feel this is related to #8353 even though the report was lacking information.

mikeller, 2020-02-25:
@TheZeroBeast: This sounds a lot like this is a problem with this particular board - especially as a different board with the same MCU and the same firmware does not have this problem.

Have you tried raising this with your supplier or the manufacturer?

TheZeroBeast, 2020-02-25:
Hi @mikeller,

I have raised this with the supplier(PhaserFPV.com.au) and the manufacturer (Heli-Nation) and so far only have a response from PhaserFPV advising that CLRacing say 85c is acceptable for this board and it doesn't sound defective.

I'm thinking there might be a bad IO config for this board, where you provided any resources to support this board?

Do you have a link to the config for this board so I can take a look, it didn't jump out at me in the filestructure?

mikeller, 2020-02-25:
@TheZeroBeast: https://github.com/betaflight/unified-targets/blob/master/configs/default/HENA-TALONF7V2.config

jflyper, 2020-02-25:
@TheZeroBeast 

How about that you first flash the board with STM32F7X2 target binary first, then add resources and from the config file mikeller mentioned line by line until you find a line (or lines) you find the chip getting hot?

TheZeroBeast, 2020-02-26:
Hi @jflyper,

Good idea. I loaded the STM32F7X2 target and temp skyrockets like before.
I also tried BF4.0.0 and same result.

I wonder why no one else that owns a TALONF7V2 isn't worried about this.

TheZeroBeast, 2020-02-26:
Hi @jflyper,

Good idea. I loaded the STM32F7X2 target and temps do appear to stabilize lower.

It seams the DMA lines of the config make the temp start jumping.

I wonder why no one else that owns a TALONF7V2 isn't worried about this, the supplier makes it sound normal for this board. 🤦‍♂️😅

jflyper, 2020-02-26:
@TheZeroBeast 
Skip the dma lines and continue adding. If nothing else makes it hot, then try narrowing it down to which particular line has the effect.

nepalien, 2020-03-04:
I am interested to find out the the root cause of this too,

jflyper, 2020-03-07:
@nepalien Then you can do your own research. Like above.

Gozzz, 2020-03-08:
I have a TalonF7V2 which does the same. As long as it's flying it is OK but sitting too long on the ground and it starts to show core temp warnings.

TheZeroBeast, 2020-03-09:
I have spent some more time on this and don't think the config is to blame.

When no config it stabilises at about 60c, doing nothing really.
My CLRacingF7V2 30x30 board idles at about 40c running PID and everything.

When flying it pulls back down to 45c which is ok but that's a damn lot of air-cooling.

Do prey you don't crash and have the FC without cooling for a period.

mikeller, 2020-03-09:
Looks like bad thermal design - probably not enough copper in the PCB to draw away a significant amount of heat, made worse by being inside a gyro cage. Getting rid of the cage and using double sided tape or foam to hold on the gyro might help. Or 3D print a cage with ventilation holes.

TheZeroBeast, 2020-03-13:
Exceeded TDP for sure.
What do you mean by gyro cage as the gyro in soldered onto the board like the larger CLRacingF7 which doesn't have any of these issues?
And if you mean when it's installed, it's held mid-air on stand-offs.

mikeller, 2020-03-13:
@TheZeroBeast: Ok, if it does not have a gyro cage then this is not an option for improvement.

nepalien, 2020-03-20:
So I go a new Talon F7 board and ran some tests. I found that setting the Piniobox to "40,255,255,255" raises the temp to ~80C (Talon F7(HENA)Target). Changing it to "39,255,255,255"(copied from CLRacing Target) seems to stablize it at 66. 
Temps were measured setting the Debug_mode = ADC_Internal. 
Test were run using new just new FC connected to USB.

mikeller, 2020-03-20:
@nepalien: That's a good find - it looks like it's the on-board VTX switch that causes the overheating then - can you please post a full `diff`?

Gozzz, 2020-03-20:
@bnn1044 is there any way to jump the transistor for the vtx switch on this board?

nepalien, 2020-03-20:
diff did not have anything since it was default but here is the dump.

Talon Dump:
[TalonF7 BF4.1.5 Dump.txt](https://github.com/betaflight/betaflight/files/4360141/TalonF7.BF4.1.5.Dump.txt)

Also here is a Config compare between Talon/CLRA:
[TalonVsCLRA_Config.txt](https://github.com/betaflight/betaflight/files/4360147/TalonVsCLRA_Config.txt)

nepalien, 2020-03-20:
I also tested this on my working quad. Pinio_box was set to 39.  With battery plugged in with VTX pit switch off Temp stabilizes at ~70. As soon as you turn on the VTX switch ON temp shoots up to the 80's.

mikeller, 2020-03-20:
@nepalien: I was asking for the `diff` because I was interested to see how you'd have the switches set up. But you are answering it in your comment, and from what you are saying it looks like it is having the VTX switch on 'on' that is causing the overheating issue.
This looks like the VTX switch is drawing too much power from the MCU pin, so this could mean that the MCU will be damaged sooner or later if the VTX switch is used.
Contemplating removing the VTX switch functionality from this target until we have clarity on this.

mikeller, 2020-03-20:
@bnn1044: See the above comment. There seems to be a hardware issue with this board that could potentially lead to hardware damage if the wrong configuration is used.

closed 2020-03-21 by mikeller

bnn1044, 2020-03-26:
> @nepalien: I was asking for the `diff` because I was interested to see how you'd have the switches set up. But you are answering it in your comment, and from what you are saying it looks like it is [...]

1. question for @nepalien  what VTX do you use? 5V VTX or vbat  input?
2. the VTX pit Mode has nothing draw too much current from the MCU pin. I can measure out the 
    exact current draw when I have my device set up. 3.3V at 10K resister. no where near the limit of the IO pin on the MCU. 
3. You compare A 25mmx25mm factor fc to 35mm x 35mm  to the thermal dissipation. the surface area is twice bigger that 25X25MM. of course the CLRACING F7 runs cooler. 
  as hardware design. there is no different how to drive the PIT mode switch on the CLF7 AND the talon F7. 

@mikeller I will provide further testing on this and provide data. I don't see that will damage to the MCU from the pit switch. and I don't think there is any hardware issue with the FC.

bnn1044, 2020-03-26:
> @bnn1044 is there any way to jump the transistor for the vtx switch on this board?

you can if you can solder on tiny pins.

mikeller, 2020-03-26:
@mikeller: Thanks for looking into this - we are just erring on the side of caution, and this difference in temperature _could_ be an indication of a problem, so we need to investigate the cause for this.

nepalien, 2020-03-27:
@bnn1044 I was using a 5V VTX. (Oscar Backpack @ 25mw).

nepalien, 2020-03-30:
@bnn1044 This is happening on a brand new blank FC with nothing connected other than the USB. I was able to get the FC to 80C by flashing the current BF and changing the "resource PINIO 1 A14" and setting the VTX switch to turn on in the modes tab. There is nothing else connected to FC.
This is my Diff:
`# 

# diff

# version
# Betaflight / STM32F7X2 (S7X2) 4.1.5 Mar 16 2020 / 05:21:46 (d4e74e39c) MSP API: 1.42
# manufacturer_id: HENA   board_name: TALONF7V2   custom defaults: YES

# start the command batch
batch start

board_name TALONF7V2
manufacturer_id HENA

# resources
resource PINIO 1 A14

# aux
aux 0 39 0 1300 1700 0 0

# master
set debug_mode = ADC_INTERNAL
set pinio_box = 39,255,255,255

profile 0

rateprofile 0

# end the command batch
batch end

# 
[image]
[image]

`

wesk-ms, 2020-04-09:
I'm reproducing the exact same symptoms on a Matek F722 SE AIO.  85C and the board starts a continuous reboot loop.  Nothing soldered to the board when connected to USB.  I'll try some older BF builds to see if the behavior changes, but this doesn't seem like a hardware design issue to me.

joshuahardee, 2020-04-22:
Any update to this issue? I'm about to buy 4x of these FC's. Should I hold off?

mikeller, 2020-04-23:
@joshuahardee: The hardware seems to be fine if you are not planning on using the built-in VTX switch.

Gozzz, 2020-04-24:
One observations from this change, if the 5v vtx power is selected and pinio box is not configured the vtx will not power up, if vbat is selected for vtx power it seems to work fine. This may be an issue with some users, I have 2 running 5v and 2 running vbat which is how I noticed.

maxn1780, 2025-05-18:
Hello. my flight controller Aocoda F722mini was warmed up to 70 degrees if the elsr receiver was connected to any uart except the 1st one. When I connected the receiver to uart1, the overheating went away.