betaflight/betaflight #9534: CMS/OSD reboot required when changing a value and then reverting it

bizmar, 2020-03-02:
**Describe the bug**
I encountered it in the Features > Blackbox menu, it may also be elsewhere. If a parameter such the device or ratio is modified and then reverted, a reboot is required. 

**To Reproduce**
Enter OSD > Features > Blackbox modify and revert the device type, exit OSD.

**Expected behavior**
A reboot is not required. 

**Flight controller configuration**
Tested with:
Betaflight / STM32F411 (S411) 4.2.0 Mar  1 2020 / 09:36:10 (86b81baa1) MSP API: 1.43
manufacturer_id: HAMO   board_name: CRAZYBEEF4DX   custom defaults: YES

Asizon, 2020-03-02:
I also can reproduce it!Maybe some missing with `(*ptr->val != previousValue)` checks @etracer65 ??

etracer65, 2020-03-02:
In general the CMS doesn't have a "before" and "after" version of the setting. Changing a setting changes the one copy of it. So we can't do "only if value changed" types of logic.

closed 2020-04-08 by stale[bot]