betaflight/betaflight #9536: No # of satellites using geprc f722 pro dual

Aarons14bugs, 2020-03-02:
# If your issue looks like a hardware fault or a configuration problem please don't raise an issue here.

## Please consider using other user support options such as asking the manufacturer of the hardware you are using, RCGroups: https://rcgroups.com/forums/showthread.php?t=2464844, or Slack (registration at https://slack.betaflight.com/) or other user support forums & groups (e.g. facebook).

## If you believe there is an issue with the firmware itself please follow these steps:
1. Describe your problem;
Number of satellites are not showing up.

2. Include ways to reproduce the problem;
Just plug everything in and go for a spin.

3. Provide information about your flightcontroller and other components including how they are connected/wired;
I am running f722 pro dual geprc fc and the problem is present in both the stock fw as well as in 4.1.3.  I've tried both open UARTS 1 and 3 at auto baud as well as every other available connection is proper tx to rx and rx to tx on two tested bn-180 and bn-220 gps modules in every configuration possible. 
4. (See above. )
5.

```# diff

# version
# Betaflight / STM32F7X2 (S7X2) 4.1.4 Feb  4 2020 / 17:27:35 (bc9715eec) MSP API: 1.42
# manufacturer_id: EXUA   board_name: EXF722DUAL   custom defaults: YES

# start the command batch
batch start

board_name EXF722DUAL
manufacturer_id EXUA

# name: CineStyle F7

# feature
feature GPS
feature TELEMETRY

# beeper
beeper -ARMED
beeper -ON_USB

# beacon
beacon RX_LOST
beacon RX_SET

# serial
serial 1 8192 115200 57600 0 115200
serial 2 2 115200 57600 0 115200
serial 3 64 115200 57600 0 115200

# aux
aux 0 0 0 1725 2100 0 0
aux 1 1 1 1275 1700 0 0
aux 2 13 3 1675 2100 0 0
aux 3 19 6 1750 2100 0 0
aux 4 26 0 1700 2100 0 0
aux 5 27 4 1700 2100 0 0
aux 6 31 4 1700 2100 0 0
aux 7 35 3 1300 1700 0 0

# rxrange
rxrange 0 988 2011
rxrange 1 988 2011
rxrange 2 988 2011
rxrange 3 988 2011

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665    0 5885 5905    0    0
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 IMD6     I CUSTOM  5732 5765 5828 5840 5866 5740    0    0
vtxtable powerlevels 4
vtxtable powervalues 0 1 2 0
vtxtable powerlabels 25 100 200 500

# master
set gyro_sync_denom = 2
set gyro_lowpass2_hz = 325
set dyn_notch_width_percent = 0
set dyn_notch_q = 200
set dyn_notch_min_hz = 90
set dyn_lpf_gyro_min_hz = 260
set dyn_lpf_gyro_max_hz = 650
set mag_hardware = NONE
set baro_hardware = NONE
set serialrx_provider = FPORT
set serialrx_inverted = ON
set serialrx_halfduplex = ON
set dshot_idle_value = 250
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set failsafe_procedure = GPS-RESCUE
set small_angle = 180
set gps_provider = UBLOX
set gps_rescue_initial_alt = 55
set gps_rescue_descent_dist = 30
set gps_rescue_min_sats = 5
set gps_rescue_allow_arming_without_fix = ON
set pid_process_denom = 1
set osd_warn_core_temp = OFF
set osd_cap_alarm = 850
set osd_alt_alarm = 120
set osd_vbat_pos = 2519
set osd_rssi_pos = 2117
set osd_link_quality_pos = 2180
set osd_tim_2_pos = 2133
set osd_flymode_pos = 2476
set osd_throttle_pos = 2497
set osd_vtx_channel_pos = 33
set osd_current_pos = 2433
set osd_mah_drawn_pos = 2465
set osd_craft_name_pos = 2504
set osd_gps_lon_pos = 2063
set osd_gps_lat_pos = 2050
set osd_gps_sats_pos = 2147
set osd_home_dir_pos = 2158
set osd_home_dist_pos = 2124
set osd_altitude_pos = 2197
set osd_warnings_pos = 14729
set osd_avg_cell_voltage_pos = 2402
set osd_flip_arrow_pos = 2253
set osd_core_temp_pos = 2487
set osd_log_status_pos = 2456
set osd_stat_rtc_date_time = ON
set osd_stat_max_dist = ON
set osd_stat_battery = ON
set osd_stat_max_g_force = ON
set osd_stat_flight_dist = ON
set vtx_band = 5
set vtx_channel = 5
set vtx_power = 3
set vtx_low_power_disarm = ON
set vtx_freq = 5806
set gyro_2_align_yaw = 900
set name = CineStyle F7

profile 0

# profile 0
set dyn_lpf_dterm_min_hz = 91
set dyn_lpf_dterm_max_hz = 221
set dterm_lowpass2_hz = 195

rateprofile 0

# rateprofile 0
set roll_srate = 77
set pitch_srate = 77
set yaw_srate = 75

# end the command batch
batch end```

Doncuppjr, 2020-03-04:
I have that controller. I know it works with the BN-880 gps+compass. I can't get the compass to work, but I do have a functional gps. Does your gps module work when you plug it into a usb-serial converter?

Aarons14bugs, 2020-03-04:
I need to close this issue.   Turned out being horrible interference from the Caddx Tarsier camera.   After learning how to power it off I instantly had 8 locked satellites in my garage!   After restarting and running the camera, signal quality went down but still gets some satellites I think I'll be fine as long as I shut down the camera while it syncs up from a cold start.   Sorry for posting this on here, I never had a problem like this before.

closed 2020-03-04 by Aarons14bugs

etracer65, 2020-03-04:
If you had included that info in your post then we could have told you that was the problem. That camera is well known to produce major interference.