betaflight/betaflight #9542: Crazybee F4 V3 F4FR soft serial not working

LawrenceLDJr, 2020-03-05:
**Describe the bug**
Soft Serial not working. You can change vtx settings in betaflight PC, OSD and LUA script and the values do change and are saved. But the actual vtx does not change bands channels or power.   
**To Reproduce**
Remap LED Strip pin to TX 11 smart audio, ports tab, the changes you make appear in OSD, betaflight PC and LUA script and they all match. The vtx remains unchanged.

**Expected behavior**
Use my PC, OSD or Jumper T16 LUA script too change vtx band, channel and power via smart audio using soft serial.

**Flight controller configuration**
# diff all

# version
# Betaflight / STM32F411 (S411) 4.1.4 Feb  4 2020 / 17:26:13 (bc9715eec) MSP API: 1.42
# manufacturer_id: HAMO   board_name: CRAZYBEEF4FR   custom defaults: YES

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name CRAZYBEEF4FR
manufacturer_id HAMO
mcu_id 002a00383437510a37313935
signature 

# name: Cinecan

# resources
resource PWM 3 NONE
resource LED_STRIP 1 NONE
resource SERIAL_TX 11 A00

# feature
feature -RX_PARALLEL_PWM
feature -AIRMODE
feature -RX_SPI
feature RX_SERIAL
feature SOFTSERIAL
feature GPS
feature TELEMETRY

# beacon
beacon RX_LOST
beacon RX_SET

# serial
serial 0 64 115200 57600 0 115200
serial 1 2 115200 57600 0 115200
serial 30 2048 115200 57600 0 57600

# aux
aux 0 0 0 1300 2100 0 0
aux 1 1 1 900 1150 0 0
aux 2 1 1 1350 2100 0 0
aux 3 2 1 1075 1150 0 0
aux 4 6 1 1475 1525 0 0
aux 5 7 1 1600 1650 0 0
aux 6 13 3 1700 2100 0 0
aux 7 15 3 900 1300 0 0
aux 8 35 1 1950 2100 0 0
aux 9 36 3 1700 2100 0 0

# adjrange
adjrange 0 0 2 900 2100 29 2 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665    0 5885 5905    0    0
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 2
vtxtable powervalues 0 1
vtxtable powerlabels 25 200

# master
set gyro_sync_denom = 2
set acc_trim_pitch = 2
set acc_calibration = -52,32,-119
set mag_hardware = NONE
set baro_hardware = NONE
set rssi_scale = 200
set serialrx_provider = FPORT
set serialrx_halfduplex = ON
set rx_spi_protocol = FRSKY_D
set blackbox_device = SPIFLASH
set failsafe_procedure = GPS-RESCUE
set ibata_scale = 399
set beeper_dshot_beacon_tone = 5
set small_angle = 180
set gps_provider = UBLOX
set gps_rescue_initial_alt = 20
set gps_rescue_allow_arming_without_fix = ON
set pid_process_denom = 2
set osd_vbat_pos = 12674
set osd_rssi_pos = 12698
set osd_link_quality_pos = 4443
set osd_tim_2_pos = 4194
set osd_flymode_pos = 366
set osd_throttle_pos = 4343
set osd_vtx_channel_pos = 4469
set osd_crosshairs_pos = 4302
set osd_ah_sbar_pos = 4303
set osd_ah_pos = 4175
set osd_current_pos = 4449
set osd_mah_drawn_pos = 4418
set osd_craft_name_pos = 4428
set osd_gps_speed_pos = 4311
set osd_gps_sats_pos = 4215
set osd_home_dir_pos = 4303
set osd_home_dist_pos = 4290
set osd_compass_bar_pos = 4203
set osd_altitude_pos = 4322
set osd_pid_roll_pos = 2152
set osd_pid_pitch_pos = 2120
set osd_pid_yaw_pos = 2184
set osd_warnings_pos = 12683
set osd_disarmed_pos = 364
set osd_nheading_pos = 4174
set osd_rtc_date_time_pos = 12326
set osd_adjustment_range_pos = 2216
set osd_stick_overlay_left_pos = 2276
set osd_stick_overlay_right_pos = 2291
set osd_profile = 2
set debug_mode = SMARTAUDIO
set vtx_band = 2
set vtx_channel = 7
set vtx_power = 1
set vtx_freq = 5847
set vcd_video_system = NTSC
set frsky_spi_tx_id = 127,95
set frsky_spi_offset = -8
set frsky_spi_bind_hop_data = 6,64,122,180,3,61,119,177,1,58,116,174,232,55,113,171,229,52,110,168,226,49,107,165,223,46,104,162,221,43,101,159,217,40,98,156,214,37,95,153,211,34,92,150,208,31,89,0,0,0
set gyro_1_align_yaw = 900
set timezone_offset_minutes = -360
set name = Cinecan

profile 0

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set throttle_limit_type = SCALE
set throttle_limit_percent = 60

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
# 

Use `resource show all` to create a resource allocation list and post it here in a code block. Put ``` (three backticks) at the start and end of the output block.
 ``` # resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: SERIAL_TX 11
A01: GYRO_EXTI
A02: SERIAL_TX 2
A03: SERIAL_RX 2
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: SERIAL_TX 1
A10: FREE
A11: USB
A12: USB
A13: FREE
A14: FREE
A15: PREINIT 2
B00: ADC_BATT
B01: ADC_CURR
B02: FREE
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: MOTOR 2
B07: MOTOR 3
B08: MOTOR 4
B09: FREE
B10: MOTOR 1
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: FREE
C03: FREE
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: LED 1
C14: FREE
C15: BEEPER
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2:
    CH3: MOTOR 1
TIM3: FREE
TIM4:
    CH1: MOTOR 2
    CH2: MOTOR 3
    CH3: MOTOR 4
TIM5:
    CH1: SERIAL_TX 11
TIM6: FREE
TIM7: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: TIMUP 4
DMA1 Stream 7: TIMUP 2
DMA2 Stream 0: ADC 1
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

# ```
**Setup / Versions**
 - Flight controller [what type is it, where was it bought from]; 

Crazybee F4 V3 Cinecan, R9mm, Eachine VTX

 - Other components [RX, VTX, brand / model for all of them, firmware version where applicable];

VTX Smart Audio LED pad, R9MM FPort TX 1 
 - how are the different components wired up.

**Additional context**
Add any other context about the problem here.

etracer65, 2020-03-05:
Try using a real UART instead of softserial to differentiate between a vtx/vtxtable problem vs. a softserial problem. In other words, make sure you can control the vtx using a real UART first.

I would say that your `vtxtable` is incorrect as it appears to be for a SA 1.0 device.

Also hopefully the vtx is not the Eachine Nano. That vtx has a broken Tramp (not SA) implementation.

LawrenceLDJr, 2020-03-06:
Thx for reply... etracer65... the Cinecan comes setup with vtx SA on UART TX 1. Does work on UART. I've done the debug and the value I get is 216. Wich requires 2.0. I rearranged the Cinecan (CRAZYBEE F4 V3) with R9mm FPort on TX1, GPS on TX RX 2, freed LED STRIP added TX11 for smart audio. Values changed via SA then saved appear on betaflight PC, fpv OSD, and LUA on my radio. Because it seems to be working I've back tracked and all my 16x16 boards using softserial smartaudio and they also don't work. All 30x30 boards are working correctly. I have a habit of putting smartaudio on a softserial port. Because all it does is change values when you not flying.

mikeller, 2020-03-07:
@LawrenceLDJr:

Try adding 

```
resource SERIAL_RX 11 A00
```

and see if this fixes it.

LawrenceLDJr, 2020-03-07:
Smart audio on an RX...? I'll give it a whirl. Just wanna let ya know I will live with out it. In a years time I haven't changed from boscam A, Channel 1, power 25 enough times to discover none of my whoops actually have smart audio working on soft serial.

mikeller, 2020-03-07:
@LawrenceLDJr: I didn't say you should be removing the `resource SERIAL_TX 11 A00` line. But SmartAudio uses _bidirectional_ communication on a single pin, so the RX side needs to be defined as well.

LawrenceLDJr, 2020-03-08:
First try was just adding SERIAL_ RX 11 on A00, no success we're on the same page mikeller. I do have working smart audio on omnibus F4 V3 with a single TX 11 on motor 6. Discovery was a recent flight into basement with Cinecan landing on dryer. Fuzzy video compelled me too switch power from 25 to 200, no change so I switched channels and bands and never lost video even though the changes are shown and saved in OSD and LUA. So like I said I checked a whoop I've been flying for a year Trashcan. This has the JHEMCU NOX target board and what I assumed working smart audio via soft serial. This one also does not work but appears to be on PC, OSD and LUA. I did try SERIAL_RX 11 only as well and of course smart audio disappears on all three platforms.

LawrenceLDJr, 2020-03-08:
Will be closing issue... woke up this morning and did some soldering... connected the little vtx to a omnibus F4 V3 which has a working vtx SA soft serial motor 6. The Cinecan, Trashcan VTX smart audio did not work soft serial. It did work on the same boards UART. Culprit is eachine VTX. https://www.banggood.com/Eachine-TRASHCAN-75mm-FPV-Racing-Drone-Spare-Part-5_8G-25mW200MW-Switchable-VTX-FPV-Transmitter-3_3-5_5V-p-1411222.html?rmmds=search&cur_warehouse=CN 
It's a good vtx... no smart audio on soft serial I'm sure is a easy fix if enough customers are bugged about it.

mikeller, 2020-03-08:
@LawrenceLDJr:

> Culprit is eachine VTX.

Yeah, Eachine is well known to dump broken hardware onto their customers and never care to address the problems.

> no smart audio on soft serial I'm sure is a easy fix if enough customers are bugged about it.

I wouldn't hold my breath - Eachine has never cared about fixing problems with the broken hardware they sold, or collaborating with Betaflight to get their hardware supported.
Best vote with your wallet, and give Eachine a big 'No' next time you spend money on hardware.

closed 2020-03-09 by LawrenceLDJr

GregorPotrebujes, 2022-01-10:
I'm having the same problem with that crazybee board - I want to use the LED_STRIP pin as softserial to control the VTX with smartaudio. In the Betaflight -> Video transmitter menu it says "Device ready - Yes". But the VTX doesn't seem to be responding to any command (change of channel, band, power). BUT... I know that this softserial port works well, because until now I've been using it for my Frsky R9MX receiver over it's F.PORT pin. And it worked without a single glitch despite all the comments and warnings that you shouldn't connect your receiver to the softserial port, because it's too slow or something. One other fact: I also know, that my VTX smartaudio works well if I connect it to an actual UART. But as I said - if I connect it to the softserial (LED_STRIP pin) it just doesn't work. I tried all permutations of serialrx_inverted and serialrx_halfduplex. Any ideas what other CLI settings could I try in order to make it work?