betaflight/betaflight #9543: Develop New Implementation for VTX Tables

shawnames, 2020-03-05:
Users are having significant issues with VTX tables and resorting to button pushing.  

I would like to see a new implementation of VTX tables that are easier for the end user to configure.  This past weekend I had 5 individual conversations with pilots struggling to get on the right power output because of errors in setting up their VTX tables.  

There has to be a better way.  Identifying the appropriate powervalues for a give vtx can be very difficult.

mikeller, 2020-03-05:
@shawnames: The way forward is to take the manufacturers / vendors of VTX devices to task - they are making money from selling these devices, they should be the ones providing VTX tables that allow legal use of the devices in the jurisdiction they sell them into.
Please direct users reporting problems with setting up VTX tables to ask their VTX manufacturers / vendors to provide them with the tables that are required to use these devices.

etracer65, 2020-03-05:
@shawnames Where do you expect us to get the various power value or band/channel list settings for every vtx? The only people who can provide it are the manufacturers who built the devices and programmed them with specific settings. There is no way for anybody to "extract" the settings from the devices.

etracer65, 2020-03-05:
> if you can't find the info then how should random software users be expected to?

By contacting the manufacturer of the product as a customer with an expectation that the product they purchased is supported. Are you under some mistaken assumption that the manufacturers of these products consult with us in anyway or even provide samples to test? Short answer is no to both.

mikeller, 2020-03-07:
@shawnames:

> Take the manufacturers to task. Make them pay to have their VTX supported, I don't care but putting this on the users is failing and causing a mess at flying fields.

Good suggestion. Can you please contact all of the VTX manufacturers and get them to post their contact details in here so we can send them invoices for the work that needs to be done to support their hardware? Or can we send you an invoice to do this work for you?

ChilledoutUK, 2020-03-10:
So erm what was wrong with the old system for VTX control?

tenten8401, 2020-03-10:
I think at the very minimum we should have the default SmartAudio/Tramp ones in the wiki page integrated directly into the configurator that way users don't have to go hunting for them and can just select a preset.

jflyper, 2020-03-10:
@tenten8401 
I think you don't understand the fundamental purpose of the VTX table, which is to free Betaflight project from potential regulatory issues by NOT providing any defaults. Users must be responsible for how they configure their radio transmitters.

tenten8401, 2020-03-10:
@jflyper 
By selecting a preset and clicking save you are configuring the video transmitter yourself without any default behavior. Having presets there simply provides an easier way to do so. Maybe add a disclaimer about how it's probably not legal with said presets and that you need to edit them if it's actually a concern? It's not like the current system of visiting a wiki page and loading a json file from the wiki has any disclaimers about legality either: https://github.com/betaflight/betaflight/wiki/VTX-tables

Many video transmitters have to be unlocked in the first place to be able to access potentially illegal channels / power levels, and even then it's not like having no easy way to control a video transmitter will make it any easier to follow regulations. Users are responsible for making sure their video equipment is on a legal frequency / power level and purposefully making it more complicated than it needs to be can make it more difficult to follow regulations. Having a way to load in presets as a starting point for additional configuration without the need to go to the wiki would make setting up video transmitters much simpler.

mikeller, 2020-03-10:
@tenten8401: Feel free to implement and maintain a 'VTX configurator' that does this - we are happy to link it from Betaflight configurator, and send the FCC your way when they come asking whom they should be suing for facilitating illegal behaviour.

tenten8401, 2020-03-11:
I think you're missing what I'm trying to suggest. I'm not saying have it how it used to be where it'd just work, but rather an easier way to load in common channel values (not necessarily power levels) from a dropdown or something. Going to an official betaflight wiki page linked from the configurator to download the required files for this wouldn't be much better legally and makes it a pain for users to deal with. Could even force the VTX into pit mode (or just send nothing to vtx) until a user checks a box saying they have verified the above frequencies and power levels are compatible with local regulations. Did one of the Betaflight devs receive a nasty letter from the FCC over the old behavior? I fail to understand why there's so much negativity towards making SmartAudio a little less painful to configure...

mikeller, 2020-03-11:
@tenten8401: Again, feel free to contribute an app that does all of this.
Just Fyi, there are roughly 50 different jurisdictions that will have to be catered for, with at least one changing the rules every couple of months, so I'd expect to have to put a considerable amount of work into keeping in a state to keep you from being accused of facilitating illegal activity with it.
Betaflight has decided we prefer to spend the time we have available for advancing the project on flight performance improvements, instead of spending it on keeping up with ever changing legal compliance requirements, and has hence externalised and crowdsourced the legal compliance part. Once more, if you want to be part of the crowd that performs this task then feel free to chip in.

tenten8401, 2020-03-11:
@mikeller By checking the box preventing saving of values the user is accepting liability for having checked their local rules and regulations. If anything this would be better legally than the current situation as it is mentioned *nowhere* on that wiki page that it may be illegal when loading the current json files in and clicking save.

mikeller, 2020-03-11:
@tenten8401: There is no such thing as an _official Betaflight wiki page_ - any GitHub user can edit these pages, and any user needs to decide for themselves what set of settings they can legally use. This is how 'crowdsourcing' works.

tenten8401, 2020-03-11:
I mean it's linked from the betaflight configurator and is part of the official betaflight github project, I'd call that close to an official wiki page as it gets. Regardless, I'd still love to see a quicker way, however would be best to do it, to configure all the channels as it's currently the most annoying part of setting up my quads and around every other day I see people asking for help with it.

mikeller, 2020-03-11:
@tenten8401: And again, Betaflight has decided to crowdsource the effort to compile and maintain a list of frequency lists that can be legally used in various durisdictions around the globe.
If you'd love to see a quicker way then maybe you can become part of this crowd and contribute one?

RipperDrone, 2020-03-13:
Maybe you wanna use SpeedyBee for VTX table config - seems they have alrdy implemented s/t close to what you're asking for... 
[image]

closed 2020-04-19 by stale[bot]