betaflight/betaflight #9547: upload txt backup file via cli tab to fc does not work

cmmoser, 2020-03-06:
I am using a t-motor f4 and just updated to the latest firmware.
before updating i safed the diff values as txt file.
i treid two options to get the backup values back to the fc:
1. chose load from file, chose the txt file, all the data are loaded into the cli. pressed enter.
2. copied the content of the txt file in the command line and pressed enter

both did not safe the values to my fc
i also tried to enter "safe" in the command line but it says that command is unknown.

how do i paste backup values from a txt file via cli tab to the fc?

thank you very much

etracer65, 2020-03-06:
Open your saved text file and copy it and paste into the cli. Then do `save`.

Also don't use settings from a previous firmware version (like don't use settings from 4.0.x on 4.1.x). Settings `diff` are not compatible between versions. In that case configure manually and refer to your saved settings.

cmmoser, 2020-03-06:
thank you very much
i just found out the problem

I wrote safe instead of save (language issue) ;-)
sorry about that
if you are an administrator you can delete this thread

closed 2020-03-06 by cmmoser