betaflight/betaflight #9552: Switch printing library to mpaland/printf

mikeller, 2020-03-07:
We need to look into switching from the currently used tfprintf library to something a bit more advanced, like https://github.com/mpaland/printf/, to get around the shortcomings that tfprintf has (like lack of `float` support).

@ledvinap: You said at some point that you had some improvements to go into mpaland/printf. Have they been integrated by now?

ledvinap, 2020-03-07:
@mikeller : There is open PR with bugfixes and preview PR with changes ... 

I'm currently rewriting it a bit for my project, I hope to be done soon.

Only minimal changes are necessary for mpaland version - remove #define printf and add some workaround for ARM eabi - varargs `float` is promoted to `double`, IMO BF is still avoiding double libraries..

mikeller, 2020-03-07:
@ledvinap:

> There is open PR with bugfixes and preview PR with changes ...

Can you post a link to it? Happy to review and ask for it to be considered.

closed 2026-09-26 by blckmn