betaflight/betaflight #9555: magnetometer QMC5883

MastroDado, 2020-03-07:
gps 880q with inbuit QMC5883, the gps works ok but the compass is not recognized. i think a driver issue because only uploading inav and not touching anything else compass is working ok.
mambaf405 target fury f4 osd betaflight 4.1.4

RipperDrone, 2020-03-07:
Yes, QMC is not recognized in many cases, had the same issue with Matek QMC. BF has QMC5883 parameter in CLI, but sometimes still doesn't support. IT was s/t about timing  in I2C comm or device IDs being incorrect. Fortunately, QMC's are detected reliably in iNAV, so when you want MAGs support, it seems wothwhile to consider using iNAV. For BF gps_return_to_home, MAG is not needed...

qwe900, 2020-03-31:
Use 4.1.1 there the QMC5883 works

closed 2020-05-07 by stale[bot]