betaflight/betaflight #9556: somethings wrong with pid_process_denom and motor_pwm_protocol

TheIsotopes, 2020-03-07:
@etracer65 something is wrong since #9544

#9545 defines the `pid_process_denom` but this is overwritten with #9544.

This means that `pid_process_denom = 1` is set as default for an fc with an 8kHz gyro,
but with `motor_pwm_protocol = ONESHOT125` as default this is corrected to `pid_process_denom = 4`.

diff output after defaults:
```
# diff

# version
# Betaflight / SPRACINGH7EXTREME (SP7E) 4.2.0 Mar  7 2020 / 03:32:35 (c002d0435) MSP API: 1.43

# start the command batch
batch start

board_name SPRACINGH7EXTREME

# master
set pid_process_denom = 4

profile 0

rateprofile 0

# end the command batch
batch end

```

I now change `motor_pwm_protocol = DSHOT600`, then I can also set  `pid_process_denom = 1`.

After a diff the `pid_process_denom = 1` is not listed, since it is already set as default.

```
# diff

# version
# Betaflight / SPRACINGH7EXTREME (SP7E) 4.2.0 Mar  7 2020 / 03:32:35 (c002d0435) MSP API: 1.43

# start the command batch
batch start

board_name SPRACINGH7EXTREME

# master
set motor_pwm_protocol = DSHOT600

profile 0

rateprofile 0

# end the command batch
batch end

```
if I then load the config from file, i always have to set afterwards the `pid_process_denom = 1`

etracer65, 2020-03-07:
Other than it's different, what's wrong? In the first case the `diff` outputs 4 because the value is not the default. In the second case you changed it back to the default of 1 so the `diff` didn't output anything (since DSHOT600 doesn't require the  denom to be lowered).

etracer65, 2020-03-07:
But this can probably be better. Let me think about it.

TheIsotopes, 2020-03-07:
after defaults `dump` says `set pid_process_denom = 4`

now i `set pid_process_denom = 1` and after `diff` it is not listed

if i now (after defaults) load the config, it is set back to `set pid_process_denom = 4` 

I can't explain it in any other way.

TheIsotopes, 2020-03-07:
The diff output in my scenario should be:

```
 # diff
 
 # version
 # Betaflight / SPRACINGH7EXTREME (SP7E) 4.2.0 Mar  7 2020 / 03:32:35 (c002d0435) MSP API: 1.43
 
 # start the command batch
 batch start
 
 board_name SPRACINGH7EXTREME
 
 # master
 set motor_pwm_protocol = DSHOT600
 set pid_process_denom = 1
 
 profile 0
 
 rateprofile 0
 
 # end the command batch
 batch end
 
 # 
```

etracer65, 2020-03-07:
I improved it a little less. See https://github.com/betaflight/betaflight/pull/9558

TheIsotopes, 2020-03-07:
working again, thx 👍

PR #9558 (2020-03-09): Revise pid_process_denom default logic - second try
Fixes #9556 

This is a revision and partial revert of #9545.

While the previous version improved the behavior for non-8K gyros, it had some unfortunate side effects on the defaults and how they affected the output of the `diff`. Basically the problem is that in most cases the default value will end up getting adjusted during the first boot to work with the default low-speed motor protocol ONESHOT125. So if an appropriate default was originally set (like in the other PR) it shows up changed in the `diff` with default settings. The only worka [... cut]
closed 2020-03-09 by mikeller