betaflight/betaflight #9557: ESC Reboot loop if beeper is enabled when booting up

Givemeurcookies, 2020-03-07:
**Describe the bug**
The ESC end up in a reboot loop if the beeper function is enabled upon booting the quadcopter. The ESC will just continue rebooting over and over until you disable the beeper. I am using DShot600 BLHeli_32 firmware (newest as of now, don't remember version, if others are unable to reproduce the bug, I will find the version).

**To Reproduce**
Enable the beeper function in the configuration, turn on the transmitter and have i.e the beeper switch enabled. The ESC will reboot over and over until transmitter is disabled or flight controller is rebooted again without the switch enabled.

**Expected behavior**
I expected the craft to boot as normal, didn't realise the quad had beeper enabled but the expected behaviour would be to boot like normal and start beeping. I can see scenarios where having beeper enabled upon boot for example if you want the quad to beep when certain switches are enabled prior to arming, alerting you.

**Flight controller configuration**
```# version
# Betaflight / STM32F7X2 (S7X2) 4.1.3 Jan 16 2020 / 11:35:53 (543a5e1ba) MSP API: 1.42
# manufacturer_id: JHEF   board_name: JHEF7DUAL   custom defaults: YES

# start the command batch
batch start

board_name JHEF7DUAL
manufacturer_id JHEF

# feature
feature -RX_PARALLEL_PWM
feature -OSD
feature -AIRMODE
feature RX_PPM
feature MOTOR_STOP

# beacon
beacon RX_LOST
beacon RX_SET

# aux
aux 0 0 1 1250 2100 0 0
aux 1 1 0 1650 2100 1 0
aux 2 27 1 1800 2100 0 0
aux 3 47 0 900 1400 1 0
aux 4 47 1 1250 1475 1 0

# master
set acc_calibration = 20,-50,22
set min_check = 1100
set serialrx_provider = IBUS
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set bat_capacity = 1600
set deadband = 10
set yaw_deadband = 10
set pid_process_denom = 1
set gyro_1_align_yaw = 900
set gyro_2_align_yaw = 900

profile 1

# profile 1
set p_pitch = 37
set i_pitch = 81
set d_pitch = 34
set f_pitch = 77
set p_roll = 34
set i_roll = 77
set d_roll = 32
set f_roll = 73
set p_yaw = 27
set i_yaw = 81
set f_yaw = 78
set d_min_roll = 18
set d_min_pitch = 20

rateprofile 1

# rateprofile 1
set thr_mid = 40
set thr_expo = 60
set throttle_limit_type = SCALE
set throttle_limit_percent = 85

# end the command batch
batch end```

**NOTE:** The diff dump is after removing the beeper function
**Setup / Versions**
 - Flight controller is a F7 flight controller from Banggood;
 - PPM transmitter
 - Connected with soldered pads, checked all wires, no issues.

mikeller, 2020-03-07:
@Givemeurcookies: To clarify, the reboot loop that you are reporting is only the ESCs rebooting, not the flight controller?

Can you try disabling the ESC beacons and see if this helps:

```
baecon -RX_LOST
beacon -RX_SET
```

If this fixes it then this looks like a bug in the ESC firmware causing problems when a beacon command is sent right after initialisation, and you should report this to the manufacturer of your ESC firmware.

etracer65, 2020-03-07:
@Givemeurcookies I can potentially see a problem with sending beeping commands to the ESCs while they're initializing. But I can't actually replicate the problem using BLHeli32 ESCs with the latest 32.7 firmware. In reality it takes a while for the flight controller to get started enough to start sending the beep commands (like more than 2 seconds). This should be more than enough time for the ESCs to finish their protocol detection and this is why I can't replicate the problem. It's possible that you have a signal wiring problem and the ESC(s) might be taking much longer to detect the signal and possibly colliding with the start of the beep commands.

etracer65, 2020-03-08:
@Givemeurcookies Looking deeper into this I think you need to check that your ESCs are up to date and/or possibly check your signal wiring. We already have a 3 second delay between when Betaflight starts talking to the ESC and the earliest that a beep command could be set. That 3 second delay is more than enough time for the ESC to detect the DSHOT signal.

helpsterTee, 2020-04-02:
@etracer65 
BLHeli32 32.7, BF 4.1.5
I can reproduce the problem. It happens for me with a long custom startup tone and bidirectional DShot enabled. I hadn't set up the receiver, so BF went into RX_LOST mode. Guess that wouldn't happen with a shorter startup tone. It could hit the 3 second delay...

mikeller, 2020-04-02:
@helpsterTee: I think the combination of bidirectional Dshot and custom startup tones is not supported - @sskaug should be able to clarify.

sskaug, 2020-04-04:
ESC input signal detection needs a stable input signal in order to work reliably. From the description, it sounds like the ESC music too long, and that the FC starts sending dshot commands while the ESC still detects the input signal.

closed 2020-05-13 by stale[bot]